espressif/esptool #235: Support uploading over UART1

projectgus, 2017-09-19:
ESP32 supports uploading over UART1 (GPIOs 9 & 10) if a sync signal is sent to UART1 RX after booting into bootloader ROM mode. However the flasher stub is currently hardcoded to UART0.

esptool.py can still be used now if the `--no-stub` option is passed, but eventually the stub should be updated to support the UART1 option.

negativekelvin, 2019-08-04:
bump

https://github.com/espressif/arduino-esp32/issues/2870

nifomorph, 2019-12-24:
Any progress on making the --no-stub work on UART1?

andrewteta, 2022-10-26:
Is there any way to access the bootloader (using esptool.py) via UART1 on ESP32C3? If anyone knows what it would take to do this (assuming it's not a hardware limitation), could they share? I have a project that would benefit from this, so I would be willing to try and implement it.