espressif/esptool #236: A fatal error occurred:failed to connect to espressif Timed out waiting for packet header

Akriti-S, 2017-10-04:
Hi,

I am getting this error when I am trying to upload the sketch after uploading the SPIFFS data folder in my esp8266. 

If I upload the code without uploading the sketch's data folder then I do not get this error however if I first upload Sketch Data folder and then try uploading the sketch. I get that error. Is there any way to get rid of it.

Thanks,
Akriti

projectgus, 2017-10-05:
Hi Akriti,

What ESP8266 board or hardware are you using?

Also, what Arduino IDE are you using to upload? I believe the ESP8266 Arduino IDE uses esptool-ck not esptool.py. esptool.py was (I thought) only used in the ESP32 Arduino IDE.

Angus

Akriti-S, 2017-10-06:
Hi, I am using Arduino 1.8.4. I am using nodemcu 1.0 ESP-12E. I upload the code by using esptool.py generally

projectgus, 2017-11-29:
Hi @Akriti-S ,

It looks like your dev board is failing to auto-reset into bootloader mode. You can try holding down the FLASH button while esptool.py tries to connect. If this doesn't work you can also try trapping RST while holding FLASH, while esptool.py tries to connect.

> If I upload the code without uploading the sketch's data folder then I do not get this error however if I first upload Sketch Data folder and then try uploading the sketch. I get that error. 

That's pretty odd, although it may also be related to auto-reset working once and then failing. Do you know what files are being flashed at which offsets? Is the flash size set correctly for your board?

projectgus, 2017-12-29:
Closing due to a lack of updates, but feel free to reopen if you'd like to continue the discussion about this configuration.

closed 2017-12-29 by projectgus