espressif/esptool #237: Wrong SPI and IROM address values for ESP32

lordcrc, 2017-10-06:
I'm trying to flash a Wemos Lolin32 Lite, but esptool fails while configuring the flash size.

While debugging I found that the `SPI_REG_BASE` as well as the `IROM_MAP_START`, `IROM_MAP_END` constants defined in `ESP32ROM` does not match the [ESP32 technical reference manual](http://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf).

Here's the values based on the technical reference manual:

```
    IROM_MAP_START = 0x400C2000 // was 0x400d0000
    IROM_MAP_END   = 0x40C00000 // was 0x40400000
    DROM_MAP_START = 0x3F400000
    DROM_MAP_END   = 0x3F800000

    # ESP32 uses a 4 byte status reply
    STATUS_BYTES_LENGTH = 4

    SPI_REG_BASE   = 0x3FF43000 // was 0x60002000
    EFUSE_REG_BASE = 0x3FF5A000 // was 0x6001a000
```
I am very new to the ESP universe, so maybe I'm missing something?

lordcrc, 2017-10-06:
I've now successfully flashed my Lolin32 Lite, turns out the main culprit was the baud rate. After reducing it to 115200 it worked like a charm. 

However, I'm still confused by the above discrepancy between code and documentation.

projectgus, 2017-12-29:
Hi @lordcrc,

Sorry, I thought someone got back to you about this a while ago.

>  IROM_MAP_START = 0x400C2000 // was 0x400d0000

This is discrepency between ESP-IDF (which uses 0x400d0000) and the reference manual. I think because of some issue using the first few pages of the IROM space. I have checked with the digital hardware team about the exact reason, and updating either the TRM or the ECO document.

>  IROM_MAP_END   = 0x40C00000 // was 0x40400000

I think this is a mistake in the TRM. Only 52 64KB pages of flash cache can be mapped concurrently. Will check.

> SPI_REG_BASE   = 0x3FF43000 // was 0x60002000
> EFUSE_REG_BASE = 0x3FF5A000 // was 0x6001a000

Some peripheral registers are mapped at two addresses. There's a small note at the top of 29.4 Register Summary in the TRM: "The base address for registers is 0x60008000 when accessed by AHB, and 0x3FF48000 when accessed by DPORT bus." We use the AHB range here (and in ESP-IDF) as a workaround for ECO 3.3 (as documented in the ECO guide).

closed 2019-10-18 by projectgus