espressif/esptool #238: Add option to use other devices instead of serial port to talk to Espressif chips

me21, 2017-10-13:
Please add option to use other devices (TCP port, FIFO file on Linux or pipe on Windows) to talk to Espressif chips. This will make it possible to flash ESP remotely by using some intermediate program that will forward the commands to real ESP port.

I think it will be impossible to set baud rate and to control DTR/RTS in that case, but it will be a well-known limitation, so that user will make necessary actions himself.

me21, 2017-10-13:
Although something similar can be implemented using https://freevirtualserialports.com/ (it can create virtual serial port connected to named pipe), but it requires another piece of software to be installed.

But it can be used as temporary workaround.

projectgus, 2017-10-16:
This functionally should be possible now but it's not documented as I haven't thoroughly tested it.

The pyserial library which esptool is built on supports RFC2217, which is a standards-compliant way to tunnel serial port traffic over a TCP telnet session. [Docs here](https://pyserial.readthedocs.io/en/latest/pyserial_api.html#rfc-2217-network-ports). If you use the form mentioned there as the port name when running esptool.py (`rfc2217://host:port`) then it should work, including flow control if DTR & RTS are hooked up at the other end.

Looks like you're on Windows. A quick search for "rfc2217 Windows" shows there are quite a few free options for serial servers on this platform.

pyserial also includes a simple RFC2217 server example:
https://github.com/pyserial/pyserial/blob/master/examples/rfc2217_server.py

Please let us know how you get on with this.

Angus

projectgus, 2017-11-14:
@me21 did you have any luck with this?

me21, 2017-11-15:
I haven't tested yet, but when I have, I'll let you know.

projectgus, 2017-11-15:
Thanks for the update @me21, no problems. I'm happy to leave this issue open until we have an officially documented way to use RFC2217.

me21, 2017-12-15:
Now I can confirm it works with RFC2217 ports with small PR #251.

projectgus, 2017-12-17:
Fantastic, thanks for confirming this @me21

projectgus, 2017-12-18:
Now documented (with some simple examples) at https://github.com/espressif/esptool/wiki/Remote-Serial-Ports

commit 234cf665 (2017-12-18): README: Add link to Remote Serial Ports wiki page

Closes #238
closed 2017-12-18 by projectgus