espressif/esptool #240: [esptool.py] A fatal error occurred in function save

YanMinge, 2017-10-24:
LOG
```
LINK build/application.elf
   text    data     bss     dec     hex filename
 732939  207404   71744 1012087   f7177 build/application.elf
Create build/application.bin
esptool.py v2.1

A fatal error occurred: Segment loaded at 0x401702a8 lands in same 64KB flash mapping as segment loaded at 0x40170290. Can't generate binary. Suggest changing linker script or ELF to merge sections.
make: *** [GNUMakefile:940：build/application.bin] 错误 2
```
How can I fix this issue?

projectgus, 2017-10-25:
Are you still using your custom modified version of MicroPython?

Are you able to attach the application.elf and application.map files  to the issue?

YanMinge, 2017-10-25:
Yes, I need my own version, I have a lot of my own code(https://github.com/YanMinge/esp32_pro). 
I hope to be able to use the latest esp-idf features

YanMinge, 2017-10-27:
Compared with the normal log, it seems that more of these things:
```
.gcc_except_table._Znaj
                0x40170290       0x16
 .gcc_except_table._Znaj
                0x40170290       0x16 c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/\libstdc++.a(new_opv.o)

.gcc_except_table.__gxx_personality_v0
                0x401702a8       0x1c
 .gcc_except_table.__gxx_personality_v0
                0x401702a8       0x1c c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/\libstdc++.a(eh_personality.o)

.gcc_except_table.__cxa_call_unexpected
                0x401702c4       0x18
 .gcc_except_table.__cxa_call_unexpected
                0x401702c4       0x18 c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/\libstdc++.a(eh_personality.o)

.gcc_except_table.__cxa_get_globals_fast
                0x401702dc       0x11
 .gcc_except_table.__cxa_get_globals_fast
                0x401702dc       0x11 c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/\libstdc++.a(eh_globals.o)

.gcc_except_table._ZN10__cxxabiv111__terminateEPFvvE
                0x401702f0       0x1d
 .gcc_except_table._ZN10__cxxabiv111__terminateEPFvvE
                0x401702f0       0x1d c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/\libstdc++.a(eh_terminate.o)
```

YanMinge, 2017-10-30:
The toolchain I'm currently using is
`esp32_win32_msys2_environment_and_toolchain-20170918`
and I replaced libstdc ++.a of toolchain version
`esp32_win32_msys2_environment_and_toolchain-20170111`
in path `xtensa-esp32-elf / lib / libstdc ++. a`

and this error does not happen again. I can successfully compile the fireware.bin file and download it to ESP32, but this will lead to some run-time error. 

I hope you can give me some possible direction to solve this problem,  I can go to try

YanMinge, 2017-11-09:
I have determined that this issue is caused by esp32_win32_msys2_environment_and_toolchain, so the issue can be closed

closed 2017-11-09 by YanMinge