espressif/esptool #241: Not able to upload from arduino ide

jayadevakash, 2017-10-25:
Arduino: 1.8.3 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Serial, 115200, 4M (3M SPIFFS)"

In file included from /Users/akashj/Desktop/esp8266_deauther-master/esp8266_deauther/esp8266_deauther.ino:11:0:
/Users/akashj/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include/ESP8266WiFi.h:28:38: fatal error: include/wl_definitions.h: No such file or directory
   #include "include/wl_definitions.h"
                                      ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

projectgus, 2017-10-25:
Hi @jayadevakash ,

This a compiling problem with Arduino for ESP8266, esptool.py is not involved.

I suggest trying here:
https://github.com/esp8266/arduino

Or you can try the esp8266 deauther project itself.

Angus

closed 2017-10-25 by projectgus

MZOHAR, 2019-05-06:
I WAS ABLE TO OVERCOME TRE PROBLEM, BUT I DID NOT UPLOAD IT (Im using UNO)
BUT AT LIST IT GIVE YOU hint :
I selected Board = NodeMCU 1.0 (ESP-12E)