espressif/esptool #242: Parallel programing

Avi-TelnT, 2017-10-28:
Hi,
Can you point me to some document that I can learn how to programming ESP32.
 I do like to connect several of ESP32 RX pin to the same one USB UART TX pin.
So with one flashing, I will get 10 devices flashed. 
I can connect the first ESP32 TX pin connected o USB UART RX pin to satisfy the esptool.py
I do not need esptool.py to confirm successful flashing, since all ESP32 devices will be 100% the same, and the user code, been flashed, will have the function to verify successful flashing.

IIM-Slide, 2018-04-19:
Did you find your answer @Avi-TelnT ? I am curioous

Avi-TelnT, 2018-04-19:
No. 
Can you help @IIM-Slide ?

projectgus, 2018-04-19:
Hi @Avi-TelnT ,

Sorry, for some reason I never replied to this when you asked it.

I think doing this with esptool.py may be possible but would require some custon modifications. The main problem I can see is that the initial "SYNC" process tends to be a little unpredictable - esptool will keep sending SYNC packets until it gets a valid response, and you'll see that even on the same hardware this can take a different number of tries each time.

The SYNC process is documented here:
https://github.com/espressif/esptool/wiki/Serial-Protocol#initial-synchronisation

You'll probably need to make some tweaks to esptool so that, for example, it sends a maximum number of SYNC packets before it assumes that all of the ESP32s are listening and that it can start flashing. Otherwise some still may still not be correctly running the bootloader at the correct baud rate.

Is this for a factory bulk flashing setup? Most of our customers who have this requirement do their flashing directly to the SPI flash chips using a bulk SPI flasher. If you're  ordering a high enough quantity of ESP-WROOM or WROVER modules from Espressif then you can also talk to our sales team about purchasing modules pre-flashed with your firmware.

Avi-TelnT, 2018-04-20:
Thank you, Mr. Angus Gratton, for the detailed replay.

IIM-Slide, 2018-04-20:
Thank you for the answers @projectgus. One precision though: does these solutions support flash encryption and Secure Boot? The way I understand these features, they need hardware to do so. 
[The doc](https://github.com/espressif/esp-iot-solution/blob/master/documents/security_solution/esp32_secure_and_encrypt_cn.md) I found (and ggl translated) implies that there is a way to program several boards in parallel with encryption, am I correct?

projectgus, 2018-04-23:
@IIM-Slide You may find the IDF docs for these features useful as well:
http://esp-idf.readthedocs.io/en/latest/security/flash-encryption.html
http://esp-idf.readthedocs.io/en/latest/security/secure-boot.html

There are at least two ways of programming multiple boards with secure boot & flash encryption enabled:
* In the recommended production config, keys are generated on the device on first boot. You can flash multiple boards in parallel for this configuration, as everything is flashed in plaintext and then security features are applied by the device on first boot. This is the recommended production configuration in the IDF docs.
* You can pre-calculate the encryption key and pre-flash it into the EFUSE of all devices. This is mentioned in the Chinese doc linked above. In this case, if you can find a way to write efuse in parallel then you can write the same flash encryption and secure boot keys to all devices and then flash. ("Reflashable" mode in the docs.) Note that using the same key for all devices in this way means that a compromise of one device's key(s) will compromise all devices.

IIM-Slide, 2018-04-23:
So if I understand correctly I have two options which allow me to
1) fully encrypt the firmware with simultaneous flashing but is time-expensive (on-device encryption) and I have no way of re-flashing them if something went wrong. 
2) re-flash with full encryption but  is extremely time-expensive (per-device programming) or extremely unsecure 
This is a real dilemma...  

The option I wish I had is to have a reflashable, uniquely fully-encrypted firmware that I could program in parallel. Even better would be to be able to program it using the Espressif Download tool. 
I guess sharing the encryption key with 10 devices (meaning I would change the key every time the download tool is done) is acceptable. 

Do you know if this is something doable?

projectgus, 2018-04-23:
Hi @IIM-Slide ,

You understand correctly.

> I have no way of re-flashing them if something went wrong.

In the most common production configurations, there's no provision for physically reflashing devices once they leave the factory (regardless of security requirements). OTA updates are supported with flash encryption enabled, so this is the commonly used way to update the device firmware (the bootloader isn't updateable, but we maintain bootloader backwards compatibility).

If you have a different production configuration that we may not have considered, please let us know.

> Do you know if this is something doable?

Something like this is technically possible. If you want to share one UART between all devices then you'll still need to be happy modifying esptool.py & espefuse.py to ignore status responses, etc. If you're happy to have one serial device per ESP then the Flash Download GUI has a batch flashing mode of this kind, already.

Another possible solution is to configure an in-factory key server of some kind, so each device comes up on first boot as part of end-of-line testing and downloads its unique keys from the server (who has a database with each key), and then enables the security features at this point. This is not currently supported in ESP-IDF, you'd have to add support yourself. However if you open a feature request on the ESP-IDF repo then we can look into the feasibility of supporting it: https://github.com/espressif/esp-idf/issues

radimkarnis, 2020-12-08:
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-08 by radimkarnis