espressif/esptool #246: Unable to kill esptool.py

liborw, 2017-11-05:
I have tried chip_id and esptool and it gets stuck, tried `ctrl-c`, `ctrl-\`, `kill -9`, force quit from activity monitor (I'm on Mac), closing the terminal, but the esptool is still there...only restart works.

projectgus, 2017-11-15:
Hi @liborw ,

Sorry for the slow reply

What hardware (including USB/serial adapter) do you have? Has esptool.py ever worked with this hardware?

There isn't anything we can do from esptool.py itself to create this kind of situation, the OS won't allow it. However the USB/serial driver which runs in kernel mode can get the operating system kernel stuck in a way where nothing can kill the process. This is particularly possible if the driver is poorly written or buggy.

Unfortunately there isn't much we can do from the esptool.py side. Looking for updated versions of the macOS USB/serial driver (or macOS itself) may resolve the issue, though.

The other thing which sometimes works is unplugging the device, which causes the operating system kernel to clean up the broken driver.

Angus

projectgus, 2017-12-29:
Closingas I don't think there's anything we can do from the esptool.py side, here.

closed 2017-12-29 by projectgus