espressif/esptool #248: Serial protocol wiki page is out of date

tehKaiN, 2017-11-15:
I'm trying to implement ESP32 flashing on platform where Python is not an option, so I thoroughly read what's on project's wiki. Also, I couldn't find any other info on the Internet about esp32 flashing process, but perhaps I dodn't know where to look.

Unfortunately it seems that protocol page is out of date, since it doesn't describe 0x10, 0x11 and 0x12 opcodes which I've captured with my logic analyzer.

Is there any reference for flashing procedure or do I have to go through esptool sources?

projectgus, 2017-11-15:
Hi @tehKaiN ,

Thanks for reminding me about this. Unfortunately that wiki page is quite old. It predates esptool being an official Espressif project and it's based on reverse engineering back when this was a community project.

I would like to update it with more details, ESP32-specific details, etc. but I haven't had time. As a stop-gap I've added a note to the top directing to this issue.

Reading the esptool source is probably the easiest way to get up to speed with the protocol quickly. Feel free to ask about any particular point which you're stuck on.

Angus

tehKaiN, 2017-11-20:
Hi there,

I've expanded some info about serial protocol on the wiki. I'm almost ready to implement everything, but my logic analyzer tells me that after writing to RAM I'm receiving kinda strange packet from ESP. Is it meaningful or should I just ignore it?

```
tx:
0xC0
0x00
0x06
0x08 0x00
0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x60 0x05 0x09 0x40
0xC0

rx:
0xC0
0x01
0x06
0x04 0x00
0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00
0xC0

rx:
0xC0
0x4F
0x48
0x41
0x49
0xC0
```

projectgus, 2017-11-28:
> after writing to RAM I'm receiving kinda strange packet from ESP. Does it have some meaning or should I just ignore it?

0x4F, 0x48, 0x41, 0x49 is the stub startup "OHAI" sequence. in a SLIP packet (you can search for it in the stub code and esptool.py. It's the first thing the stub sends once it starts running (This is why it comes after the ESP_MEM_END (0x6) command with the entry point address of the stub code.)

This is understandably a bit weird, and is mostly for historical reasons. Cesanta wrote the original version of the flasher stub, and their stub wrote OHAI when it started up and then spoke a custom protocol from then on. I rewrote the stub code to use a compatible protocol to the ESP8266 & ESP32 ROM loaders, so we could have the `--no-stub` option without too much duplicate code. But I kept "OHAI", I guess I found it kind of charming...

It's also the only thing the stub sends unsolicited. I guess the alternative would be to have the host send commands and poll for a running stub.

> I get response packets that have body and no status/error:
> And also packets with status/error but no body.
> So I guess status/error only occurs when no body is present. Am I right? And then there is the MD5 checksum packet, which response doesn't apply to this rule

... Yes. This is (also) for historical reasons due to the way the ROM loader evolved between ESP8266 and ESP32. I won't try to defend it as a coherent design!

* ESP_MEM_END (0x6) not returning status bytes, I had actually not noticed. The command code in esptool.py doesn't verify the return length at all, it just assumes the last two bytes are status. And as these are always zero, it was happy.
* The ESP_SPI_FLASH_MD5 (0x13) command is indeed the (very) odd one out here.

In general, if you can structure parsing of responses in a similar way to esptool.py (read an entire variable-length SLIP packet, then make assumptions based on the structure of the SLIP packet) you'll probably have a better time than trying to parse from the beginning based on the (inconsistent as you've seen) length headers. *sorry*.

tehKaiN, 2017-11-29:
Nothing to _sorry_ about. ;) I'd also keep OHAI, but I guess this should be documented somewhere. So on wiki I've added new paragraph called "protocol flow" and added info there. Also, there was no info that sync packet should be repeated until one gets response, so I've added that too.

I guess that after my struggle, docs will be as complete as necessary to implement new solution from scratch. ;)

bvernoux, 2017-11-29:
I have done some reverse engineering on that if you are interested
https://gist.github.com/bvernoux/6ddc02ed157417a4df37861488357c5a
Feel free to update the Serial protocol wiki with those informations

nkolban, 2018-01-08:
@tehKaiN  @bvernoux 

I find myself in need for a C library that is a functional equivalent for the esptool.py.  To that end I started reverse engineering the code.  However, I'd like to try and practice what I preach relating to open source and not attempt to re-invent the wheel for the sake of it.  Is there a project or repository that maybe y'all have started that we could use as the basis for a common/shared piece of work?  If not, would there be interest in starting one in the belief that there may be value if we all pulled together to compare notes and collaborate on such a task?

tehKaiN, 2018-01-08:
I guess it's a good idea. My code is unfortunately closed source since it's for commercial project of my company and I really can't do anything about it. Also it's not that generic since it's tailored to run on one specific microcontroller.

I've tried to leave as much clues as possible on esptool protocol wiki page, feel free to expand it if needed. Also, reading esptool's sources isn't that painful once you get used to python syntax.

Also, I'm going to use esp32 as part of my hobby projects so you can count me in for such rewrite, because having reasonably licensed open source tool written in C would be great.

ygator, 2018-01-08:
@nkolban 
I have a simple C program I threw together to test my problem related to #250.
If you are interested in starting with it then I can make it available.

nkolban, 2018-01-08:
@ygator Please and thank you sir!!

ygator, 2018-01-08:
[esp.zip](https://github.com/espressif/esptool/files/1613011/esp.zip)

projectgus, 2018-01-31:
Thanks again @tehKaiN for expanding the Wiki page.

I've been through and expanding it some more, and added some detail about common command flows for flashing, etc.:
https://github.com/espressif/esptool/wiki/Serial-Protocol

I'm considering that this issue can be closed now. If you find inaccuracies or shortcomings in the wiki protcol description, please open new issues to track this.

closed 2018-01-31 by projectgus