espressif/esptool #249: esptool: timeout after sending one command to esp8266 on Opensmart Adaptor

ctshh, 2017-11-20:
I have an ESP-01 sitting on top of a Opensmart USb to ESP-01 adaptor. It shows up as /dev/tty.wchusbserial1410 and can be access, for example, to read the flash or chip ID just fine (eg. the GND-GPIO0 connection is made ;):

```
$ esptool.py --port=/dev/tty.wchusbserial1410 chip_id
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x004c4de9
Hard resetting...
$
```

One problem seems to be that this setup only seems to accept one command at a time, then needs reboot by unplugging/plugging in:
```
$ esptool.py --port=/dev/tty.wchusbserial1410 chip_id
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x004c4de9
Hard resetting...
$ # second try, device not unplugged / plugged in
$ esptool.py --port=/dev/tty.wchusbserial1410 chip_id
esptool.py v2.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
$
```

Flashing seems to issue multiple commands, thus flashing (even after I just resetted the combo) always fails:

```
$ esptool.py --port=/dev/tty.wchusbserial1410 --baud 460800 write_flash --flash_size=1MB 0 Downloads/esp8266-20171101-v1.9.3.bin
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash params set to 0x0020
Compressed 600888 bytes to 392073...

A fatal error occurred: Timed out waiting for packet header
$
```

Is there a way to only issue one flashing command at a time, basically chopping up the flow of commands apparently issued when flashing?

projectgus, 2017-11-20:
I think two different things are happening here:

* The adapter you have doesn't seem to have DTR wired up for automatic resetting of the ESP8266. This means that esptool.py is not able to automatically reset the chip during the "Connecting..." phase and relies on you having manually done it (via power / unpower). You may have some luck passing `--after no_reset` on the command line, allowing you to run multiple esptool.py commands in a row, but this may not work correctly either...

* Flashing multiple files in one pass is not the same as running esptool.py multiple times. So this is something else, maybe to do with the ESP-01 or the usb/serial chip. Some things to try:
   * Lower the baud rate (try the default 115200, perhaps)
   * Try `esptool.py ... write_flash --no-compress --flash-size=1MB ...`
   * Try `esptool.py --no-stub .. write_flash ...`

Please let us know if any of these options help.

ctshh, 2017-11-21:
Thank you!!

`$ esptool.py --port /dev/tty.wchusbserial1410 --baud 115200  write_flash --no-compress --flash_size=1MB 0 esp8266-20171101-v1.9.3.bin` worked like a charm.
So, for future generations of knowledge seekers (do these issues stay "up", in anyway?), the above works for an Opensmart USB to ESP-01 adaptor.

projectgus, 2017-11-21:
> do these issues stay "up", in anyway?

They do.

> the above works for an Opensmart USB to ESP-01 adaptor.

I'm glad you found a workaround, but I'd like to fix this bug as well. Did any of the other suggestions work for you, or only the one posted here?

projectgus, 2017-11-21:
It's possible this is the same underlying issue solved by #247.

ctshh, 2017-11-24:
The `--no-stub` option worked, too:

```
$ esptool.py --port=/dev/tty.wchusbserial1410 --baud 115200 --no-stub write_flash --flash_size=1MB 0 Downloads/esp8266-20171101-v1.9.3.bin
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Enabling default SPI flash mode...
Configuring flash size...
Erasing flash...
Flash params set to 0x0020
Took 2.63s to erase flash block
Wrote 601088 bytes at 0x00000000 in 71.1 seconds (67.6 kbit/s)...

Leaving...
Hard resetting...
$
```
->

```
ets_task(40100130, 3, 3fff837c, 4)
OSError: [Errno 2] ENOENT

MicroPython v1.9.3-8-g63826ac5c on 2017-11-01; ESP module with ESP8266
Type "help()" for more information.
>>>
```

👍 😄

KaranArora1, 2018-07-09:
Thank you so much! Ran into the same issue (and my ESP wouldn't flash) and after hours of googling and wasting time, this solved it.

projectgus, 2018-07-09:
Closing as this seems to be resolved.

closed 2018-07-09 by projectgus