espressif/esptool #250: Commands being sent too fast for ESP32

ygator, 2017-12-10:
I have an esp32 that used to work and have now ran into a problem where I am always getting timeouts. 
 After debugging for 2 days, learning the protocol and trying different things I have determined that the commands are being sent to the esp32 too quickly.  My solution was putting a time.sleep(.25) right before the self._port.write(buf) in the write function.  There may be other places as well, but since this change I've not had a timeout since.

projectgus, 2017-12-10:
HI @ygator ,

Thanks for reporting this. Can you give any more details about your setup, such as:

What host OS are you using, and which development board? At what point were you getting timeouts? (If you get the latest esptool.py, can you run some commands with `esptool.py --trace` and let us know the output?)

Thanks,

Angus

projectgus, 2017-12-10:
This sounds very similar to the issues fixes in #247, although without more details I'm guessing...

ygator, 2017-12-11:
This was on both Windows 10 and Arch linux.  I was getting timeouts on the early stuff and was not able to get to the real stuff such as flashing.  Timeout on sync, detecting chip, stub, etc...  The board I have is a DOIT ESP32 DEVKIT V1.  I manually wrote a little vb6 serial port program and was following what the esptool was doing.  Other then having to send sync multiple times to get a response the other commands I tried worked fine every time.  So I figured the board is fine and it is something with the esptool.  Of course running it myself there was a lot of time between the commands and so I tried the sleep before the write and it worked great.

Below is a trace I still had.  Good sync, good UART_DATA_REG_ADDR read, timeout on efuse(3) read.  The failure on it is just the normal timeout waiting for a reply from a command.  Since the ESP32 never gets (or only partially gets the command) nothing is returned and hence the timeout.

Detecting chip type...
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=b'x\x00\x00`'
TRACE +0.000 Write 14 bytes: b'\xc0\x00\n\x04\x00\x00\x00\x00\x00x\x00\x00`\xc0'
TRACE +0.004 Read 1 bytes: b'\xc0'
TRACE +0.000 Read 13 bytes: b'\x01\n\x04\x00\x00%\x12\x15\x00\x00\x00\x00\xc0'
TRACE +0.000 Full packet: b'\x01\n\x04\x00\x00%\x12\x15\x00\x00\x00\x00'
 ESP32
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=b'\x0c\xa0\x01`'
TRACE +0.000 Write 14 bytes: b'\xc0\x00\n\x04\x00\x00\x00\x00\x00\x0c\xa0\x01`\xc0'
TRACE +3.004 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header

ygator, 2017-12-11:
In regards to #247, my sleep is before the write with no change when reading and the driver is for CP210X, but on both Windows and Linux.  @rmokerone may want to revert back and try this fix to see if it works.

ygator, 2017-12-11:
By the way the 0.25 sleep was just something I found worked.  I just did some more testing and it appears that sleeping for 0.01 is working as well.  Though, anything smaller and I am getting timeouts again or maybe the resolution of sleep is only 0.01.  I'll do some more testing tomorrow.

projectgus, 2017-12-11:
Thanks for providing all these details. We use CP2102 in the official Espressif dev boards (the DOIT boards are a very similar design) with no issues. I wonder what the difference is, I have heard of fake CP2102s so this is a possibility.

Sleep 0.010 will have a lot less of an impact than 0.25, which makes it a more viable fix (I'm reluctant to impact performance of all chips on all platforms because of bugs that only effect a small subset).

I'm also curious if the fix in this branch helps, if you have time to try it out:
https://github.com/espressif/esptool/tree/experiment/ch340_timeout_bug

(This is for the possibility that it's not the write which is lost, but somehow the following read is lost at the driver layer.)

If you're feeling particularly adventurous, it would be interesting to see the USB-level behaviour of your chip (which can be captured from inside Arch using usbmon):
https://wiki.wireshark.org/CaptureSetup/USB

ygator, 2017-12-11:
python esptool.py --trace --port /dev/ttyUSB0 --baud 115200 --before no_reset --after no_reset read_mac
esptool.py v2.3-dev
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.015 Read 1 bytes: b'\xc0'
TRACE +0.000 Read 111 bytes: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00\xc0\xc0\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00\xc0\xc0\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00\xc0\xc0\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00\xc0\xc0\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00\xc0\xc0\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00\xc0\xc0\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00\xc0\xc0\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00\xc0'
TRACE +0.000 Full packet: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00'
TRACE +0.000 Full packet: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00'
TRACE +0.000 Full packet: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00'
TRACE +0.000 Full packet: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00'
TRACE +0.000 Full packet: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00'
TRACE +0.000 Full packet: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00'
TRACE +0.000 Full packet: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00'
TRACE +0.000 Full packet: b'\x01\x08\x04\x00\x07\x07\x12 \x00\x00\x00\x00'

Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=b'x\x00\x00`'
TRACE +0.000 Write 14 bytes: b'\xc0\x00\n\x04\x00\x00\x00\x00\x00x\x00\x00`\xc0'
TRACE +3.005 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header

ygator, 2017-12-11:
Frame 31: 78 bytes on wire (624 bits), 78 bytes captured (624 bits) on interface 0
    Interface id: 0 (usbmon3)
        Interface name: usbmon3
    Encapsulation type: USB packets with Linux header and padding (115)
    Arrival Time: Dec 11, 2017 11:35:48.190835000 EST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1513010148.190835000 seconds
    [Time delta from previous captured frame: 0.003400000 seconds]
    [Time delta from previous displayed frame: 0.003400000 seconds]
    [Time since reference or first frame: 1.168104000 seconds]
    Frame Number: 31
    Frame Length: 78 bytes (624 bits)
    Capture Length: 78 bytes (624 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: usb]
USB URB
    [Source: host]
    [Destination: 3.2.1]
    URB id: 0xffff9b239e374240
    URB type: URB_SUBMIT ('S')
    URB transfer type: URB_BULK (0x03)
    Endpoint: 0x01, Direction: OUT
        0... .... = Direction: OUT (0)
        .... 0001 = Endpoint number: 1
    Device: 2
    URB bus id: 3
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1513010148
    URB usec: 190835
    URB status: Operation now in progress (-EINPROGRESS) (-115)
    URB length [bytes]: 14
    Data length [bytes]: 14
    [bInterfaceClass: Unknown (0xffff)]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000000
    Number of ISO descriptors: 0
Leftover Capture Data: c0000a04000000000078000060c0

Frame 32: 64 bytes on wire (512 bits), 64 bytes captured (512 bits) on interface 0
    Interface id: 0 (usbmon3)
        Interface name: usbmon3
    Encapsulation type: USB packets with Linux header and padding (115)
    Arrival Time: Dec 11, 2017 11:35:48.191415000 EST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1513010148.191415000 seconds
    [Time delta from previous captured frame: 0.000580000 seconds]
    [Time delta from previous displayed frame: 0.000580000 seconds]
    [Time since reference or first frame: 1.168684000 seconds]
    Frame Number: 32
    Frame Length: 64 bytes (512 bits)
    Capture Length: 64 bytes (512 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: usb]
USB URB
    [Source: 3.2.1]
    [Destination: host]
    URB id: 0xffff9b239e374240
    URB type: URB_COMPLETE ('C')
    URB transfer type: URB_BULK (0x03)
    Endpoint: 0x01, Direction: OUT
        0... .... = Direction: OUT (0)
        .... 0001 = Endpoint number: 1
    Device: 2
    URB bus id: 3
    Device setup request: not relevant ('-')
[... 444 lines cut ...]
    [Source: 3.2.0]
    [Destination: host]
    URB id: 0xffff9b239e0a8d80
    URB type: URB_COMPLETE ('C')
    URB transfer type: URB_CONTROL (0x02)
    Endpoint: 0x00, Direction: OUT
        0... .... = Direction: OUT (0)
        .... 0000 = Endpoint number: 0
    Device: 2
    URB bus id: 3
    Device setup request: not relevant ('-')
    Data: not present ('>')
    URB sec: 1513010151
    URB usec: 204369
    URB status: Success (0)
    URB length [bytes]: 0
    Data length [bytes]: 0
    [Request in: 41]
    [Time from request: 0.001952000 seconds]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000000
    Number of ISO descriptors: 0
    [bInterfaceClass: Unknown (0xffff)]

ygator, 2017-12-11:
So as you can see the v2.3-dev still gives timeout.  The usb capture shows nothing returned for the UART_DATA_REG_ADDR register read command.

projectgus, 2017-12-11:
Thanks for this, @ygator . Any chance you could please post the USB capture as a .pcap or .pcapng file? Easier to follow than plain text.

ygator, 2017-12-11:
[usb.zip](https://github.com/espressif/esptool/files/1549682/usb.zip)

projectgus, 2017-12-11:
That's interesting. Here's that same command running successfully on a genuine Core Board:
[coreboard_sync.zip](https://github.com/espressif/esptool/files/1549717/coreboard_sync.zip)

I don't see any particular differences. The response to the sync packet has a different payload, but I think this is mostly a coincidence rather than indicative of any issues with the loader behaviour. And it seems like sometimes your ESP32 responds to the first command after the sync, then stops responding, so that indicates it's not a problem with the sync itself.

I am inclined to think this is a hardware bug in the "CP2102N" rather than anything with the driver or the ESP32... Although it could be power related, with the extra delays smoothing out the power draw a bit.

Can you try swapping a different USB cable and see if the behaviour changes? I wouldn't expect it to, but still...

projectgus, 2017-12-11:
Also, do you have a link to the place you bought the DOIT from? I'd like to get one for myself if I can.

ygator, 2017-12-12:
I purchased it on aliexpress back in June, but the store does not exist any more.  The product title was "Official DOIT ESP32 Development Board WiFi+Bluetooth Ultra-Low Power Consumption Dual Core ESP-32 ESP-32S ESP 32 Similar ESP8266".  I don't believe I had any timeout issues when I first was working with it.  I then flashed it and it was in place and any updates were flashed OTA.

I tried several different usb cables with no change.

projectgus, 2017-12-12:
That's interesting. One last question then I promise I'll stop asking you to test things, if you roll back to [esptool.py v2.0](https://github.com/espressif/esptool/releases/tag/v2.0) (which was the first stable release with ESP32 support), does anything change?

ygator, 2017-12-12:
No change.  Only thing that works is adding the sleep before the write.

projectgus, 2017-12-12:
(Also, I picked up one DOIT ESP32 board from their official Taobao store, plus two cheaper ones that may be clones from different stores. Will be interested to see how they all behave when they arrive...)

projectgus, 2017-12-12:
>  I don't believe I had any timeout issues when I first was working with it.

> No change. Only thing that works is adding the sleep before the write.

That seems really strange... What could have changed?

ygator, 2017-12-12:
A lot has changed for me.  I recently upgraded to Windows 10 Fall Creator Update.  I have gone through many versions of the Arduino IDE and each time I installed the ESP32 and ESP8266 boards.  When I was getting the timeouts I read about the older CP210X might be the issue so I updated those to the latest.  Also maybe python and/or python-serial changed during the past 6 months as well.

Most of that is ruled out since it happens the same when just using esptool on arch.
So that would leave python and/or python-serial.
Or of course it could just be my ESP32 board.  It would be interesting to know if all the other timeouts people are seeing (which are not related to not being in programming mode) can be remedied by the sleep.

projectgus, 2017-12-12:
I see. I guess on the Linux side there is also kernel version (USB host behaviour & cp2102 driver), although at the usbmon level these seem fairly standard.

I'm going to wait until my boards turn up and see if I can reproduce this. I'm not ruling out adopting your suggested fix (especially if it's only a 10ms sleep), but I'm wary about merging changes like this before we know what the scope is and what exactly we're fixing (like you said, this may just be your board).

projectgus, 2017-12-27:
Hi @ygator ,

I got three totally different DOIT boards from three different taobao sellers:
[image]

(Pen marked one is official from doit.am, the other two are clones.)

All three seem to work fine from my Arch Linux development system. Does yours match any of those?

Is it possible you have a defective development board?

ygator, 2017-12-27:
[image]

In a C program I got the wait time down to usleep(5000) = 5ms, 4ms is not enough.  The board works fine otherwise.

Don't get hung up on Arch has I am just using it now since it is easier to test with.  I first encountered the problem on Windows 10 trying to flash from arduino ide.

projectgus, 2017-12-27:
That's interesting. Your board layout is identical to the middle one in the photo I posted, although my one has an ESP32S module not a genuine ESP-WROOM32.

Is this the only ESP32 development board you have?

> Don't get hung up on Arch has I am just using it now since it is easier to test with. I first encountered the problem on Windows 10 trying to flash from arduino ide.

I'm not, it just happens that's what I use as my development machine. I'm away from my physical Windows 10 computer right now but I can test with it next week.

ygator, 2017-12-27:
The bottom has:
doit logo
ESP32 DEVKIT V1
www.doit.am

I do have another board I just got in November, but there is something wrong with it as I cannot send to the board.  I can receive, but just cannot send to it.  So I think there is something wrong with the rx line.  I was able to get a credit for it.  Btw, it was the same seller as the other board.  The ESP32 writing is a little different.  It just has a bigger CE marking without the 1313 after it.

projectgus, 2018-01-23:
Hi @ygator,

Have fixed an issue that might be the same as this one. Can you please try the latest master branch (with commit linked above) and let me know if it fixes the problem.

Angus

ygator, 2018-01-23:
It does work, but unfortunately the other versions are working as well.  I'll try on my windows computer and let you know.

projectgus, 2018-01-23:
Interesting, thanks for checking. The bug we fixed just now only seems prominent at higher baud rates, and possibly is a race condition with the host computer (for what that's worth.)

ygator, 2018-01-23:
It worked, but unfortunately so did the older versions.  On Windows manually running the installed esptool.exe times out, but running esptool.py works.  Also running the esptool.py v.2.3dev works as well.  How do I get the .py version into an .exe version to try?

projectgus, 2018-01-24:
The only .exe I know of is built as part of arduino-esp32. I'm not sure if @me-no-dev publishes the scripts for this - @me-no-dev can you help us out with that?

The difference may be the pyserial version (a pyserial version will be bundled in the .exe)

me-no-dev, 2018-01-24:
Compilation is done really easy. I have python2 installed on a windows machine, also the [pyinstaller](http://www.pyinstaller.org) module. For easier command prompt, I use gitbash that is bundled with [GitHub Desktop](https://central.github.com/deployments/desktop/desktop/latest/win32) for Windows and have enabled in the right click menu. From then on I open the folder that contains `esptool.py` => rightclick =>open git bash in that folder => execute ```pyinstaller --distpath . -F esptool.py``` =>exe is in that folder :)

ygator, 2018-01-24:
@me-no-dev  Thanks

@projectgus It is working on windows.

projectgus, 2018-01-24:
Thanks @me-no-dev & @ygator. In that case I'm going to close this issue, as it seems like this particular case is fixed in v2.2.1.

closed 2018-01-24 by projectgus