espressif/esptool #252: Cannot flash Feather Huzzah with CircuiPython

mkingdom, 2017-12-16:
I am trying to flash CircuitPython 2.1.0 to my Feather Huzzah.

The erase_flash function works fine and the boards red led is lit low ready to be flashed.
When I issue the write_flash command I'm getting a fatal error:

Marks-iMac:desktop mark$ esptool.py --port /dev/tty.SLAB_USBtoUART erase_flash
esptool.py v2.2
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.6s
Hard resetting...
Marks-iMac:desktop mark$ esptool.py --port /dev/tty.SLAB_USBtoUART --baud 460800 write_flash 0 adafruit-circuitpython-feather_huzzah-2.1.0.bin
esptool.py v2.2
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
Marks-iMac:desktop mark$

ygator, 2017-12-16:
You can try lowering the baud rate or if you look at #250 you might want to try adding time.sleep(0.01) right before the self._port.write(buf) in the write function.

mkingdom, 2017-12-16:
@ygator You are a star. Thanks. Reduced baud rate 230400 and problem solved.

closed 2017-12-16 by mkingdom