espressif/esptool #253: ESP32 target not being reset on Cygwin

sysprogs, 2017-12-20:
When running esptool.py with ESP32 under Cygwin, resetting the target after programming leaves the device in the bootloader. Modifying hard_reset() to set DTR to False explicitly seems to solve this.

```
    def hard_reset(self):
        self._port.setDTR(False)  # IO0=HIGH
        self._port.setRTS(True)  # EN->LOW
        time.sleep(0.1)
        self._port.setRTS(False)

```

Could this workaround be added to the repo, or would it break some scenarios on Linux?

projectgus, 2017-12-29:
Hi @sysprogs ,

This seems reasonable as a change. Are you using `--before no_reset` when seeing this behaviour?

I'm asking because DTR should already be False due to the reset logic. The only remaining case is when `--before no_reset` is passed, when DTR looks like it will default to asserted (True) and remains this way. This may also cause a bug with hard_reset on other platforms.

Angus

sysprogs, 2018-01-02:
Hi,
Nope, this happens without the "--before" argument. Maybe the stub closes and reopens the port at some point (resetting DTR under Cygwin and keeping it off under Linux)?

If this helps, you can get a full Cygwin environment where this can be reproduced here: http://sysprogs.com/files/gnutoolchains/esp32/esp32-gcc5.2.0-r7.exe

radimkarnis, 2020-12-08:
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-08 by radimkarnis