espressif/esptool #256: A fatal error occurred: MD5 of file does not match data in flash!

jnachtigall, 2017-12-27:
I purchased this [NodeMCU Lua Amica Module ESP8266 ESP-12E](https://www.amazon.de/gp/product/B01N4OYOKD/ref=oh_aui_search_detailpage?ie=UTF8&psc=1) at amazon.

When trying to flash I always get
```
$ esptool.py --port /dev/ttyUSB0 write_flash 0x00000 ~/Downloads/nodemcu-master-4-modules-2017-12-27-22-09-15-integer.bin 
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x4a25bf, SizeID=0x4a), defaulting to 4MB
Flash params set to 0x0240
Compressed 389344 bytes to 251389...
Wrote 389344 bytes (251389 compressed) at 0x00000000 in 22.2 seconds (effective 140.0 kbit/s)...
File  md5: 1d2be650c78635463e5d1fe9551121df
Flash md5: d5a03ae3037154fe9bb5f6155f3974fe
MD5 of 0xFF is 04139a8f075f846eaa8362e20893563e

A fatal error occurred: MD5 of file does not match data in flash!

```

I tried different version of the [cloud build service](https://nodemcu-build.com/): First 10 different modules, then I read that this might result in a too big flash image. So I went down to 7 and now 4 modules. But flashing always results in this `A fatal error occurred: MD5 of file does not match data in flash!` error. I tried a dozen times now and what irritates me is the `Unknown` flash size.

Don't know if important (I am a NodeMCU beginner) but `flash_id` says:
```
$ esptool.py --port /dev/ttyUSB0 flash_id
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Manufacturer: bf
Device: 254a
Detected flash size: Unknown
Hard resetting...
```

jnachtigall, 2017-12-27:
FWIW, same error when flashing with `-b 9600`

projectgus, 2017-12-27:
Hi @jnachtigall ,

Looks like you have a dev board with an SST (now Microchip) flash chip:
https://chromium.googlesource.com/chromiumos/third_party/flashrom/+/master/flashchips.h#697
https://chromium.googlesource.com/chromiumos/third_party/flashrom/+/master/flashchips.h#721

I haven't heard of ESP8266 boards with these chips before (and they don't follow the more common flash ID numbering scheme, which is why esptool.py can't detect the size).

To work around the flash size warning you can pass `write_flash -fs 4MB` to esptool.py.

This won't solve the problem that esptool.py is reading back different data after it's written, though (the MD5 mismatch errors). Looking at the Microchip datasheet for the flash chip, it looks like it should support the standard flash commands which are used by ESP8266.

Do you get the same incorrect "Flash md5: " hash each time? If it's the same each time, this can indicate that some write protection bits are set. What is the output of `esptool.py read_flash_status --bytes 1`?

jnachtigall, 2017-12-27:
@projectgus, Thanks for the quick reply 👍 

Adding `fs 4MB` I get:
```
$ esptool.py --port /dev/ttyUSB0 write_flash -fs 4MB 0x00000 ~/Downloads/nodemcu-master-4-modules-2017-12-27-22-09-15-integer.bin 
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0240
Compressed 389344 bytes to 251389...
Wrote 389344 bytes (251389 compressed) at 0x00000000 in 22.3 seconds (effective 139.9 kbit/s)...
File  md5: 1d2be650c78635463e5d1fe9551121df
Flash md5: d5a03ae3037154fe9bb5f6155f3974fe
MD5 of 0xFF is 04139a8f075f846eaa8362e20893563e

A fatal error occurred: MD5 of file does not match data in flash!

```

> Do you get the same incorrect "Flash md5: " hash each time?

Yes, it is always the same.

 > If it's the same each time, this can indicate that some write protection bits are set. 

I don't know if related but I also tried to flash using the Arduino IDE (after following various tutorials on how to set it up for the NodeMCU). This also always ends up with the following in the console which seems strange as I have seen screenshots where this did not end up so quickly:

```
Archiving built core (caching) in: /tmp/arduino_cache_876296/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_12d2032a6629f55d6f3287f8fcf91765.a
Sketch uses 226617 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31824 bytes (38%) of dynamic memory, leaving 50096 bytes for local variables. Maximum is 81920 bytes.
Uploading 230768 bytes from /tmp/arduino_build_349017/TestEspApi.ino.bin to flash at 0x00000000
................................................................................ [ 35% ]
................................................................................ [ 70% ]
..................................................................               [ 100% ]

```
So this also looks kind of "hanging" at 100%. As said I do not know if this is related but maybe writing here also does not work, because even the simplest Hello-Worlds via Serial.println does not work.

> What is the output of esptool.py read_flash_status --bytes 1?

```
$ esptool.py read_flash_status --bytes 1
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Status value: 0x0000
Hard resetting...
```

jnachtigall, 2017-12-28:
Uh, sorry, I might have misunderstood your question about the `same incorrect "Flash md5: " hash each time`. It is the same incorrect Flash md5 for the same image file. However, it is different if the image file is different. For instance, for the 7-module flash image it is `Flash md5: b3c0b4a6c6a4253b3ea94a506b2248c7` whereas it was `Flash md5: d5a03ae3037154fe9bb5f6155f3974fe` for the 4-modules flash image from above.

See here:
```
$ esptool.py --port /dev/ttyUSB0 write_flash 0x00000 ~/Downloads/nodemcu-master-7-modules-2017-12-27-21-32-40-float.bin esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x4a25bf, SizeID=0x4a), defaulting to 4MB
Flash params set to 0x0240
Compressed 415552 bytes to 267685...
Wrote 415552 bytes (267685 compressed) at 0x00000000 in 23.7 seconds (effective 140.1 kbit/s)...
File  md5: 689366eca7ddf796aba779d732beda0f
Flash md5: b3c0b4a6c6a4253b3ea94a506b2248c7
MD5 of 0xFF is fe640e2d5882be7a50cde1238b6da159

```

Anyway, I've ordered a NodeMCU board from another vendor on amazon now, hopefully this will work. 

But let me know if I can add any other info here or should any other test.

projectgus, 2017-12-28:
Hi,

You understood correctly. The diffrentiator is that for random data corruption (maybe one of the SPI data lines is faulty) the same "File md5" will read back a different "Flash md5" each time. For consistent corruption (write protection, problems erasing, command compatibility problems with the flash chip) the same "File md5" will read back the same "Flash md5" each time.

Moving to a new development board with a known-compatible flash chip is probably the best option at this stage. You could try to diagnose what specifically is incompatible with the Microchip flash chip, but probably it won't be a problem which is easy to fix (even if esptool.py can be made compatible, the ESP8266 ROM routines and SDK probably won't be.)

jnachtigall, 2017-12-29:
Thanks for the follow-up. The new board I ordered works flawlessly. Hence, I am closing since there's nothing actionable for esptool that could be done as far as I see it.

Thanks again very much for the answers and help. Very much appreciated!

closed 2017-12-29 by jnachtigall