espressif/esptool #259: ESP8266 12E Timeout/Invalid head

karenYerevan, 2018-01-04:
Hi
I have ESP8266 12E and arduino uno
python esptool.py --port COM5 flash_id
I have error  

> A fatal error occurred: Timed out waiting for packet header
OR
A fatal error occurred: Invalid head packet  ('x')

Arduino uno ------------ ESP8266 12E
TX ---> TX
RX ---> RX
GND ---> GND
GND ---> GPIO0

[image]

projectgus, 2018-01-04:
Hi @karenYerevan ,

Try adding `esptool.py --before no_reset`, I think the Arduino USB/serial interface can be confused by the automatic reset DTR/RTS operations.

If this doesn't work, can you please paste the full output from esptool.py not just the error?

Angus

karenYerevan, 2018-01-04:
Thank you for attention

[image]

projectgus, 2018-01-04:
Do you get anything if you add `esptool.py --trace --before no_reset ...`?

If you don't get any trace output, triple-check all the pin connections (TX/RX aren't swapped, ESP8266 has been reset manually with GPIO0=GND, etc.)

karenYerevan, 2018-01-04:
ohhhh
i get this ((((

[image]

karenYerevan, 2018-01-04:
sometimes i have error like this 
A fatal error occurred: Invalid head of packet ('\x01')

projectgus, 2018-01-04:
Do you see any lines saying "Read"? You may see something useful in the output when you get "invalid head of packet", ie some input from the ESP8266.

karenYerevan, 2018-01-04:
Unfortunately there is no line Read,
There is only this lines is repeat many time 

> _TRACE +0.063 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
> TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
> TRACE +0.109 Timed out waiting for packet header

karenYerevan, 2018-01-04:
now i see Read lines

[image]

projectgus, 2018-01-04:
It looks like the serial data is being corrupted, maybe by the Arduino USB/serial somehow.

You can try lowering the baud rate (try really low like `-b 9600`). If this doesn't work then you may need to seek out a different USB/serial interface.

karenYerevan, 2018-01-05:
bound doesn't work for me ((((
which usb serial you advice ?

projectgus, 2018-01-07:
Almost any 3.3V "low voltage TTL" serial adapter should work. I have some based on FTDI chips and some based on Cypress chips, but they're all very similar. :)

If you want an ESP8266 development board, you might also consider something like "NodeMCU" where the USB/serial is integrated on the development board.

projectgus, 2018-01-23:
I'm going to close this as I think it's a hardware issue. However please reopen if you think esptool.py is doing the wrong thing here.

closed 2018-01-23 by projectgus

nikyc, 2018-04-18:
Hello
It looks like I have exactly the same problem. I'm using a safe-made board with ESP12-F and FT231X. The hardware is triple-checked and everything looks fine, this being the reason I write here.
Every time I use _**esptool**_, I get back the following:

```
PS H:\> esptool.py -p COM5 -t  --chip esp8266 erase_flash
esptool.py v2.2
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUU
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.106 Timed out waiting for packet header
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.106 Timed out waiting for packet header
.TRACE +0.053 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.106 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
```

I saw the FT231X being used in a couple of projects, successfully. 
@karenYerevan did you had any results with another USB-to-Serial converter?

projectgus, 2018-04-18:
Hi @nikyc ,

It looks like your ESP8266 is never sending any serial data. If you open a normal serial connection to the ESP8266 and 74880bps and reset the board (with GPIO0 high or low), do you see any output at all on the serial link? If not, this is a hardware issue of some kind - either the ESP8266 isn't enabling correctly or the USB/serial chip is not correctly relaying the serial output.

nikyc, 2018-04-19:
Thanks for your answer @projectgus !

Let's start by saying that my board has the following schematic:
[image]

I did what you asked for in the last message. I also changed both the ESP module and the FT231. The results were the same as before. Below you can see the trace output and some scope captures done on the RX and TX pins of the ESP module:

**GPIO0 high @ reset:**
`PS H:\> esptool.py -p COM9 -t -b 74880 --before no_reset --after no_reset erase_flash
esptool.py v2.2
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.010 Read 1 bytes: b'('
TRACE +0.000 Read invalid data: b'('
TRACE +0.001 Remaining data in serial buffer: b''
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.010 Read 1 bytes: b'('
TRACE +0.001 Read invalid data: b'('
TRACE +0.001 Remaining data in serial buffer: b''
.TRACE +0.057 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.010 Read 1 bytes: b'('
TRACE +0.000 Read invalid data: b'('
TRACE +0.006 Remaining data in serial buffer: b''
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.011 Read 1 bytes: b'('
TRACE +0.000 Read invalid data: b'('
TRACE +0.001 Remaining data in serial buffer: b''
_

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (b'(')`
[image]

**GPIO low @ reset:**
`PS H:\> esptool.py -p COM9 -t -b 74880 --before no_reset --after no_reset erase_flash
esptool.py v2.2
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.102 Timed out waiting for packet header
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.102 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.103 Timed out waiting for packet header
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.102 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.103 Timed out waiting for packet header
.TRACE +0.053 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.106 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header`
[image]

I think that the last thing I can do is to change the FT231 with another converter and see what happens.

projectgus, 2018-04-20:
> As you can see the signal is a lot noisier when trying to boot (GPIO0 low) and the TX of ESP module doesn't work. it seems like it can't be pulled low.

The noise is a little odd, but it doesn't look like enough to cause any permanent problems.

I think you're not seeing anything on U0TX with GPIO0 low because the ESP8266 isn't sending any response (after all, it works with GPIO0 high). This is normal in bootloader mode, until a command is successfully received - so it could be a boot mode issue, or it could be that the U0RX pin is open circuit at the ESP8266 pad (cold joint?) so no command is getting through.

To check the boot mode, measure the voltages of the EN, RESET, GPIO0, GPIO15, GPIO2 at the module and consult this docum,entation: https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection

You may also discover something interesting by using miniterm.py to read the full bootloader output when GPIO0 is high: `miniterm.py COMx 74880`

nikyc, 2018-04-25:
Hello again,

It seems that, in my case, the problem was a weird one. I have used a ground plane under the ESP module which, for some reason, was messing with the whole memory communication procedure.

This is the layout before, at the time I was unable to talk with the ESP:
[image]

This is the layout now when everything goes fine, even at 460800 baud:
[image]

Nothing else changed and now both the manual and auto-reset procedures are working. Maybe someone can find this useful in the future.

@projectgus, thanks for your guidance!

projectgus, 2018-04-25:
Glad you got it sorted.

BTW, one thing that might explain this, the ground plane under the module in the first revision doesn't have a clear path to the the rest of the ground or the decoupling caps (only the small notch back at the GND pin). This may cause it to couple high frequencies, kind of like an antenna. Adding some stitching vias into the bottom ground plane, and making sure you have a short straight (as possible) path back to the decoupling caps, may also help.

But removing it entirely sounds like it worked, too. :)

Amrandroid, 2020-03-07:
It worked after adding a capacitor btw enable and ground 
here is a helpful link
https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/

jiangytcn, 2020-04-13:
hi @Amrandroid I had exactly same issue as yours with ESP8266-01, however per the datasheet of the esp-01, the `EN` pin should be in `HIGHER` value, and how do you wire it to the ground with `200mF`  capacitor ?  is there any chance that show us the layout? 

BTW, I'm using raspberry to operate the esp8266-01 chip