espressif/esptool #261: Timeout under windows

ftoyon, 2018-01-10:
I want flash ESP32 without installing the esp-idf.

I try to use esptool but it's fail whit time out.
```
c:\...>esptool.py --chip esp32 --port COM7 --baud 921600 read_mac
esptool.py` v2.2
Connecting...
Traceback (most recent call last):
  File "C:\Python36\Scripts\esptool.py-script.py", line 11, in <module>
    load_entry_point('esptool==2.2', 'console_scripts', 'esptool.py')()
  File "c:\python36\lib\site-packages\esptool.py", line 2585, in _main
    main()
  File "c:\python36\lib\site-packages\esptool.py", line 2313, in main
    esp.connect(args.before)
  File "c:\python36\lib\site-packages\esptool.py", line 397, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "c:\python36\lib\site-packages\esptool.py", line 377, in _connect_attempt
    self.sync()
  File "c:\python36\lib\site-packages\esptool.py", line 331, in sync
    timeout=SYNC_TIMEOUT)
  File "c:\python36\lib\site-packages\esptool.py", line 272, in command
    self.write(pkt)
  File "c:\python36\lib\site-packages\esptool.py", line 241, in write
    self._port.write(buf)
  File "c:\python36\lib\site-packages\serial\serialwin32.py", line 323, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
```
the "same" command on the same computer but from esp-idf environement is working :
```
$ python ${IDF_PATH}/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/COM7 --baud 921600 read_mac
esptool.py v2.0
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
MAC: 24:0a:c4:09:64:84
Hard resetting...
```

projectgus, 2018-01-10:
Hmm, that's odd! I've not seen a "write timeout" before.

* For your "not IDF" environment, can you please post the output of "python -v"?

* In the IDF environment, can you confirm you're using mingw32.exe (ie the prompt should say MINGW32 in large letters). If you specify the port as `--port COM7` inside IDF as well, does this make any difference?

ftoyon, 2018-01-11:
For esp-idf environment I'm using the mingw32.exe and it's working fine.

The result of python -v under windows command line
```
c:\...>python -v C:\msys32\home\Frederic\esp\esp-idf-v3.0-rc1\components\esptool_py\esptool\esptool.py --chip esp32 --port COM7 --baud 921600 read_mac
import _frozen_importlib # frozen
import _imp # builtin
import sys # builtin
import '_warnings' # <class '_frozen_importlib.BuiltinImporter'>
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
import '_frozen_importlib_external' # <class '_frozen_importlib.FrozenImporter'>
import '_io' # <class '_frozen_importlib.BuiltinImporter'>
import 'marshal' # <class '_frozen_importlib.BuiltinImporter'>
import 'nt' # <class '_frozen_importlib.BuiltinImporter'>
import _thread # previously loaded ('_thread')
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import _weakref # previously loaded ('_weakref')
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
import 'winreg' # <class '_frozen_importlib.BuiltinImporter'>
# installing zipimport hook
import 'zipimport' # <class '_frozen_importlib.BuiltinImporter'>
# installed zipimport hook
# C:\Python36\lib\encodings\__pycache__\__init__.cpython-36.pyc matches C:\Python36\lib\encodings\__init__.py
# code object from 'C:\\Python36\\lib\\encodings\\__pycache__\\__init__.cpython-36.pyc'
# C:\Python36\lib\__pycache__\codecs.cpython-36.pyc matches C:\Python36\lib\codecs.py
# code object from 'C:\\Python36\\lib\\__pycache__\\codecs.cpython-36.pyc'
import '_codecs' # <class '_frozen_importlib.BuiltinImporter'>
import 'codecs' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A4299E8>
# C:\Python36\lib\encodings\__pycache__\aliases.cpython-36.pyc matches C:\Python36\lib\encodings\aliases.py
# code object from 'C:\\Python36\\lib\\encodings\\__pycache__\\aliases.cpython-36.pyc'
import 'encodings.aliases' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5A2630>
import 'encodings' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A41EF28>
# C:\Python36\lib\encodings\__pycache__\utf_8.cpython-36.pyc matches C:\Python36\lib\encodings\utf_8.py
# code object from 'C:\\Python36\\lib\\encodings\\__pycache__\\utf_8.cpython-36.pyc'
import 'encodings.utf_8' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5B36A0>
import '_signal' # <class '_frozen_importlib.BuiltinImporter'>
# C:\Python36\lib\encodings\__pycache__\latin_1.cpython-36.pyc matches C:\Python36\lib\encodings\latin_1.py
# code object from 'C:\\Python36\\lib\\encodings\\__pycache__\\latin_1.cpython-36.pyc'
import 'encodings.latin_1' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5B3B00>
# C:\Python36\lib\__pycache__\io.cpython-36.pyc matches C:\Python36\lib\io.py
# code object from 'C:\\Python36\\lib\\__pycache__\\io.cpython-36.pyc'
# C:\Python36\lib\__pycache__\abc.cpython-36.pyc matches C:\Python36\lib\abc.py
# code object from 'C:\\Python36\\lib\\__pycache__\\abc.cpython-36.pyc'
# C:\Python36\lib\__pycache__\_weakrefset.cpython-36.pyc matches C:\Python36\lib\_weakrefset.py
# code object from 'C:\\Python36\\lib\\__pycache__\\_weakrefset.cpython-36.pyc'
import '_weakrefset' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5BA9E8>
import 'abc' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5BA0B8>
import 'io' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5B3CC0>
# C:\Python36\lib\__pycache__\site.cpython-36.pyc matches C:\Python36\lib\site.py
# code object from 'C:\\Python36\\lib\\__pycache__\\site.cpython-36.pyc'
# C:\Python36\lib\__pycache__\os.cpython-36.pyc matches C:\Python36\lib\os.py
# code object from 'C:\\Python36\\lib\\__pycache__\\os.cpython-36.pyc'
import 'errno' # <class '_frozen_importlib.BuiltinImporter'>
# C:\Python36\lib\__pycache__\stat.cpython-36.pyc matches C:\Python36\lib\stat.py
# code object from 'C:\\Python36\\lib\\__pycache__\\stat.cpython-36.pyc'
import '_stat' # <class '_frozen_importlib.BuiltinImporter'>
import 'stat' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5E1630>
# C:\Python36\lib\__pycache__\ntpath.cpython-36.pyc matches C:\Python36\lib\ntpath.py
# code object from 'C:\\Python36\\lib\\__pycache__\\ntpath.cpython-36.pyc'
# C:\Python36\lib\__pycache__\genericpath.cpython-36.pyc matches C:\Python36\lib\genericpath.py
# code object from 'C:\\Python36\\lib\\__pycache__\\genericpath.cpython-36.pyc'
import 'genericpath' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5ED588>
import 'ntpath' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5E1CF8>
# C:\Python36\lib\__pycache__\_collections_abc.cpython-36.pyc matches C:\Python36\lib\_collections_abc.py
# code object from 'C:\\Python36\\lib\\__pycache__\\_collections_abc.cpython-36.pyc'
import '_collections_abc' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5EDB38>
import 'os' # <_frozen_importlib_external.SourceFileLoader object at 0x000001A69A5D05C0>
# C:\Python36\lib\__pycache__\_sitebuiltins.cpython-36.pyc matches C:\Python36\lib\_sitebuiltins.py
# code object from 'C:\\Python36\\lib\\__pycache__\\_sitebuiltins.cpython-36.pyc'
[... 376 lines cut ...]
# cleanup[3] wiping encodings.latin_1
# cleanup[3] wiping abc
# cleanup[3] wiping os
# destroy abc
# destroy errno
# destroy ntpath
# cleanup[3] wiping stat
# cleanup[3] wiping _stat
# destroy _stat
# cleanup[3] wiping genericpath
# cleanup[3] wiping encodings.cp1252
# cleanup[3] wiping _struct
# cleanup[3] wiping importlib._bootstrap
# cleanup[3] wiping ctypes
# destroy _struct
# destroy ctypes._endian
# destroy ctypes.wintypes
# cleanup[3] wiping _ctypes
# cleanup[3] wiping sys
# cleanup[3] wiping builtins
# destroy ctypes
# destroy os
# destroy stat
# destroy genericpath
```

projectgus, 2018-02-21:
Closing due to a lack of information. Feel free to reopen if you have some more details to help debug this.

closed 2018-02-21 by projectgus