espressif/esptool #262: Add partition support to write_flash command.

OtherCrashOverride, 2018-01-11:
Currently, a fixed offset must be specified for the write_flash command.  An option to specify ESP32 partition name or ID should be added.  This would allow either an offset that does not change or a partition that potentially changes to be used.

projectgus, 2018-01-11:
I agree this will be a nice feature to have.

The plan is to add this to the partition table tool, not esptool.py. esptool.py currently knows nothing about the partition table. This is by design (as partition tables are not part of the ESP32 hardware but part of ESP-IDF).

Some discussion on a similar ESP-IDF feature PR:
https://github.com/espressif/esp-idf/pull/612#issuecomment-302270524

I'm going to close this but feel free to reopen the request on the IDF repo (or Watch the linked PR, as we'll probably deal with both of these shortcomings at the same time.)

closed 2018-01-11 by projectgus

OtherCrashOverride, 2018-01-11:
I am attempting to divorce the flashing of arbitrary data from the build system.  This functionality needs to be available in the absence of any source code or build tools or dependencies thereof.

I do not believe this concept is aligned with the goals of espressif/esp-idf#612.

OtherCrashOverride, 2018-01-11:
A real world example is the case of a SPIFFS filesystem partition.  In the case where only the filesystem is to be updated, the offset where it resides needs to found and may vary across different builds and/or app revisions.  It is not possible to OTA update in all environments where an ESP32 may be deployed.

projectgus, 2018-01-11:
> I am attempting to divorce the flashing of arbitrary data from the build system

The plan is to build a tool that doesn't depend from the ESP-IDF build system (the current gen_esp32part.py partition table tool can be run standalone as well).

On the other hand, the ESP-IDF build system will depend on this tool to run checks as part of the build system.

Does that make sense?

projectgus, 2018-01-11:
> A real world example is the case of a SPIFFS filesystem partition. In the case where only the filesystem is to be updated, the offset where it resides needs to found and may vary across different [...]

We can build a tool for this, but it still makes more sense in a standalone tool that's part of ESP-IDF than having ESP-IDF's software design bleed into this tool.

projectgus, 2018-01-11:
BTW, feel free to open a ticket on the ESP-IDF repo with this request and the use case.