espressif/esptool #264: ESP8089 and ESP8266 manufacturer code

voloviq, 2018-01-18:
Hello,
I bought a few days ago a few chips esp8266 and esp8089. My intention is to use SDIO interface. After solder chip into to my board chip not working. My question is do manufacturer write some firmware into internal flash or chip has empty internal flash memory.
Thanks

projectgus, 2018-01-18:
Hi @voloviq,

ESP8266 and ESP8089 have no internal flash. They can be used with an external SPI flash chip (which is usally pre-flashed by the module/board manufacturer, not Espressif) or they can be used in SDIO Slave mode (ie as an external WiFi interface for Linux, etc.) The chips do have an internal mask ROM which contains some utility functions, etc. However this mask ROM is not progammable or intended for direct user access.

Hope that helps. If you have more general questions about ESP8266 or ESP8089, the best place to post is http://bbs.espressif.com/

Angus

closed 2018-01-18 by projectgus

voloviq, 2018-01-18:
Sorry for question here, I do not known where shall I ask. You gave me exactly such kind of information I looking for. In meantime I was able to run esp8266 on Linux, I hope esp8089 will be work as esp8266 in Linux using the same driver.

Thanks
Best Regards

projectgus, 2018-01-18:
No problems. :)

If you're using ESP8089/ESP8266 from Linux then no flash is involved. The driver pushes a small firmware into the ESP's IRAM, and it calls ROM routines for the rest.

I believe the same kernel driver should work for both.

Best of luck!

Angus

nimaltd, 2018-12-11:
Have you any example sdio 
I want drive it by stm32

voloviq, 2018-12-11:
I run it with succes for am3352. Probably for stm32 should work as well.
What kind of help You need?