espressif/esptool #269: AttributeError: 'module' object has no attribute 'serial_for_url'

alessss, 2018-01-29:
Hi.
Running python esptool.py chip_id causes:
`python esptool.py chip_id
esptool.py v2.3-dev
Traceback (most recent call last):
  File "esptool.py", line 2606, in <module>
    _main()
  File "esptool.py", line 2599, in _main
    main()
  File "esptool.py", line 2320, in main
    esp = ESPLoader.detect_chip(args.port, initial_baud, args.before, args.trace)
  File "esptool.py", line 223, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "esptool.py", line 197, in __init__
    self._slip_reader = slip_reader(self._port, self.trace)
AttributeError: 'ESPLoader' object has no attribute '_port'`

projectgus, 2018-01-30:
Hi,

Can you please give some details about what OS you are using?

Also, the error you posted in the title of the issue and the error in the body of the issue are different, although looks like they're in a related part of esptool.py. Can you please clarify what the full output from esptool.py is?

alessss, 2018-01-30:
Sorry, trace in previous comment appeared after an attempt to fix the issue.
Correct trace is:
`python esptool.py chip_id
esptool.py v2.3-dev
Traceback (most recent call last):
  File "esptool.py", line 2605, in <module>
    _main()
  File "esptool.py", line 2598, in _main
    main()
  File "esptool.py", line 2319, in main
    esp = ESPLoader.detect_chip(args.port, initial_baud, args.before, args.trace)
  File "esptool.py", line 222, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "esptool.py", line 193, in __init__
    self._port = serial.serial_for_url(port)
AttributeError: 'module' object has no attribute 'serial_for_url'`

OS: MacOS sierra 10.12.6
Script was run in Python venv (Python 2.7.10)

pip list
ecdsa (0.13)
esptool (2.2.1)
future (0.16.0)
iso8601 (0.1.12)
pip (9.0.1)
pyaes (1.6.1)
pyserial (3.4)
PyYAML (3.12)
serial (0.0.38)
setuptools (38.4.0)
wheel (0.30.0)

projectgus, 2018-01-30:
Thanks for clarifying with these extra details. It's odd, serial_for_url has been part of pyserial since version 2.5 so 3.4 (latest) should definitely have it.

Interesting that you seem to have esptool 2.2.1 installed via pip, but you're running esptool.py in place from the command line somewhere else and getting v2.3-dev. If you change directories and try to run version 2.2.1 from pip, does this version run correctly?

Finally, can you please double-check pyserial by running this in the exact same command line, virtualenv, & directory that you run the broken esptool.py from:

`/usr/bin/env python -c "import serial; print serial.VERSION; print serial.__dict__.keys()"`

and post the output? Thanks.

projectgus, 2018-01-30:
Maybe this as well:

`/usr/bin/env python -c "import serial; print serial.__file__"`

alessss, 2018-01-30:
>>> print serial.VERSION
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'VERSION'
>>> print serial.__dict__.keys()
['bytearray', 'IndexError', 'all', 'help', 'vars', 'hooks', 'SyntaxError', 'absolute_import', '__path__', 'unicode', 'UnicodeDecodeError', 'memoryview', 'isinstance', 'copyright', 'NameError', 'abs', 'generators', 'input', 'oct', 'bin', 'SystemExit', 'StandardError', 'format', '__file__', 'repr', 'sorted', 'False', 'RuntimeWarning', 'list', 'iter', 'reload', 'Warning', 'round', 'dir', 'cmp', 'set', 'bytes', 'with_statement', 'intern', 'meta', 'issubclass', 'Ellipsis', 'EOFError', 'locals', 'BufferError', 'slice', 'FloatingPointError', 'sum', 'getattr', 'BytesWarning', 'exit', 'print', 'True', 'FutureWarning', 'ImportWarning', 'model', 'None', 'hash', 'any', '__builtins__', 'ReferenceError', 'len', 'credits', 'frozenset', '__name__', 'ord', 'super', 'properties', 'print_function', 'sys', 'TypeError', 'license', 'KeyboardInterrupt', 'UserWarning', 'filter', 'range', 'staticmethod', 'SystemError', 'BaseException', 'pow', 'RuntimeError', 'float', 'MemoryError', 'StopIteration', 'globals', 'divmod', 'enumerate', 'apply', 'LookupError', 'open', 'quit', 'basestring', 'UnicodeError', 'zip', 'hex', 'long', 'next', 'int', 'chr', 'dict', 'type', '__doc__', 'Exception', 'nested_scopes', 'tuple', 'reduce', 'reversed', 'UnicodeTranslateError', 'UnicodeEncodeError', 'IOError', 'hasattr', 'delattr', 'setattr', 'raw_input', 'SyntaxWarning', 'compile', 'ArithmeticError', 'str', 'property', 'GeneratorExit', 'division', 'unicode_literals', 'xrange', 'standard_library', 'KeyError', 'coerce', 'PendingDeprecationWarning', 'file', 'EnvironmentError', 'unichr', 'ascii', 'id', 'OSError', 'DeprecationWarning', 'min', 'UnicodeWarning', 'execfile', '__package__', 'complex', 'bool', 'test', 'ValueError', 'NotImplemented', 'map', 'buffer', 'max', 'object', 'utilities', 'TabError', 'callable', 'ZeroDivisionError', 'errors', 'eval', 'IndentationError', 'AssertionError', 'classmethod', 'request', 'UnboundLocalError', 'ImportError', 'native_str', 'NotImplementedError', 'AttributeError', 'OverflowError']
>>> print serial.__file__
/********/********/.venvs/esp8266/lib/python2.7/site-packages/serial/__init__.pyc - this is the venv path.

projectgus, 2018-01-30:
Thanks for that. I'm not sure what's happened, but there's something very wrong with your pyserial installation - it doesn't seem to have any of the pyserial API exposed in the module. Maybe try reinstalling pyserial via pip and see if the problem goes away?

(BTW, if you use triple-backticks before and after blocks of text like this, they will be `quoted as code` without the weird formatting.)

alessss, 2018-01-30:
Thanks. Recreating venv fixed the trace. I don't have esp module currently at the office, but it seems the problrm is fixed.

shirish47, 2018-03-15:
I am also having this issue. Recently I was learning python I don't know I installed python3 and now when I try to flash my esp32 I get exact same error 

I tired the command above 
```
Shirishs-MacBook-Pro:gatt_client shirishbytelens$ /usr/bin/env python -c "import serial; print serial.VERSION; print serial.__dict__.keys()"
  File "<string>", line 1
    import serial; print serial.VERSION; print serial.__dict__.keys()
                              ^
SyntaxError: invalid syntax

```
and this too
```
Shirishs-MacBook-Pro:gatt_client shirishbytelens$ /usr/bin/env python -c "import serial; print serial.__file__"
  File "<string>", line 1
    import serial; print serial.__file__
                              ^
SyntaxError: invalid syntax
```

projectgus, 2018-03-15:
Hi shirish47,

If you're using Python 3 then please try this:
```
python -c "import serial; print(serial.VERSION); print (serial.__dict__.keys())"
```

shirish47, 2018-03-16:
here is the out put ..

```python -c "import serial; print(serial.VERSION); print(serial.__dict__.keys()); print(serial.__file__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: module 'serial' has no attribute 'VERSION'

projectgus, 2018-03-18:
Hi @shirish47 ,

Thanks for the update. Maybe you have a file called "serial.py" somewhere in your path so Python is importing this is instead of Pyserial.

Can you please try this shorter line as well:
```
python -c "import serial; print(serial.__dict__.keys()); print(serial.__file__)"
```

shirish47, 2018-03-19:
YA I tried it I get this : 
` python -c "import serial; print(serial.__dict__.keys()); print(serial.__file__)"
dict_keys(['__name__', '__doc__', '__package__', '__loader__', '__spec__', '__path__', '__file__', '__cached__', '__builtins__', 'absolute_import', 'division', 'generators', 'nested_scopes', 'print_function', 'unicode_literals', 'with_statement', 'standard_library', 'abs', 'all', 'any', 'ascii', 'bin', 'callable', 'chr', 'compile', 'delattr', 'dir', 'divmod', 'eval', 'exec', 'format', 'getattr', 'globals', 'hasattr', 'hash', 'hex', 'id', 'input', 'isinstance', 'issubclass', 'iter', 'len', 'locals', 'max', 'min', 'next', 'oct', 'ord', 'pow', 'print', 'repr', 'round', 'setattr', 'sorted', 'sum', 'vars', 'None', 'Ellipsis', 'NotImplemented', 'False', 'True', 'bool', 'memoryview', 'bytearray', 'bytes', 'classmethod', 'complex', 'dict', 'enumerate', 'filter', 'float', 'frozenset', 'property', 'int', 'list', 'map', 'object', 'range', 'reversed', 'set', 'slice', 'staticmethod', 'str', 'super', 'tuple', 'type', 'zip', 'BaseException', 'Exception', 'TypeError', 'StopAsyncIteration', 'StopIteration', 'GeneratorExit', 'SystemExit', 'KeyboardInterrupt', 'ImportError', 'ModuleNotFoundError', 'OSError', 'EnvironmentError', 'IOError', 'EOFError', 'RuntimeError', 'RecursionError', 'NotImplementedError', 'NameError', 'UnboundLocalError', 'AttributeError', 'SyntaxError', 'IndentationError', 'TabError', 'LookupError', 'IndexError', 'KeyError', 'ValueError', 'UnicodeError', 'UnicodeEncodeError', 'UnicodeDecodeError', 'UnicodeTranslateError', 'AssertionError', 'ArithmeticError', 'FloatingPointError', 'OverflowError', 'ZeroDivisionError', 'SystemError', 'ReferenceError', 'BufferError', 'MemoryError', 'Warning', 'UserWarning', 'DeprecationWarning', 'PendingDeprecationWarning', 'SyntaxWarning', 'RuntimeWarning', 'FutureWarning', 'ImportWarning', 'UnicodeWarning', 'BytesWarning', 'ResourceWarning', 'ConnectionError', 'BlockingIOError', 'BrokenPipeError', 'ChildProcessError', 'ConnectionAbortedError', 'ConnectionRefusedError', 'ConnectionResetError', 'FileExistsError', 'FileNotFoundError', 'IsADirectoryError', 'NotADirectoryError', 'InterruptedError', 'PermissionError', 'ProcessLookupError', 'TimeoutError', 'open', 'quit', 'exit', 'copyright', 'credits', 'license', 'help', 'native_str', 'errors', 'utilities', 'properties', 'meta', 'model', 'hooks', 'test', 'request'])
/usr/local/lib/python3.6/site-packages/serial/__init__.py`

curusarn, 2018-04-26:
Same problem as @shirish47:
`python -c "import serial; print(serial.__dict__.keys()); print(serial.__file__)"` gives me:  
```
dict_keys(['__name__', '__doc__', '__package__', '__loader__', '__spec__', '__path__', '__file__', '__cached__', '__builtins__', 'absolute_import', 'division', 'generators', 'nested_scopes', 'print_function', 'unicode_literals', 'with_statement', 'standard_library', 'abs', 'all', 'any', 'ascii', 'bin', 'callable', 'chr', 'compile', 'delattr', 'dir', 'divmod', 'eval', 'exec', 'format', 'getattr', 'globals', 'hasattr', 'hash', 'hex', 'id', 'input', 'isinstance', 'issubclass', 'iter', 'len', 'locals', 'max', 'min', 'next', 'oct', 'ord', 'pow', 'print', 'repr', 'round', 'setattr', 'sorted', 'sum', 'vars', 'None', 'Ellipsis', 'NotImplemented', 'False', 'True', 'bool', 'memoryview', 'bytearray', 'bytes', 'classmethod', 'complex', 'dict', 'enumerate', 'filter', 'float', 'frozenset', 'property', 'int', 'list', 'map', 'object', 'range', 'reversed', 'set', 'slice', 'staticmethod', 'str', 'super', 'tuple', 'type', 'zip', 'BaseException', 'Exception', 'TypeError', 'StopAsyncIteration', 'StopIteration', 'GeneratorExit', 'SystemExit', 'KeyboardInterrupt', 'ImportError', 'ModuleNotFoundError', 'OSError', 'EnvironmentError', 'IOError', 'EOFError', 'RuntimeError', 'RecursionError', 'NotImplementedError', 'NameError', 'UnboundLocalError', 'AttributeError', 'SyntaxError', 'IndentationError', 'TabError', 'LookupError', 'IndexError', 'KeyError', 'ValueError', 'UnicodeError', 'UnicodeEncodeError', 'UnicodeDecodeError', 'UnicodeTranslateError', 'AssertionError', 'ArithmeticError', 'FloatingPointError', 'OverflowError', 'ZeroDivisionError', 'SystemError', 'ReferenceError', 'BufferError', 'MemoryError', 'Warning', 'UserWarning', 'DeprecationWarning', 'PendingDeprecationWarning', 'SyntaxWarning', 'RuntimeWarning', 'FutureWarning', 'ImportWarning', 'UnicodeWarning', 'BytesWarning', 'ResourceWarning', 'ConnectionError', 'BlockingIOError', 'BrokenPipeError', 'ChildProcessError', 'ConnectionAbortedError', 'ConnectionRefusedError', 'ConnectionResetError', 'FileExistsError', 'FileNotFoundError', 'IsADirectoryError', 'NotADirectoryError', 'InterruptedError', 'PermissionError', 'ProcessLookupError', 'TimeoutError', 'open', 'quit', 'exit', 'copyright', 'credits', 'license', 'help', 'native_str', 'errors', 'utilities', 'properties', 'meta', 'model', 'hooks', 'test', 'request'])
/usr/lib/python3.6/site-packages/serial/__init__.py
```

projectgus, 2018-04-26:
Hi @curusarn ,

Are you also using MacOS? How did you install esptool, via pip or via some other method?

If either of you run "sudo pip install pyserial" (to re-install pyserial into the system packages), does this help?

curusarn, 2018-04-29:
Hi,

I'm using Linux. I have installed `pyserial` via pip. I'm not using virtualenv. I think `pyserial` was clashing with `serial` package which I have installed earlier (via pip). Uninstalling `serial` and reinstalling `pyserial` fixed my issue. I tried to reproduce it afterwards but I didn't manage to.

Thanks for your response.

projectgus, 2018-04-30:
The underlying issue is here: https://github.com/pyserial/pyserial/issues/319

Seems like 'serial' is a dependency of the popular 'requests' library, which explains how it can be installed unexpectedly.

projectgus, 2018-04-30:
For now, I've added a note so esptool.py prints a better error message if 'serial' is installed. The error notes that these commands may work around the problem:

```
pip uninstall serial
pip install serial
```

The catch is that this may break other Python software which requires 'serial'.

There's no great fix for this right now, apart from [setting up a separate virtualenv for each use](https://virtualenv.pypa.io/en/stable/). I'll try and make contact with the serial & pyserial maintainers, see if there's a better fix that can be implemented.

davebelais, 2018-05-05:
I added a check for the `pyserial` package in the setup script for `serial`—to prevent users from accidentally installing `serial` when intending to upgrade a version of `pyserial`. If there is anyone using `pyserial` with a use case for having both installed—please feel free to raise this topic on bitbucket (https://bitbucket.org/davebelais/serial/issues).

projectgus, 2018-06-13:
I'm going to close this as the specific error has been resolved (since 5988e03). 'serial' and 'pyserial' are still incompatible, but there's nothing esptool can do about this - except for print a more helpful error (which it now does).

closed 2018-06-13 by projectgus

p0we7, 2018-06-14:
I uninstall serial and reinstall pyserial, esptool is not any error now.

gojimmypi, 2018-06-29:
esptool.py used to work well for me. I've not used it in some time, as I've been using C/C++ in Visual Studio with the Visual Micro add-in. Recently I was revisiting MicroPython, which builds in Linux and uploads to the ESP8266 with esptool.py - however I now get the `ImportError: No module named tools.list_ports` error.

I've read a variety of resources online, none of which have helped, including this one from stackoverflow that sounded promising:

https://stackoverflow.com/questions/14108428/pyserial-no-module-named-tools

I tried installing these, but it did not help:

```
sudo apt-get install python-serial
sudo apt-get install pyflakes
sudo apt-get install python-pip
sudo pip install pyserial
sudo apt-get install python3-pyqt5
```

Also tried the pip uninstall:
```
0 $ pip uninstall serial
Cannot uninstall requirement serial, not installed
Storing complete log in /home/pi/.pip/pip.log
```
pyserial is installed:
```
0 $ pip install pyserial
Requirement already satisfied (use --upgrade to upgrade): pyserial in /usr/lib/python2.7/dist-packages
Cleaning up...

```
latest version of  pyserial is installed:

```
0 $ pip install pyserial --upgrade
Cannot fetch index base URL http://pypi.python.org/simple/
Could not find any downloads that satisfy the requirement pyserial in /usr/lib/python2.7/dist-packages
No distributions at all found for pyserial in /usr/lib/python2.7/dist-packages
Storing complete log in /home/pi/.pip/pip.log
```
but the latest esptool.py is failing:

```
0 $ sudo ~/workspace/esptool/esptool.py
Traceback (most recent call last):
  File "/home/pi/workspace/esptool/esptool.py", line 35, in <module>
    import serial.tools.list_ports as list_ports
ImportError: No module named tools.list_ports
```
my esptool.py is dates 6/27/2018 and 125,993 bytes, (`__version__ = "2.4.1"`)

any suggestion what I might be doing wrong? This is on a Raspberry Pi (wheezy) with latest upgrade / updates applied.

gojimmypi, 2018-07-02:
The problem here for me was due to the new requirement of **_pyserial 3_**, and upgrades to it being ignored since an older version was "Owned by OS". Newer versions (in my case at least _Jesse_ for Raspberry Pi) can be forcibly upgraded. Older ones - (in my case _wheezy_ for Raspberry Pi) appear to have no options, not even `apt-get dist-ugrade`.  I was also able to get this working working with Debian 8; no problems there.

See this similar issue where I have more details: https://github.com/espressif/esptool/issues/324#issuecomment-401504758

specifically the _ignore-installed_ option:

```
sudo apt-get update
sudo apt-get upgrade
sudo pip install esptool
sudo pip install --ignore-installed pyserial
sudo esptool.py --port /dev/ttyUSB0 --baud 115200 flash_id
```

See also my project for building MicroPython for the ESP8266, which got me started on this whole path when it stopped working with the latest esptool:

https://github.com/gojimmypi/build_esp8266

amitkray31, 2018-07-26:
Hi, I am having this issue to running python esptool.py. please help.
```
esptool.py v2.5.1-dev
Serial port `/dev/ttyUSB0`
Traceback (most recent call last):
  File "esptool.py", line 2851, in <module>
    _main()
  File "esptool.py", line 2844, in _main
    main()
  File "esptool.py", line 2545, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "esptool.py", line 244, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "esptool.py", line 213, in __init__
    self._port = serial.serial_for_url(port)
  File "/home/akr/.local/lib/python2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/home/akr/.local/lib/python2.7/site-packages/serial/serialposix.py", line 286, in open
    self._update_dtr_state()
  File "/home/akr/.local/lib/python2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 71] Protocol error
```

projectgus, 2018-07-26:
@amitkray31 Please open a New Issue for this, and fill in the details which are asked for on the New Issue page.

answerquest, 2018-12-02:
Just chipping in with what worked at my end: I had to uninstall both serrial and pyserial, then install pyserial again. And because by ubuntu 18.04 system didn't allow python to access the port when on simple user or from inside a virtual environment, I had to install pyserial on the main system. The --user flage helps though.  
`pip3 install --user pyserial`  
I had to run my program with a `sudo` prefix. Wishing there was a way around that.

If the "serial" namespace is contended by two separate packages, then it's all the more case for the module to have its main name equal to the module's name. ie, if it's pyserial, then it's better to just call it keep the program line as `import pyserial` and get on with the show instead of coveting `serial`.

projectgus, 2018-12-03:
@answerquest You should be able to access the port even when pyserial itself is in the "user"/virtualenv install, but your user has to be in the dialout group. See last paragraph here:
https://github.com/espressif/esptool#serial-port

> all the more case for the module to have its main name equal to the module's name

I generally agree, although `pyserial` has been around for a very long time now as `serial` (a good deal longer than the package called "serial"), it may be possible for it to register two package names (I think) and slowly deprecate `import serial`. But this is a matter to discuss with the pyserial maintainers, there's nothing esptool can do about it.

wv1124, 2019-07-07:
Pyserial is not installed for /usr/local/opt/python/bin/python3.7. Check the README for installation instructions.
Traceback (most recent call last):
  File "/bin/esptool.py", line 37, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

projectgus, 2019-07-08:
@wv1124 The error says "Check the README for installation instructions." Did you check the README for installation instructions? https://github.com/espressif/esptool/#installation--dependencies

amitchakor21, 2020-02-14:
Traceback (most recent call last):
  File "/home/amit/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 37, in <module>
    import serial
ImportError: No module named serial
Multiple libraries were found for "WiFi.h"
 Used: /home/amit/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /home/amit/arduino-nightly-linux64/arduino-nightly/libraries/WiFi
exit status 1
Error compiling for board AI Thinker ESP32-CAM.

plz help to solve this, unable to find anything out of readme as stated above
python version 3.7.4
pyserial 3.4 installed
serial 0.0.97 uninstalled

iparfen0xff, 2023-04-18:
Read the instruction: [https://docs.espressif.com/projects/esptool/en/latest/esp32/](url)
1) sudo pip uninstall serial
2) sudo pip install esptool
All works