espressif/esptool #270: Fails to reset ESP8266 after upload; Adafruit Huzzah

jeffsf, 2018-01-31:
With an Adafruit Huzzah board connected through an Adafruit FTDI Friend, v1.2 successfully flashes and restarts the ESP8266. However, current versions do not properly reset the chip at the conclusion of the flash (even though "hard resetting..." is displayed). A "manual" restart of the ESP8266 suggests that the flash was successful.

Mac OS X 10.11.6
python 2.7.10 at /usr/bin/python
Adafruit FTDI Friend (data and power only, no RTS/DTR connection)
Adafruit Huzzah (no additional hardware attached)

python /Volumes/esp-build/esptool/esptool.py -p /dev/tty.usbserial-AH02F0VW --baud 115200 write_flash -fs 16m -fm qio -ff 40m 0x0 ../../bootloader/firmware_prebuilt/rboot.bin 0x1000 ../../bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/blink.bin

```
[jeff@miniup /Volumes/esp-build/esp-open-rtos/examples/blink]$ python /Volumes/esp-build/esptool/esptool.py -p /dev/tty.usbserial-AH02F0VW --baud 115200 write_flash -fs 16m -fm qio -ff 40m 0x0 ../../bootloader/firmware_prebuilt/rboot.bin 0x1000 ../../bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/blink.bin
WARNING: Flash size arguments in megabits like '16m' are deprecated.
Please use the equivalent size '2MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.3-dev
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 3104 bytes to 2169...
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 119.3 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 1001.0 kbit/s)...
Hash of data verified.
Compressed 253344 bytes to 185856...
Wrote 253344 bytes (185856 compressed) at 0x00002000 in 16.5 seconds (effective 122.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```
("blink" does not run until the board is manually reset with the "Reset" switch)

Also fails with 

python /Volumes/esp-build/esptool/esptool.py **_--no-stub_** -p /dev/tty.usbserial-AH02F0VW --baud 115200 write_flash -fs 16m -fm qio -ff 40m 0x0 ../../bootloader/firmware_prebuilt/rboot.bin 0x1000 ../../bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/blink.bin

`git bisect` suggests the issue was introduced around Aug 31 2016

```
commit 9c51b69600b0f667975b7de9885f9ff762d1c477 (refs/bisect/bad)
Author: Angus Gratton <angus@espressif.com>
Date:   Thu Sep 1 16:36:55 2016 +1000

    Add ESP8266 & ESP32 stubs, change stub generation to make a .py snippet not JSON

commit 9cb2aa32c629307435e7b7674e35c48578293801
Author: Angus Gratton <angus@espressif.com>
Date:   Thu Sep 1 15:57:34 2016 +1000

    Allow for very slow chip_erase, some datasheets say max 128s(!!)

commit 1742ae3bebb9881dc68378a0450e9c478a531824
Author: Angus Gratton <angus@espressif.com>
Date:   Thu Sep 1 14:09:55 2016 +1000

    ESP32 stub running, test_esptool.py tests pass on ESP32 & ESP8266

commit f2bc2e437132cb9611951d928e97c384ae91a8e5
Author: Angus Gratton <angus@espressif.com>
Date:   Wed Aug 31 17:44:42 2016 +1000

    Compile stub for ESP8266 and ESP32

commit 52d13b1a524ab2d8d0378f54065b95c796e02fb0
Author: Angus Gratton <angus@espressif.com>
Date:   Wed Aug 31 17:20:22 2016 +1000

    Add erase_region command to erase part of flash

commit 780c9c938be943b4926433d076c8bb8e21c1928b
Author: Angus Gratton <angus@espressif.com>
Date:   Wed Aug 31 17:09:11 2016 +1000

    Add test cases for flash_id, erase_flash

commit efa36980d630575d2013ab6a8a4a994601fa0484
Author: Angus Gratton <angus@espressif.com>
Date:   Wed Aug 31 16:52:00 2016 +1000

    stub flasher: Use SPIBlockErase where possible
    
    This brings performance up to slightly better than V1.1 levels.

commit f72dc4ada62f54f44b226505448d3936b5e3a9e4
Author: Angus Gratton <angus@espressif.com>
Date:   Wed Aug 31 16:30:27 2016 +1000

    Refactor stub loader to speak same serial protocol as ROM bootloaders

commit 08f382b47e4fb123dbb189f9618cf91c171292e7 (refs/bisect/good-08f382b47e4fb123dbb189f9618cf91c171292e7)
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Aug 30 10:48:34 2016 +1000

    gitignore: Add .eggs directory
```

Unable to easily narrow it further as several of the commits over those days appear to be interim development commits and raise exceptions when run.

projectgus, 2018-02-01:
Hi @jeffsf ,

Thanks for the excellent comprehensive bug report. Much appreciated.

In IDF v2.0 we added hard vs soft reset after flashing (as soft reset has some faults of its own, and doesn't work quite right on ESP32). The default option is hard reset, but this requires RTS to be connected.

If you pass `--after soft_reset` before the `write_flash` option then you should get the old behaviour back.

Seeing this bug report I realise that the only documentation for this is buried a bit in the Advanced Options section:
https://github.com/espressif/esptool/wiki/Advanced-Options#reset-modes

Not sure if the solution is to add a mention for this somewhere in the README, or maybe add some kind of default "--after hard_first" which tries a hard reset and falls back on a soft reset as a fallback. Or maybe it'd be enough to say "Hard resetting via DTR" in the log message, so it's clearer what esptool.py is doing here.

jeffsf, 2018-02-01:
Yes, the unpredictable reset behavior of the board I work with has been challenging. A couple of revs of firmware ago I could "trust" that calling `sdk_system_restart_in_nmi()` would reboot the chip to the bootloader, well, at least if it had just been flashed and run. I'm going to have to solder a line to the back of the USB-serial adapter so I can get both DTR and RTS so I don't have to fiddle with the board's switches during development and test. 

Thanks for the explanation and how to move ahead. Knowing that it is related to the lack of DTR reset explains the "Works for me" responses I got. 

For me, the "Hard resetting via DTR" message would have immediately identified the problem (or is it RTS?). I'm not a big fan of too many options, especially when there are enough already to accomplish the task.

projectgus, 2018-02-01:
Thanks @jeffsf . :)

It should be RTS. There's some documentation about it here:
https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection#automatic-bootloader

Will change the message as suggested.

Am toying with the idea of sending a "soft reset" command during the hard_reset sequence while RTS is asserted, anyhow, as in the case of a board with RTS connected the chip will be held in reset so it won't even see the command. Can't think of a downside, but am worried it's too clever...

commit b2027906 (2018-02-01): Add note to "Hard resetting..." message about RTS pin

Closes #270
closed 2018-02-01 by projectgus