espressif/esptool #271: write_flash cannot find binary file : No such file or directory: 'bootloader.bin'

hiphigi, 2018-02-03:
I'm sure this is a silly issue due to my own lack of experience, so I apologize in advance for asking.

I've been getting "esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'bootloader.bin'

* Operating system: Windows 10 64bit
* Python version: (`python -V` to check this)
* ESP hardware in use: NodeMCU ESP32 Dev board 

# Full esptool.py command line as run: 
py tools/esptool.py --chip esp32 --port COM4 --baud 921600 --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions_espruino.bin 0x10000 espruino_esp32.bin

# Full output from esptool.py (please copy and paste all lines of output)
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spit-connection SPI_CONNECTION] [--no-progress] 
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]

# What is the expected behaviour? 
Flash the Espruino binaries to the device.

# Do you have any other information from investigating this?
Seems like the main issue is it cannot find the binary files.

# Is there any other information you can think of which will help us reproduce this problem?
I am running `Python 3.6.2` from `C:\Dev\Python with` `esptool.py` residing in `C:\Dev\Python\tools\esptool.py` - have tried putting the `bin` files in `C:\Dev\Python`, `C:\Dev\Python\tools`, and `C:\Dev\Python\tools\partitions` as well but same error.

closed 2018-02-03 by hiphigi

gbourdon1, 2019-11-14:
Having the same issue

vec123, 2019-12-04:
Hi, me too... i'm also pretty new to this.

projectgus, 2019-12-04:
Hi @gbourdon1 & @vec123 ,

Are you following some instructions from Espruino or somewhere else?

Esptool is looking for a file called bootloader.bin in the current directory and it can't find it.

If you can find this file and change the path to it on the esptool.py command line to match, it should flash.

Or alternatively use `cd \path\to\bin` to change into the `bin` directory with the binaries, and then run `esptool.py` with this command line.

Without knowing what steps you are following or other details it's not possible to help you more from the esptool side, sorry.