espressif/esptool #272: FreeBSD: esptool.py 2.2.1 erase_flash (ESP8266) did not work

ds2k5, 2018-02-04:
FreeBSD 11.1

Python 2.7.14

ESP8266

esptool.py --port /dev/ttyU0 erase_flash

```
esptool.py v2.2.1
Connecting...
....._____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

```

projectgus, 2018-02-05:
Hi,

Can you please give some details about your hardware configuration? (ie what development board or other hardware). The USB ID only gives us the USB/serial chip model.

Do you know if this hardware works on any of the supported OSes?

Can you please post the output of `esptool.py --trace --port /dev/ttyU0 flash_id`?

Thanks,

Angus

ds2k5, 2018-02-06:
Hi Angus,
sorry for the delay

Board: ESP8266, NodeMcu V3

<code>
# esptool.py --trace --port /dev/ttyU0 flash_id
esptool.py v2.2.1
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.107 Timed out waiting for packet header
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.107 Timed out waiting for packet header
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.107 Timed out waiting for packet header
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.107 Timed out waiting for packet header
.TRACE +0.053 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.107 Timed out waiting for packet header
.TRACE +1.863 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.106 Timed out waiting for packet header
_TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.102 Timed out waiting for packet header
_TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
...
....
..... 
</code>

projectgus, 2018-02-09:
It looks like absolutely nothing is being read from the serial port.

My best guess is that this is a problem with the CH340G driver on FreeBSD, or maybe with the [pyserial support for FreeBSD](https://github.com/pyserial/pyserial).

You can test the CH340G driver on FreeBSD by using any serial port program to connect to the ESP8266 and 74880 bps, pressing reset, and looking for any serial output at all. You can test the pyserial support for this by running `miniterm.py /dev/ttyU0 74880`, pressing reset, and looking for any serial output at all.

I don't think there is anything we can do about this from the esptool side, though, if the underlying support isn't sending us any bytes. Sorry!

closed 2018-02-09 by projectgus

ds2k5, 2018-02-09:
Thanks, but i am sure the is installed.
I but will check

ds2k5, 2018-02-09:
tryed to install on a fresh freeBSD insalled system

root@freeBSD11:~ # pip install esptool
Collecting esptool
  Downloading esptool-2.2.1.tar.gz (70kB)
    100% |################################| 71kB 2.2MB/s 
**Collecting pyserial>=2.5 (from esptool)
  Downloading pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |################################| 194kB 2.9MB/s** 
Collecting pyaes (from esptool)
  Downloading pyaes-1.6.1.tar.gz
Collecting ecdsa (from esptool)
  Downloading ecdsa-0.13-py2.py3-none-any.whl (86kB)
    100% |################################| 92kB 5.2MB/s 
Installing collected packages: pyserial, pyaes, ecdsa, esptool
  Running setup.py install for pyaes ... done
  Running setup.py install for esptool ... done
Successfully installed ecdsa-0.13 esptool-2.2.1 pyaes-1.6.1 pyserial-3.4

probonopd, 2021-01-13:
Exact same issue using `py37-esptool-2.5.1_2` from FreeBSD packages on FreeBSD 12.1-RELEASE.

```
% uname -v
FreeBSD 12.1-RELEASE r354233 GENERIC 

% sudo dmesg
(...)
uchcom0: <vendor 0x1a86 USB2.0-Serial, rev 1.10/2.54, addr 9> on usbus0
uchcom0: CH340 detected
```

probonopd, 2021-01-13:
But in my case, something is read (see log):

```
TRACE +0.000 Remaining data in serial buffer: 
    303a30303a303020 4346473a204c6f61 | 0:00:00 CFG: Loa
    6465642066726f6d 20666c6173682061 | ded from flash a
    742046412c20436f 756e742031300d0a | t FA, Count 10..
    30303a30303a3030 204150503a20536f | 00:00:00 APP: So
    6d65207365747469 6e67732068617665 | me settings have
    206265656e207265 7365742028313229 |  been reset (12)
    0d0a30303a30303a 30302050726f6a65 | ..00:00:00 Proje
    6374207461736d6f 7461205461736d6f | ct tasmota Tasmo
    7461205665727369 6f6e20382e312e30 | ta Version 8.1.0
    287461736d6f7461 292d325f365f310d | (tasmota)-2_6_1.
    0a                                | .
```

probonopd, 2023-04-02:
Still an issue on `FreeBSD Users-Laptop 13.1-RELEASE-p3 FreeBSD 13.1-RELEASE-p3 GENERIC amd64`. Cannot communicate with the ESP8266. ESP32 is working fine.

```
esptool.py v4.4
Found 1 serial ports
Serial port /dev/cuaU0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.007 Read 1 bytes: 06
TRACE +0.000 Read invalid data: 06
TRACE +0.000 Remaining data in serial buffer: 784fc563e9
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.023 Read 1 bytes: 0d
TRACE +0.000 Read invalid data: 0d
TRACE +0.000 Remaining data in serial buffer: 
    0a30303a30303a30 302e303031204844 | .00:00:00.001 HD
    573a204553503832 363645580d0a     | W: ESP8266EX..
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 1 bytes: 30
TRACE +0.000 Read invalid data: 30
TRACE +0.000 Remaining data in serial buffer: 
    303a30303a30302e 303530204346473a | 0:00:00.050 CFG:
    204c6f6164656420 66726f6d20666c61 |  Loaded from fla
    7368206174204639 2c20436f756e7420 | sh at F9, Count 
    313332330d0a3030 3a30303a30302e30 | 1323..00:00:00.0
    3534204652433a20 536f6d6520736574 | 54 FRC: Some set
    74696e6773206861 7665206265656e20 | tings have been 
    7265736574202837 290d0a           | reset (7)..
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.020 Read 1 bytes: 78
TRACE +0.000 Read invalid data: 78
TRACE +0.000 Remaining data in serial buffer: 
    fb6fe018030c04c3 0c240c849c0c040c | .o.......$......
    230c67e37c02e46c c70487238cf36f67 | #.g.|..l...#..og
    ef006c8c8f6c2003                  | ..l..l .
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
[... 256 lines cut ...]
TRACE +0.101 No serial data received.
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.107 No serial data received.
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.107 No serial data received.
.
/dev/cuaU0 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
```

probonopd, 2023-04-16:
Is this an issue in the [uchcom](https://man.freebsd.org/cgi/man.cgi?query=uchcom&sektion=&manpath=freebsd-release-ports) FreeBSD driver?

Looking at https://github.com/freebsd/freebsd-src/commits/main/sys/dev/usb/serial/uchcom.c, maybe @hselasky, @avg-I, or @[thompsa know how to debug this further?

Reference:
* https://github.com/helloSystem/ISO/issues/479