espressif/esptool #273: Flashing Sonoff RF bridge MD5 error

tyjtyj, 2018-02-05:
(Please delete any lines which don't apply.)

* Operating system: Win10
* Python version: (`python -V` to check this) 3.5.2.3
* ESP hardware in use:  Sonoff RF bridge 

# Full esptool.py command line as run: as below

# Full output from esptool.py (please copy and paste all lines of output) as below

# What is the expected behaviour?

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?
the MD5 is same each time

Sonoff RF 433 bridge 

>python esptool.py --port COM4 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 495808 bytes to 341889...
Wrote 495808 bytes (341889 compressed) at 0x00000000 in 30.2 seconds (effective 131.4 kbit/s)...
File  md5: 0b3cd34b181cc9a7a30a08a33b051783
Flash md5: dfa42a29a5e2a78fd1edd7c6f8daf672
MD5 of 0xFF is 8e969d31e5c853ad827fd6fa36704dbf

A fatal error occurred: MD5 of file does not match data in flash!

=============================================================
>python esptool.py --port COM4 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 495808 bytes to 341889...
Wrote 495808 bytes (341889 compressed) at 0x00000000 in 30.2 seconds (effective 131.4 kbit/s)...
File  md5: 0b3cd34b181cc9a7a30a08a33b051783
Flash md5: dfa42a29a5e2a78fd1edd7c6f8daf672
MD5 of 0xFF is 8e969d31e5c853ad827fd6fa36704dbf

A fatal error occurred: MD5 of file does not match data in flash!

>python esptool.py --port COM4 flash_id
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Uploading stub...
Running stub...
Stub running...
Manufacturer: 51
Device: 4014
Detected flash size: 1MB
Hard resetting...

What could possible go wrong ?

projectgus, 2018-02-06:
Can you please try

`esptool.py verify_flash --diff yes 0x0 sonoff.bin`

(May produce a lot of output if there are a lot of differences...)

tyjtyj, 2018-02-06:
The device still boot to sonoff original firmware. I am pretty sure it does not write anything

Too much different... already out of screen buffer 

Not sure if I run this correct time 
I run this after i flash. reenter flash mode 

```
   00076ced ff 87
   00076cee ff 14
   00076cef ff 07
   00076cf0 ff 51
   00076cf1 ff 55
   00076cf2 ff fe
   00076cf3 ff c0
   00076cf4 ff 20
   00076cf5 ff 00
   00076cf6 ff 49
   00076cf7 ff 05
   00076cf8 ff 07
   00076cf9 ff 73
   00076cfa ff 09
   00076cfb ff 31
   00076cfc ff 70
   00076cfe ff 0c
   00076cff ff 04
   00076d00 ff c0
   00076d01 ff 20
   00076d02 ff 00
   00076d03 ff 49
   00076d04 ff 03
   00076d05 ff 1b
   00076d06 ff 66
   00076d07 ff 62
   00076d08 ff 47
   00076d09 ff 00
   00076d0a ff 86
   00076d0b ff 13
   00076d0c ff 00
   00076d0d ff 82
   00076d0e ff 45
   00076d0f ff 00
   00076d10 ff 58
   00076d11 ff 03
   00076d12 ff cc
   00076d13 ff 35
   00076d14 ff 52
   00076d15 ff 42
   00076d16 ff 00
   00076d17 ff 0d
   00076d18 ff f0
   00076d19 ff 50
   00076d1a ff 50
   00076d1b ff f4
   00076d1c ff 8c
   00076d1d ff 65
   00076d1e ff 61
   00076d1f ff 48
   00076d20 ff fe
   00076d21 ff c0
   00076d22 ff 20
   00076d23 ff 00
   00076d24 ff 59
   00076d25 ff 06
   00076d26 ff 38
   00076d27 ff 03
   00076d28 ff 07
   00076d29 ff 73
   00076d2a ff 09
   00076d2b ff 31
[... 8914 lines cut ...]
   000790ab 00 40
   000790ac d0 a4
   000790ad 27 6f
   000790ae 00 20
   000790af 4c 40
   000790b0 28 18
   000790b1 32 9d
   000790b2 d0 20
   000790b3 27 40
   000790b4 00 48
   000790b5 4c a2
   000790b6 28 fe
   000790b7 33 3f
   000790b8 0a 18
   000790b9 0a 31
   000790ba 07 ff
   000790bb 01 3f
   000790bc ff 00
   000790bd ff 00
   000790be ff 00
   000790bf 00 e5

A fatal error occurred: Verify failed.

```

tyjtyj, 2018-02-09:
power issue... need to connect to USB vcc instead of 3.3v

closed 2018-02-09 by tyjtyj