espressif/esptool #274: Unable to flash sonoff basic (Timed out waiting for packet header)

lyghtnox, 2018-02-08:
(Please delete any lines which don't apply.)

* Operating system: Windows 7
* Python version: (`python -V` to check this): 2.7.14
* ESP hardware in use: Sonoff basic (esp8266)

# Full esptool.py command line as run: 
esptool.py --port COM7 --baud 115200 write_flash -fs detect -fm dout 0x00000 sonoff.bin

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 467984 bytes to 340377...
Writing at 0x00008000... (14 %)
A fatal error occurred: Timed out waiting for packet header

# What is the expected behaviour?
Sonoff.bin flash succeeded

# Do you have any other information from investigating this?
erase_memory succeeded but I'm unable to write on it

SaimenSays, 2018-02-08:
How is your Sonoff supplied during flashing?
I had some trouble using supply from USB-serial adapter. After using an external 3,3V supply it worked for me.

lyghtnox, 2018-02-08:
I tried to supplied it with an arduino uno but I got the same issue. And I'm unable to connect to the esp if I supply it with a power module.

lyghtnox, 2018-02-08:
I tried with a lower baudrate (38400) and I got the same error but this time at 0x00018000.

projectgus, 2018-02-09:
Are you sure the external power is connected correctly and can supply enough? These kind of failures are often due to power stability problems. We recently had issue #273 on Sonoff which was resolved by connecting power correctly.

> EDIT: Ok I don't know why but I can upload my backup of the original sonoff bin file with :
> esptool.py --port COM7 --baud 115200 write_flash -fs 1MB -fm dio 0x00000 sonoff-backup.bin

Does this work every time?

Try passing the `write_flash --no-compress`, and maybe turning baud rate all the way down to 9600, then flashing your new image. If this works, it's almost certainly a power issue.

lyghtnox, 2018-02-09:
So, flashing my backup works almost every time.

> Try passing the `write_flash --no-compress`, and maybe turning baud rate all the way down to 9600, then flashing your new image.

This doesn't work at all I get the same error (crash at 0x00018000).

Here start the strange part:
I restored my backup and I supplied my sonoff with an arduino uno without entering in flash mode. The led started blinking that means the sonoff is powered on.
But when I enter in flash mode (with the same supply), I can't connect to it and I get:
```
Connecting........_____....._____....._____....._____....._____....._____.....__
___....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting
 for packet header
```
However I still can connect to the sonoff if I supply it with my USB-serial adapter (but I get into the error).

lyghtnox, 2018-02-10:
That was a firmware problem. I tried another firmware and it worked perfectly. Thanks for the answers!

closed 2018-02-10 by lyghtnox

dalulli, 2019-03-04:
what kind of firmware? Python Version? I'm a complete nooby and I always get the timed out thing when having the sonoff with serial adapter connected to my macbook regardless RX/TX orientation

becker666, 2020-08-23:
I'm having the same issue I'm on a Mac(Yosemite) I was able to read_flash but tried different firmwares, baudrates same message try these switches too:

   esptoo.py -p [myport] write_flash -fs 1MB -fm dout 0x0 [sonoff bin version]

heckler, 2020-09-17:
This is a bit anecdotal, since it comes from a sample of one, but it might help others:

I just successfully flashed a Sonoff Basic RF with ESPHome (which uses plaftorm.io and esptool under the covers).  My first attempt failed with the same "Timed out waiting for packet header" mentioned here.

Since I had problems in the past with higher baud rates (ESPHome defaults to 460800 baud), I re-tried lowering the baud rate to 115200 baud, and that did the trick. Upload was successful and after that I disconnected it and did everything else via OTA.

If anyone is still unsuccessful after trying with a lower baud rate, consider the following additional checks:

1. Check if the Tx and Rx are not mixed-up (Tx from the USART goes to Rx on the Sonoff, and vice-versa)
2. Make sure the connections are good (especially if you're holding the connector against the vias, rather than having pin headers soldered)
3. Before starting the upload, the Sonoff must be unpowered; press and hold the devices' button, then apply power; after a couple of seconds, let go of the button and then start the upload (this sequence at power-up puts the Sonoff in flashing mode)

mmccutch, 2021-03-07:
I had the same problem, timed out waiting for packet header. The issue turned out to be simple: with a Sonoff RF2, the enter programming mode procedure of holding down the button while powering on does not work. The solution is documented here https://tasmota.github.io/docs/devices/Sonoff-RF  If you short K_P to GND while powering on, the Sonoff does enter programming mode and then the connection works. Hope that saves someone some frustration.