espressif/esptool #275: esptool using arduino's usb2serial

dynek, 2018-02-16:
* Operating system: Mac OS 10.12.5
* Python version: 2.7.10
* ESP hardware in use: ESP-12

Hello,

Anyone already used Arduino's USB2Serial to communicate with an ESP through esptool?

https://store.arduino.cc/arduino-usb-2-serial-micro

I am able to send AT commands and see the output but I'm unable to connect to the ESP using esptool, it always ends up with "A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header".

Thank you

closed 2018-02-16 by dynek

projectgus, 2018-02-18:
FWIW, you can probably use the Arduino usb/serial if you pass `--before none` to disable the automatic reset behaviour. The Arduino serial firmware doesn't seem to work well with this option.

dynek, 2018-02-19:
I have seen this when googling for this problem, tried it without success :-|

Thanks for your answer.