espressif/esptool #276: Document write_mem option

GTP95, 2018-02-20:
The wiki doesn't mention the write_mem option. You should provide a description that points out the difference (if any) between write_flash and write_mem. Is this one supposed to be used to upload files to the esp filesystem?

projectgus, 2018-02-21:
Thanks for pointing this out, will add documentation for this command.

`write_mem` is for writing into the RAM of the device. It's mostly useful for loading small pieces of code that you want to run directly from IRAM. Not useful for general purposes.

`write_flash` writes to the attached SPI flash, which is what most people want. The ESP doesn't really have a "filesystem" as such, esptool.py just sees the flash as a big range of binary data that it can read & write to. Some development environments may emulate filesystems on top of regions of the flash, but this is one layer higher than esptool's abstraction layer.

projectgus, 2018-05-07:
This is now documented in the "Advanced Commands" section of the wiki:
https://github.com/espressif/esptool/wiki/Advanced-Commands#read_mem--write_mem

The documentation for `write_flash` has also been expanded:
https://github.com/espressif/esptool#write-binary-data-to-flash-write_flash

closed 2018-05-07 by projectgus