espressif/esptool #277: Unable to flash firmware ("Failed to connect") on Wemos D1 with D0/RST soldiered

ShakataGaNai, 2018-02-22:
* Operating system: OSX High Sierra
* Python version: 2.7.10
* ESP hardware in use: Wemos D1 Mini 3.0.0

# Full esptool.py command line as run:
esptool.py --port /dev/cu.wchusbserial1450 --baud 115200 write_flash --flash_size=detect 0 ~/Downloads/esp8266-20171101-v1.9.3.bin

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.2.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

# What is the expected behaviour?
Flashing of firmware

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?
``esptool.py --trace --port /dev/cu.wchusbserial1450 flash_id
esptool.py v2.2.1
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.006 Read 1 bytes: '$'
TRACE +0.000 Read invalid data: '$'
TRACE +0.000 Remaining data in serial buffer: ' UUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.005 Read 1 bytes: '\x08'
TRACE +0.000 Read invalid data: '\x08'
TRACE +0.000 Remaining data in serial buffer: '\x1b[K$ UUUUUUUUUUUUUUUUUUUUUUUUUU'
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.005 Read 1 bytes: '\x08'
TRACE +0.000 Read invalid data: '\x08'
TRACE +0.000 Remaining data in serial buffer: '\x1b[K$ UUUUUUUUUUUUUUUUUUUUUUUUUU'
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.005 Read 1 bytes: '\x08'
TRACE +0.000 Read invalid data: '\x08'
TRACE +0.000 Remaining data in serial buffer: '\x1b[K$ UUUUUUUUUUUUUUUUUUUUUUUUUU'
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.005 Read 1 bytes: '\x08'
TRACE +0.000 Read invalid data: '\x08'
TRACE +0.000 Remaining data in serial buffer: '\x1b[K$ UUUUUUUUUUUUUUUUUUUUUUUUUU'
``

projectgus, 2018-03-13:
Hi @ShakataGaNai ,
 
I've not heard of this WeMos hardware feature. Do you have any more information about it?

The schematic here https://wiki.wemos.cc/products:d1:d1_mini only shows one solder jumper which is RST<->GPIO16 (which is required for deep sleep to work).

If the jump does bridge RST to GPIO0, this will prevent the ESP8266 correctly entering the bootloader mode. This is because to do so RST needs to transition LOW->HIGH while GPIO0 is kept held low. You can read more about this here: https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection

projectgus, 2018-03-13:
Ah, my mistake I forgot that WeMos use their own numbering scheme so their "D0" is ESP8266 GPIO16.

Looking more closely at the trace data you provided, it looks like the device is going into bootloader mode correctly but the first byte of the response is being corrupted.

I'll try to get hold of a WeMos D1 Mini to attempt to reproduce this myself.

ShakataGaNai, 2018-03-13:
It's a long shot, but I appreciate it! The answer for anyone else running across this issue, especially in the near term, is to simply make the Deep Sleep connection something that can be disabled. Either by switch, jumper or using headers and pulling the D1 during programming.

030jmk, 2018-04-02:
I ran into the same issue but fixed it with a slower baud rate. 115200 led to the same fatal error but 57600 ran just fine.

projectgus, 2018-04-09:
Closing as this seems to be resolved.

closed 2018-04-09 by projectgus