espressif/esptool #278: ESP32 Verison 1 - reading and reflashing problems

brendonshaw, 2018-02-26:
I have a custom board and we have been using ESP32 version 0 boards for a few months and now we have switched over version 1 boards. Out of 15 boards I can only read & reflash three boards, keep getting the same error, I have tried everything but cannot solve it.

What has changed between version 0 and version 1, to cause these problems?

esptool.py --port COM17 --baud 115200 read_mac
esptool.py v2.2
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

The command with the trace:

Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.109 Timed out waiting for packet header
.TRACE +0.063 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.109 Timed out waiting for packet header
.TRACE +0.063 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.078 Read 1 bytes: b'e'
TRACE +0.000 Read invalid data: b'e'
TRACE +0.000 Remaining data in serial buffer: b'ts Jun  8 2016 00:22:57\r\n\r\nrst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)\r\nconfigsip: 0, SPIWP:0x00\r\nclk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\nmode:DIO, clock div:2\r\nload:0x3fff0008,len:8\r\nload:0x3fff0010,len:3480\r\nload:0x40078000,len:7804\r\nho 0 tail 12 room 4\r\nload:0x40080000,len:252\r\nentry 0x40080034\r\n\x1b[0;32mI (46) boot: ESP-IDF v2.0-3-gbef9896 2nd stage bootloader\x1b[0m\r\n\x1b[0;32mI (46) boot: compile time 05:59:45\x1b[0m\r\n\x1b[0;32mI (46) boot: Enabling RNG early entropy source...\x1b[0m\r\n\x1b[0;32mI (65) boot: SPI Speed      : 40MHz\x1b[0m\r\n\x1b[0;32mI (78) boot: SPI Mode       : DIO\x1b[0m\r\n\x1b[0;32mI (90) boot: SPI Flash Size : 4MB\x1b[0m\r\n\x1b[0;32mI (102) boot: Partition Table:\x1b[0m\r\n\x1b[0;32mI (114) boot: ## Label            Usage          Type ST Offset   Length\x1b[0m\r\n\x1b[0;32mI (136) boot:  0 phy_init         RF data          01 01 0000f000 00001000\x1b[0m\r\n\x1b[0;32mI (160) boot:  1 otadata          OTA data         01 00 00010000 00002000\x1b[0m\r\n\x1b[0;32mI (183) boot:  2 nvs              WiFi data        01 02 00012000 0000e000\x1b[0m\r\n\x1b[0;32mI (206) boot:  3 at_customize     unknown          40 00 00020000 000e0000\x1b[0m\r\n\x1b[0;32mI (229) boot:  4 ota_0            OTA app          00 10 00100000 00180000\x1b[0m\r\n\x1b[0;32mI (253) boot:  5 ota_1            OTA app          00 11 00280000 00180000\x1b[0m\r\n\x1b[0;32mI (276) boot: End of partition table\x1b[0m\r\n\x1b[0;32mI (289) boot: Disabling RNG early entropy source...\x1b[0m\r\n\x1b[0;32mI (306) boot: Loading app partition at offset 00100000\x1b[0m\r\n'

projectgus, 2018-02-26:
It looks like your board is resetting into a normal boot mode, not the UART bootloader. There were some differences in power on reset between ESP32 r0 and r1.

How are you pulling GPIO0 low on reset? If you are using an automatic reset circuit, try adding some capacitance to the EN pin (so it rises slower when released by the serial control lines).

brendonshaw, 2018-02-27:
We go through a transistor on the GPIO 0 pin, is there any recommendation on the enable line i.e. 100 nF from the enable line to the supply voltage with a pull up resistor in parallel?

projectgus, 2018-02-27:
Yes, something like should help. Recent Espressif dev boards use a 100nF capacitor with a 10K pullup. Not in parallel, but both attached to the pin (capacitor is EN<->GND, Pullup is EN<->VCC). You can find schematics on the Getting Started guides in the ESP-IDF docs: http://esp-idf.readthedocs.io/en/latest/hw-reference/modules-and-boards.html

In the past we've recommended higher value capacitors. There's no harm in having the EN pin rise more slowly, at least in the realm of a few extra milliseconds it won't make any difference.

brendonshaw, 2018-03-08:
Found the problem to be with pin IO2 (24) needed to be unconnected and left floating. I had it connected to a transitor and for some odd reason it was holding the pin in a state. 

I understand this pin controls the flash programming voltage and cannot be tied/connected to any voltage.

The odd thing is I have other boards with the same circuit and they program correctly.

Many thanks for your help and support

closed 2018-03-08 by brendonshaw