espressif/esptool #279: A fatal error occurred: Timed out waiting for packet header from Configuring flash size...

CEStanley58, 2018-03-01:
(Please delete any lines which don't apply.)

* Operating system: Windows 10 x64 32MB DRAM
* Python version: (`python -V` to check this) v34
* ESP hardware in use: WeMOS ESP32 MINI 32 V1.0.0

# Full esptool.py command line as run: Using Arduino, so I'm not sure.

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 512000
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

# What is the expected behavior?
FLASH being programmed.

# Do you have any other information from investigating this?
BLE_Client, sketch uses 931953 bytes (71%)
# Is there any other information you can think of which will help us reproduce this problem?
SOmetimes this works and other times it gets stuck in "A fatal error:..."

CEStanley58, 2018-03-03:
Really, No response?

projectgus, 2018-03-13:
Hi @CEStanley58 ,

Sorry for the slow response, I've been on leave.

Usually these kind of intermittent problems point to hardware signal integrity or other issues.

The fact that it fails after setting 512000bps makes me think there may be a baud rate mismatch betwen the ESP32 and the USB/serial chip on the device. The WeMOS board probably has a CP2104 serial chip, which divides it baud rate from 48MHz. If you try 400000bps (an integer divisor of both 40MHz and 48MHz),  or the default 115200, does the problem go away?

CEStanley58, 2018-03-13:
Hi projectsgus,
  I found on a different thread that lowering the baud from 921600 to anything else might be the ticket. I did change it to 115400 and, while it takes much longer,  it loads everytime.
  Thanks,

CEStanley58, 2018-03-13:
Oops, 115200...

projectgus, 2018-03-13:
Hi @CEStanley58 ,

Good to hear, thanks. You may also want to try some lower multiples of 115200 - 230400, 460800 - and see if these still work reliably. There's a strong chance multiples of 100,000 will work with this particular board as well (maybe up to 900000 or more, depending on the hardware quality.)

closed 2018-03-13 by projectgus

CEStanley58, 2018-03-13:
Hi Angus,

 

  230400 baud works but 512000 did not. So there is a timing issue between PC and ESP32 that is not there with ESP8266. In Arduino, I am only able to select set baud rates. Seems 230400 is the fastest reliable connection on ESP32 and my system.

 

Charles Stanley

congkeodhbk, 2018-10-03:
Pull IO12 LOW

tux-1950, 2020-07-26:
I have had the same problem. I use a USB-TTL converter with two kinds of power (3,3V and 5V selected by a jumper). IThe wrong jumper was selected, 3,3 instead of 5V. 
 After fixing this, all works fine.

rafasimionato, 2020-09-04:
Very same problem here using IDF ... just changed to baud rate to 921600 and it woked fine. As the same as in the Arduino IDE. See IDF logs below:

rafael@r-simionato01:~/esp/hello_world$ idf.py -p /dev/ttyS11 -b 921600 flash
Executing action: flash
Running ninja in directory /home/rafael/esp/hello_world/build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cd /home/rafael/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D ...ld/build" -P /home/rafael/esp/esp-idf/components/esptool_py/run_esptool.cmake
esptool.py --chip esp32 -p /dev/ttyS11 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v3.0-dev
Serial port /dev/ttyS11
Connecting........_____....._____....._____....._____....._____....._____....._
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:62:ab:fe:d2:f0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 9138.2 kbit/s)...
Hash of data verified.
Compressed 25024 bytes to 15349...
Writing at 0x00001000... (100 %)
Wrote 25024 bytes (15349 compressed) at 0x00001000 in 0.2 seconds (effective 1147.8 kbit/s)...
Hash of data verified.
Compressed 152144 bytes to 80787...
Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 152144 bytes (80787 compressed) at 0x00010000 in 1.0 seconds (effective 1191.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
rafael@r-simionato01:~/esp/hello_world$