espressif/esptool #280: A fatal error occurred: Failed to write to target RAM (result was 0107)

emeyedeejay, 2018-03-02:
(Please delete any lines which don't apply.)

* Operating system: Windows 10
* Python version: 3.64
* ESP hardware in use: ESP8266 on Itead SonOff smart WiFi switch (basic)
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi

# Full esptool.py command line as run: python esptool.py --port COM3 read_flash 0x00000 0x100000 image1M.bin

# Full output from esptool.py (please copy and paste all lines of output) D:\Downloads\esptool-2.3.1>python esptool.py --port COM3 read_flash 0x00000 0x100000 image1M.bin
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 0107)

# What is the expected behaviour? Backup of current firmware

# Do you have any other information from investigating this? Tried with 2 different USB -> Serial devices. FTDI232 and PROLIFIC USB-to-serail adapter

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2018-03-13:
Hi @emeyedeejay,

Thanks for reporting this. The device is  suddenly failing while uploading the "stub loader" code into its IRAM. Usually this suggests a power supply problem or a signal integrity problem with the USB/serial connections.

Check all power and signal/ground connections to the device. You could try setting a lower baud rate (for example `-b 9600`) and see if the problem improves at all.

Angus

emeyedeejay, 2018-03-13:
Many thanks for the feedback/input.

I tried another actual FTDI USB->serial cable and was successful so it looks like it was my Chinese "ftdi" and USB-> serial components that are causing the issue here. It; odd though, they have never given me any issues with anything else I have ever used them for (Arduino based activity).

I'll try again with these devices to confirm them as the problem but it looks like you can close this issue.

If there is any additional info you want form me, please let me know...

Best,
Mike

projectgus, 2018-03-13:
Thanks for the update. Feel free to reopen if you think there's something esptool should be doing differently here.

closed 2018-03-13 by projectgus

vgonet, 2018-08-29:
Hello,

I've the same problem than emeyedeejay. I'm trying to program the ESP32 through a STM32. I convert the USB frames to USART (and USART to USB). The information sent by the ESP32 are OK until the Uploading stub. Do you know what is the problem ?

Thank you
Best
Vincent

Flashing binaries to serial port COM14 (app at offset 0x10000 )...
esptool.py v2.5.0
Serial port COM14
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01050000)
make: *** [/home/Vincent/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2

stigok, 2018-10-21:
**FWIW:** If using esptool on a Raspberry Pi using the GPIO serial interface at `/dev/serial0` you have to disable the login shell being available on that device. You can disable this in the interface options in `raspi-config`.

mh4nx7net, 2020-01-30:
done with turn another plug usb

tneish, 2020-04-12:
If you're flashing e.g. a sonoff make sure the "OTA" jumper is on/pin is bridged.  Without it I think you get 0107 because the flash is write protected.

alunomakerdigital, 2020-12-08:
Boa Tarde.
Estava com esse problema e foi RESOLVIDO COM AS ORIENTAÇÕES DESTE TÓPICO. 
TROQUEI O CABO USB ...
PARABÉNS A TODOS.
MUITO GRATO.

azrin1972, 2021-05-23:
Hi,

Im using a LilyGo TTGO T-Call Sim800L v1.4

The upload UART is using the USB C. here is the error I'm getting

[image]

lordofswe, 2021-05-27:
> Im using a LilyGo TTGO T-Call Sim800L v1.4

Same here, same problem

lordofswe, 2021-05-27:
> The upload UART is using the USB C. here is the error I'm getting

I am using another LilyGo product but the same problem.

fhb, 2021-06-06:
This happens to me acactly too, tried a lot of things. Drivers are up to date. It's a ESP32-WROM with 30 Pins and USB/UART CH9102x. 

`/Users/Felix/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool --chip esp32 --port /dev/cu.usbmodem52870049811 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 /Users/Felix/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/partitions/boot_app0.bin 0x1000 /Users/Felix/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/bin/bootloader_qio_40m.bin 0x10000 /var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino_build_58002/sketch_jun06c.ino.bin 0x8000 /var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino_build_58002/sketch_jun06c.ino.partitions.bin 
esptool.py v3.0-dev
Serial port /dev/cu.usbmodem52870049811
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:14:e8:14
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
A fatal error occurred: Failed to write to target RAM (result was 01070000)
`

lmayencourt, 2021-06-10:
I'm facing the same issue with a few ESP32 DEVKITV1 I got from aliexpress:
```
esptool.py v2.6
Serial port /dev/cu.usbmodem52870048231
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 3c:61:05:14:57:34
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
A fatal error occurred: Failed to write to target RAM (result was 01070000)
```

Any hint to resolve this?

azrin1972, 2021-06-11:
I've contacted LilyGo and it seems the FTDI chip they are using is not compatible with Mac OS Big Sur UART Driver. My solution right now is yo use windows under Virtual box and flat from Windows.

Hope this helps

glowingkitty, 2021-06-13:
@azrin1972 cannot even get it to work under a virtual machine with windows 10... Any update on how to make it work on MacOS Big Sur? (M1 Mac)
Else I guess I need to return the ESP32 from LilyGo... Or do you know what the best way is to message them?

don41382, 2021-06-13:
At first I thought the chip is broken. But yes, I am running on a M1 too and get the same error. Did anyone contact LilyGo already?

fhb, 2021-06-13:
For me this error occurs on an older Intel machine running MacOS Big Sur. I didn't try flashing with virtual machine and windows 10 yet, will do in a few days though and report.

don41382, 2021-06-13:
@fhb I used Parallels Software on my M1 running Windows 10.

I could upload the image without any issues.

solocatz, 2021-06-15:
> I've contacted LilyGo and it seems the FTDI chip they are using is not compatible with Mac OS Big Sur UART Driver. My solution right now is yo use windows under Virtual box and flat from Windows.
> 
> Hope this helps

I might got same issue here, Intel CPU with MacOS 11.4 . 
For me only affect Lilygo product, my other low cost esp32 clones are fine. 

esptool.py v3.0-dev
Serial port /dev/cu.usbmodem526F0051341
Connecting........__
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:47:3b:2c
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
A fatal error occurred: Failed to write to target RAM (result was 01070000)

squix78, 2021-08-28:
I had the same issue with a LilyGO device with CH9102 UART chip. The driver in this github resolved the issue for me completely on Mac OS X 11.5.2 Big Sur: https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139 But be warned: you have to install a driver and trust it in the security settings. All driver installation instructions are in Chinese 😱

wangkangman, 2021-09-15:
I think it must be something wrong with the big Sur that every thing running well at windows on same pc with another disk.

BaltarJoaoPaulo, 2022-06-27:
I solved this problem changing the ESP01 USB adapter

esptool.FatalError: Failed to write to target RAM (result was 0107)

the problem was in the USB-SERIAL converter

bytemain, 2022-06-29:
> I had the same issue with a LilyGO device with CH9102 UART chip. The driver in this github resolved the issue for me completely on Mac OS X 11.5.2 Big Sur: [Xinyuan-LilyGO/LilyGo-T-Call-SIM800#139](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139) But be warned: you have to install a driver and trust it in the security settings. All driver installation instructions are in Chinese 😱

download this new driver is worked for me !!!

http://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html

and you need select the new port

[image]

raulclaudino, 2022-07-26:
> 

English version:
http://www.wch-ic.com/downloads/CH34XSER_MAC_ZIP.html

raulclaudino, 2022-07-26:
> download this new driver is worked for me !!!
>
> http://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html
>
> and you need select the new port
>
> [image]

English version:
http://www.wch-ic.com/downloads/CH34XSER_MAC_ZIP.html

kanstantsin-bucha, 2023-05-31:
This issue keeps coming - I have it on the ESP32C3 board for the serial monitor USB-C, I am glad I have the second straight from chip USC-C that does the job. For me, it means I should plug both USB-C when writing firmware. Why do they use that CH9102 UART chip in the first place? I used to buy boards with CP210x and never had a single problem.

gkuhlmann14, 2023-07-13:
> English version: http://www.wch-ic.com/downloads/CH34XSER_MAC_ZIP.html

If you're using platformio I would recommend the following setting in your platformio.ini file: `upload_port = /dev/cu.wchusbserialxxxxxxxxx ` (where xxxxxxxxx is the serial # from the downloaded chinese driver which you can find in either /dev folder or your computers System Information as detailed below). I was experiencing issues with auto detecting upload port for my board and after hard coding to the driver in the ini file no fatal error.

[image]

[image]

Rob2, 2023-08-05:
'A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)'

This problem was solved for me on an old Macbook Air running macOS 10.13.6 connected to an 'M5Stack Tough' with ESP32-D0WDQ6-V3 by updating to the latest driver for the CH9102 USB/UART chip that the 'Tough' uses.

Fully documented driver downloads for various commonly used WCH chips are available here https://wch-ic.com/products/categories/46.html?pid=1#data

mrsndmn, 2024-01-31:
I had a similar problem on Mac with `ESP32-C6-DevKitC-1-N4 ESP32-C6 RISC-V ESP32`

This board has two usb-c ports. And problem was solved by connecting to the second port `ESP32C6`)

christiangda, 2024-03-17:
Same issue for this [ESP32-S3 N16R8](https://es.aliexpress.com/item/1005006028969168.html) 

and after read and install this on my **MacBook Pro M1** [https://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html](https://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html) start working.

Thanks to @raulclaudino for shared the link and info

pauljmartinez, 2024-07-01:
I also had this issue. I'm on an Apple M1 Max, Sonoma 14.5 (23F79) and was getting this error when uploading to my ESP32-C6-DEV-KIT-N8.

```shell
esptool.py v4.6
Serial port /dev/cu.usbmodem57350148971
Connecting....
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:4f:90:b0:ff:fe
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
Failed uploading: uploading error: exit status 2
```

However, I hadn't noticed that there were 2 new `dev/cu.*` entries added when connecting my ESP via the one USB-C port.

```
$  ls -l /dev/cu*
crw-rw-rw- 1 root wheel 9, 5 Jun 17 09:56 /dev/cu.Bluetooth-Incoming-Port
crw-rw-rw- 1 root wheel 9, 3 Jun 30 16:24 /dev/cu.usbmodem57350148971
crw-rw-rw- 1 root wheel 9, 1 Jun 30 17:42 /dev/cu.wchusbserial57350148971
```

Choosing the second entry in Arduino IDE solved my issue.
`/dev/cu.wchusbserial57350148971`

sridhar2rok, 2024-09-24:
> I had a similar problem on Mac with `ESP32-C6-DevKitC-1-N4 ESP32-C6 RISC-V ESP32`
> 
> This board has two usb-c ports. And problem was solved by connecting to the second port `ESP32C6`)

followed the same using second usb-c port solved the issue

bingcheng1998, 2025-03-15:
this works for me: https://github.com/espressif/esptool/issues/908#issuecomment-2726138451