espressif/esptool #284: Possible to monitor device UART output after flashing ?

stefaanv, 2018-03-22:
Is it possible to monitor the device UART output after flashing like with the ESP-IDF command "make flash monitor" ?

projectgus, 2018-03-25:
Hi @stefaanv ,

Not at the moment. However, you can emulate what ESP-IDF does by running miniterm.py immediately after esptool.py. For Linux & OS X this looks something like:

```
esptool.py -p PORT -b 921600 write_flash (flash args) && miniterm.py -p PORT -b 115200
```

For Windows command prompt I think you should use `&` instead of `&&`.

projectgus, 2018-04-09:
Closing this as it's unlikely esptool.py will be extended will serial monitor functionality, rather than using miniterm.py directly.

If there's something which doesn't work with this usage pattern, please let us know and I'll reopen.

closed 2018-04-09 by projectgus