espressif/esptool #291: MD5 of file does not match data in flash!

nhondong, 2018-03-30:
* Operating system: Ubuntu 17.04
* Python version: 2.7.14
* ESP hardware in use:

Kesen KS-601 (3CH) -> Like Sonoff T3
FTDI232 

# Full esptool.py command line as run:
esptool.py -p /dev/ttyUSB0 write_flash -fm dout  0x0 DE-firmware-5.12.0i-mod-1.18.9.1.bin 

# Full output from esptool.py
```
esptool.py v2.3.2-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 484400 bytes to 332821...
Wrote 484400 bytes (332821 compressed) at 0x00000000 in 29.6 seconds (effective 131.1 kbit/s)...
File  md5: 3eff822e6f0c0739c42c5186873d99ed
Flash md5: f42c87786010893519cfe18960034044
MD5 of 0xFF is 66febdacf2859a91a76b932b4810f145
```

A fatal error occurred: MD5 of file does not match data in flash!

Bevor I flashed this one, I flashed 2 other Kesen KS-601 (1CH)

I have read in this forum, that this Issue is related to a power Issue.

Issue #273 says:
power issue... need to connect to USB vcc instead of 3.3v

So, do I really put the 5V output to the 3,3V from the Switch?
I think this will gonna kill the switch.

# Further Infos:
esptool.py -p /dev/ttyUSB0 flash_id

esptool.py v2.3.2-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Uploading stub...
Running stub...
Stub running...
Manufacturer: 51
Device: 4014
Detected flash size: 1MB
Hard resetting via RTS pin...
```

nhondong, 2018-04-01:
Hi,

I decided to test it with 5 Volt on the 3.3V connector.
It works.

Kind regards, Nils

closed 2018-04-01 by nhondong

tcpipchip, 2019-09-10:
crazy

moelfus, 2020-01-30:
Experienced the same. Summed it up here: https://github.com/arendst/Tasmota/issues/2344