espressif/esptool #293: A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

JKTrowling, 2018-03-31:
(Please delete any lines which don't apply.)

* Operating system: Windows 10
* Python version: 2.7.14
* ESP hardware in use:  Adafruit HUZZAH ESP8266 Breakout
# Full esptool.py command line as run: ./esptool.py -p /dev/ttyS4 --trace erase_flash

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.3.2-dev
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.116 Timed out waiting for packet header
.TRACE +0.063 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.137 Timed out waiting for packet header
.TRACE +0.063 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.105 Timed out waiting for packet header
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.126 Timed out waiting for packet header
.TRACE +0.053 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.110 Timed out waiting for packet header
.TRACE +1.831 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.123 Timed out waiting for packet header
_TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.120 Timed out waiting for packet header
_TRACE +0.064 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.115 Timed out waiting for packet header
_TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.116 Timed out waiting for packet header
_TRACE +0.064 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.120 Timed out waiting for packet header
_TRACE +0.223 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.127 Timed out waiting for packet header
.TRACE +0.063 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.106 Timed out waiting for packet header
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.145 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.135 Timed out waiting for packet header
.TRACE +0.064 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.115 Timed out waiting for packet header
.TRACE +1.843 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.107 Timed out waiting for packet header
_TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.117 Timed out waiting for packet header
_TRACE +0.059 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.130 Timed out waiting for packet header
_TRACE +0.065 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.136 Timed out waiting for packet header
_TRACE +0.065 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.130 Timed out waiting for packet header
_TRACE +0.217 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
[... 223 lines cut ...]
TRACE +0.117 Timed out waiting for packet header
.TRACE +1.836 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.115 Timed out waiting for packet header
_TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.104 Timed out waiting for packet header
_TRACE +0.060 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.125 Timed out waiting for packet header
_TRACE +0.059 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.116 Timed out waiting for packet header
_TRACE +0.070 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.120 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

# What is the expected behaviour?
Erase the current firmware

# Is there any other information you can think of which will help us reproduce this problem?
Connected from the esp8226 breakout to my windows 10 machine using https://www.amazon.com/gp/product/B014GZTCC6

projectgus, 2018-04-02:
Hi @JKTrowling ,

Thanks for providing those details. The trace indicates that no data at all was received from the ESP8266. Please check the following:

* Correct serial connections https://github.com/espressif/esptool#serial-connections
* Correct boot mode selection (see https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection) - you'll need to hold GPIO0 while pressing and releasing Reset, and then run esptool.py.
* Try the troubleshooting issues mentioned in the README: https://github.com/espressif/esptool#bootloader-wont-respond

If none of this works, it may be a hardware issue - I suggest talking to Adafruit.

PabloF7, 2018-04-23:
Hi,
I have a similar error,

`esptool.py v2.3.1
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data='\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: '\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.001 Read 1 bytes: '\xc0'
TRACE +0.001 Read 1 bytes: '\x00'
TRACE +0.000 Read 11 bytes: '\x08$\x00\x00\x00\x00\x00\x07\x07\x12 '
TRACE +0.001 Read 1 bytes: 'U'
TRACE +0.000 Read 10 bytes: 'UUUUUUUUUU'
TRACE +0.001 Read 1 bytes: 'U'
TRACE +0.000 Read 10 bytes: 'UUUUUUUUUU'
TRACE +0.001 Read 1 bytes: 'U'
TRACE +0.000 Read 10 bytes: 'UUUUUUUUU\xc0'
TRACE +0.000 Full packet: '\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.100 Timed out waiting for packet header
`
In my case, apparently the bytes are read, but it times out anyway. I tryed all links in previous post without any success.. Is this hardware problem?

projectgus, 2018-04-26:
Hi @PabloF7 ,

It looks like your serial stream may be being corrupted, probably due to some hardware problem. The "full packet" for a sync response normally looks like this: `'\x01\x08\x04\x00\x07\x12 U\x00\x00\x00\x00'` (ESP32) or this `'\x01\x08\x02\x00\x07\x07\x12 \x00\x00'` (ESP8266).

Your board seems to be replying with `''\x08$\x00\x00\x00\x00\x00\x07\x07\x12 '` and then `'\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'` which is similar but not the same...?

coffee-cup, 2018-05-04:
I seem to be having a similar problem. My trace is

```
esptool.py v2.3.1
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.003 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.100 Timed out waiting for packet header
.TRACE +0.057 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.103 Timed out waiting for packet header
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.104 Timed out waiting for packet header
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.000 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.105 Timed out waiting for packet header
```

I can successfully view the boot messages over serial with a baud rate of 74880, but am unable to get `esptool` to connect. I was also able to successfully send AT commands before I bricked it with the wrong baud rate command. Another module I bought at the same time works without issue using the same setup. I am using a 3.3V breadboard power supply and have grounded GPIO0. The Rx and Tx are 5V this doesn't seem to be a problem for the other esp8266 module.

projectgus, 2018-05-09:
Hi @PabloF7 @coffee-cup 

If you run `esptool.py --baud 9600 ...` to sync at a very low baudrate, does it start working? (If it does, this implies a physical signal integrity limitation with your serial connection.)

Angus

coffee-cup, 2018-05-09:
Unfortunately not :( It does not work at any baud rate

theproxy, 2018-06-29:
Are you on a Ryzen chipset? I was going to submit a same issue but I am trying same with esptool 2.4.1 on a Dell Inspiron 13 7000 2-in-1. Windows 10 reports Processor is "AMD Ryzen 5 2500U with Radeon Vega Mobile Gfx".

I tried running esptool --chip esp32 --port COM3 --baud 9600 read_mac and get same results. 

I am successful talking to the device using TeraTerm so I believe it's not a serial terminal issue.

theproxy, 2018-06-29:
My data: 
TRACE +0.110 Timed out waiting for packet header
.TRACE +0.078 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.063 Read 1 bytes: 5b
TRACE +0.000 Read invalid data: 5b
TRACE +0.000 Remaining data in serial buffer:
    494e464f5d205265 6164696e6720616c | INFO] Reading al
    61726d2066696c65 0d0a             | arm file..
.TRACE +0.109 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU

projectgus, 2018-07-02:
Hi @theproxy ,

You have some different symptoms here to @JKTrowling , because it looks like some data is being read from your ESP8266 - it's just not what esptool is expecting to see.

Usually this means the chip is not resetting into bootloader mode correctly. See here: https://github.com/espressif/esptool/#entering-the-bootloader

Angus

alesrebec, 2018-07-12:
I'm having the same issue with ESP8266-01.
If I power on the ESP8266 the blue light is solid ON. In the monitor I get constant garbage.
All connections are fine because I'm reading and flashing other ESPs !! 
I may only bricked this one after a wrong connection or some corruption in code.

If I boot it up (GPIO0 LOW + RESET - like I do with other where it works) and try to flash it I get the same error.
It cannot talk with ESP.

`A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header`

Is it dead or can I somehow unbrick it?

backpacked, 2018-09-14:
> My data:
> TRACE +0.110 Timed out waiting for packet header
> .TRACE +0.078 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
> 0707122055555555 5555555555555555 | ... UUUUUUUUUUUU [...]

I see similar output on my Sonoff Basic 2017-10-11. Have not been able to flash it yet. @theproxy, did you have any success? If so, what did you do?

RockfordWei, 2018-10-01:
I have a similar issue but I am not sure if the problem is the command line parameters.

My win7 VM works well for flashing as the screenshot.
[image]

However, my mac with `sudo pip install esptool` doesn't work at all:

```
$ esptool.py --port /dev/tty.usbmodem1411 --baud 115200 --trace \
	write_flash \
	--flash_freq=40m \
	--flash_mode=dio \
	--flash_size=1MB  \
	0x00000 boot_v1.6.bin \
	0x01000 build/qbiq_esp_app.user1.1024.0x01000.bin \
	0x81000 build/qbiq_esp_app.user2.1024.0x81000.bin \
	0xFC000 esp_init_data_default.bin \
	0x7E000 blank.bin
```

will cause this:
```
Uploading stub...
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 2853, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 2846, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2571, in main
    esp = esp.run_stub()
  File "/usr/local/bin/esptool.py", line 579, in run_stub
    self.mem_block(stub[field][from_offs:to_offs], seq)
  File "/usr/local/bin/esptool.py", line 492, in mem_block
    self.checksum(data))
  File "/usr/local/bin/esptool.py", line 340, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/usr/local/bin/esptool.py", line 308, in command
    self.write(pkt)
  File "/usr/local/bin/esptool.py", line 271, in write
    self._port.write(buf)
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 552, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
```

The full trace looks like this:
```
esptool.py v2.5.1
Serial port /dev/tty.usbmodem1411
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.106 Timed out waiting for packet header
.TRACE +0.058 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.027 Read 1 bytes: c0
TRACE +0.000 Read 95 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
[... 860 lines cut ...]
    e5dbdce08293ed08 4603000c0c060100 | ........F.......
    000000cd05dd0c7c de313dfa62212b28 | .......|.1=.b!+(
    23d066dbdcd022db dcdaffd11cfa2923 | #.f...".......)#
    380d711cfacac370 3cdbdc0cc0       | 8.q....p<....
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 2853, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 2846, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2571, in main
    esp = esp.run_stub()
  File "/usr/local/bin/esptool.py", line 579, in run_stub
    self.mem_block(stub[field][from_offs:to_offs], seq)
  File "/usr/local/bin/esptool.py", line 492, in mem_block
    self.checksum(data))
  File "/usr/local/bin/esptool.py", line 340, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/usr/local/bin/esptool.py", line 308, in command
    self.write(pkt)
  File "/usr/local/bin/esptool.py", line 271, in write
    self._port.write(buf)
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 552, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
```

projectgus, 2018-10-25:
Hi @RockfordWei ,

This looks like a different issue again, maybe something with the USB to Serial driver on your Mac. WriteTimeoutError means that the adapter hasn't allowed serial data to be written to it for longer than the timeout period (10 seconds). This basically never happens under normal conditions, the problem is either the OS-level driver is buggy or there's some problem with the USB connection to the device.

Angus

alesrebec, 2018-11-27:
The error says it failed to connect to Espressif device: Invalid head of packet.

Two things look weird in your esptool.py call:
esptool.py --port /dev/**tty.usbserial-00000000** --baud **74880** --trace read_mac

Check if the device name is valid and baud looks wrong.

projectgus, 2018-11-30:
@ice2hu Possibly also the device is not in the correct boot mode. Check the wiki at https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection about entering bootloader mode.

I'm going to close this issue because the original reporter never provided any follow-up details, and the other traces posted here do not seem to be the same.

*Everyone: If you're having trouble connecting to your ESP8266 or ESP32, please [check the troubleshooting list here](https://github.com/espressif/esptool/#bootloader-wont-respond). If it still doesn't work, and you can't find a 100% exact match for your situation in the open issues, then please open a new issue and provide all of the details asked for in the issue template. Thanks!*

closed 2018-11-30 by projectgus

gbourdon1, 2019-11-17:
I'm also having a similar issue, except I don't receive any TRACE messages. It simply says 

C:\Users\gbourdon1>esptool.py --port COM9 --before no_reset --baud 115200 write_flash 0 NINA_W102-1.5.0.bin
esptool.py v2.8
Serial port COM9
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x08)

I tried lowering the baud rate, which lead me to this error: A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

Any thoughts on how to debug?

onpagecorp, 2020-02-02:
try the following:
1. disconnect from USB
2. press and hold RESET button
3. connect USB
4. release RESET button now
5. try to burn now

Bernd58, 2020-05-15:
Hi All! This is my first comment here because I started with ESP8266 4 weeks ago. Today I used first a FTDI adapter with CH340G chip. While trying programming ESP-01 with Tasmota via esptool.py I got the same error:
**"Failed to connect to Espressif device: Timed out waiting for packet header"**.
Solution to fix it: It is absolutely necessary to connect the "chip_enable" port to high level (3.3V).
Connect also RESET and GPIO0 to GND with 10k resistor each. Then it worked!!!

surrounds, 2020-08-13:
I had same issue, algo that helped for me

1. disconnected from usb (or just power if you use external power)
2. press and hold RESET Button
3. connect usb (power)
4. release RESET button
5. wait 2 sec
6. Press and hold RESET button for 2 sec

Also you can try to use external power, I used Arduino 3.3V pin