espressif/esptool #297: How to download the factory image (.bin) file from esp32?

anupamb062, 2018-04-06:
* ESP hardware in use:esp32

projectgus, 2018-04-09:
Hi @anupamb062 ,

You can use the read_flash command to read the flash contents, ie
```
esptool.py -p PORT -b 921600 read_flash 0 0x400000 flash_dump.bin
```

Replace 0x400000 above with the size of your flash, if different to 4MB.

This will read the entire contents of the flash. To read just a single partition (like the factory image), you'll need to determine the partition offset and size and pass these to read_flash instead. Usually this information is printed to the serial console by the bootloader.

closed 2018-04-09 by projectgus

anupamb062, 2018-04-09:
Thanks projectgus. Can you suggest some materials to gain knowledge about the flash partitions and offsets?

projectgus, 2018-04-10:
If the device is running something based on ESP-IDF (AFAIK all ESP32 software environments are based on ESP-IDF in some way), it will have an ESP-IDF partition table on it:
https://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html

esptool.py doesn't know about partitioning (it just sees the flash as a big array of bytes), but there is a tool in ESP-IDF that you can use to convert the binary partition data to a text-based summary (documented at the above link).

anupamb062, 2018-04-23:
Hi @projectgus ,
                           What if I forget to give 0 in the command line to download the bin in the flash? Will it have any effect on the bin file? The command I might have used : esptool.py -p PORT -b 115200 read_flash 0x400000 flash_dump.bin. As you can see I might have missed the zero after the read_flash function call. Appreciate any sort of help.

projectgus, 2018-04-23:
@anupamb062 If you miss one of these arguments then esptool.py will error out. Here it is on my computer:

```
$ esptool.py -p /dev/ttyUSB0 -b 115200 read_flash 0x400000 flash_dump.bin
usage: esptool read_flash [-h] [--spi-connection SPI_CONNECTION]
                          [--no-progress]
                          address size filename
esptool read_flash: error: argument size: invalid arg_auto_int value: 'flash_dump.bin'
```

chegewara, 2019-07-19:
Hi @projectgus 
i have maixduino board, which is RISC V with esp32 wroom. I can monitor esp32 with `make monitor ESPPORT=com22` but i can dump flash with this command:
`esptool.py -p ESPPORT=com22 -b 115200 read_flash 0 0x400000 flash_dump.bin`

i am getting error that msys2 can't open port:
```
esptool.py v2.7-dev
Serial port ESPPORT=com22
Traceback (most recent call last):
  File "F:/esp32_master/home/darek/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 3065, in <module>
    _main()
  File "F:/esp32_master/home/darek/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 3058, in _main
    main()
  File "F:/esp32_master/home/darek/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2750, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "F:/esp32_master/home/darek/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 261, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "F:/esp32_master/home/darek/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 225, in __init__
    self._port = serial.serial_for_url(port)
  File "F:/esp32_master/mingw32/lib/python2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "F:/esp32_master/mingw32/lib/python2.7/site-packages/serial/serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'ESPPORT=com22': WindowsError(2, 'Nie mo\xbfna odnale\x9f\xe6 okre\x9clonego pliku.')
```
I am using msys2 on windows 10 with esp-idf master.

projectgus, 2019-07-22:
@chegewara You don't need to pass `ESPPORT=` when using esptool.py directly, try `esptool.py -p COM22 ...`