espressif/esptool #300: erase_flash command timing out after 120 seconds

projectgus, 2018-04-11:
As reported by @loadedtech and @dehart, here:
https://github.com/espressif/esptool/pull/247#issuecomment-355700683
https://github.com/espressif/esptool/pull/247#issuecomment-380618886

Both cases the erase_flash operation times out with
```
Erasing flash (this may take a while)...
TRACE +0.000 command op=0xd0 data len=0 wait_response=1 timeout=120.000 data=''
TRACE +0.000 Write 10 bytes: '\xc0\x00\xd0\x00\x00\x00\x00\x00\x00\xc0'
TRACE +0.024 Read 1 bytes: '\xc0'
TRACE +0.000 Read 8 bytes: '\x01\xd0\x00\x00\x00\x00\x00\x00'
TRACE +120.007 Timed out waiting for packet content
```

projectgus, 2018-04-11:
Hi folks,

I think the issue here may be that the flash is taking more than 120 seconds to erase. Some flash chips can take a while, we've had reports of a few models of "ultra low power" flash where erase & write operations are very slow.

The erase is probably completing OK, it's just that esptool.py has timed out waiting for it to happen.

* If you re-run the same command on the same chip, does it pass then? (Usually the length of time needed for a chip erase is proportional to the number of unerased blocks in the chip).
* What hardware (dev board, module, etc) do you each have?
* Can you please post the output of `esptool.py flash_id`? If the flash chip is exposed on the board (not under an RF shielding can) then can you please read the part number off the top of it?

Thanks,

Angus

dehart, 2018-04-13:
- The same command has been executed multiple times the outcome is the same.
- Lolin v3 NodeMcu. ESP8266
- flash_id output:
```
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Manufacturer: ff
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...
```

projectgus, 2018-04-16:
Hi @dehart,

Thanks for the update. Reporting 0xFF for manufacturer & device id is an indication that the ESP8266 can't talk to the attached flash chip at all, it's sending commands and getting garbage back. Sometimes this can be because of power issues or because something is connected to the ESP8266 pins which are also wired to the flash chip (GPIOs 6-11).

If your development board is inserted in a breadboard or similar then you can remove it and see if anything changes.

Otherwise, the hardware is probably faulty (flash chip is defective or not soldered correctly). Replacing is the best option if you don't feel like rework soldering it yourself.

Angus

dehart, 2018-04-16:
I should add that `esptool.py read_flash` does compleet.
And `esptool.py flash_id` sometimes outputs
```
Manufacturer: 00
Device: 0000
```
After power on and off a few times pressing the flash and reset buttons,
I am going to try soldering but I've ordered another board because this usualy doesn't end well for me 🤕

projectgus, 2018-04-16:
Hi @dehart ,

Out of curiosity, when you read_flash do you get all 0xFF or 0x00 in the output, or is it correct data?

The top bridge in your photo looks like the WP and CS pins to the SPI flash chip, so this would explain flash interaction problems.

(The bottom bridge in the photo is two 3.3V power pins so that's likely harmless.)

If you're going to scrap this anyway and you have some liquid flux and a soldering iron then you may be able to unbridge by applying some liquid flux to the bridge and then popping the iron on it (tinned but with no liquid solder on the tip) for 5 or so seconds. Those little bridges will sometimes pop right off with a bit of flux!

dehart, 2018-04-17:
read_flash got nothing but 0x00
I've ordered some liquid flux because I wanted to find out. 
If I can unbridge it I'll try to flash again. Fingers crossed 🤐

projectgus, 2018-04-17:
Good luck @dehart. :)

@loadedtech do you think the issue you've reported may have the same root cause?

dehart, 2018-04-18:
Worked like a charm, Thanks again @projectgus 

[image]

```
python esptool.py -p /dev/cu.wchusbserial1420 -b 115200 flash_id
esptool.py v2.3.2-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Manufacturer: c2
Device: 2516
Detected flash size: 4MB
Hard resetting via RTS pin...
```

closed 2018-06-13 by projectgus