espressif/esptool #301: esptool does not work when minicom is opened

vinifr, 2018-04-14:
* Operating system: Ubuntu 16
* Python version: 2.7.6
* ESP hardware in use: ESP32-DevKitC

# Full esptool.py command line as run:
sudo python tools/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x10000 pppos_client.bin 0x8000 partitions_singleapp.bin

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.3.1
Connecting....
Traceback (most recent call last):
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 2637, in <module>
    _main()
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 2630, in _main
    main()
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 2356, in main
    esp.connect(args.before)
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 407, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 387, in _connect_attempt
    self.sync()
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 341, in sync
    timeout=SYNC_TIMEOUT)
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 294, in command
    p = self.read()
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 239, in read
    return next(self._slip_reader)
  File "tools/esp-idf/components/esptool_py/esptool/esptool.py", line 1651, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 453, in read
    buf = os.read(self.fd, size-len(read))
OSError: [Errno 11] Resource temporarily unavailable

Minicom is open in /dev/ttyUSB0 as well for Debug.

projectgus, 2018-04-16:
Hi @vinifr ,

Unfortunately this is is an operating system level restriction. Linux doesn't reliably allow multiple programs to access the same serial port - in this case it's returning a "Resource temporarily unavailable" error when esptool tries to read from it.

There isn't anything we can do about this, except for request exclusive access to the serial port (which would mean esptool errors out earlier if minicom is open, but still wouldn't allow them both to use it at once). The configuration we have now does allow people to (sometimes) share the serial port successfully under some (poorly understood or defined) conditions, though.

> Minicom is open in /dev/ttyUSB0 as well for Debug.

Are you looking to debug the program after it's flashed? You can write a command line like this for that:

```
esptool.py -p /dev/ttyUSB0 write_flash (...) && minicom -D /dev/ttyUSB0 (...)
```

Which will flash and then run minicom immediately if flashing was successful.

I'm a fan of "miniterm.py", which is bundled with the pyserial library that esptool uses. Miniterm doesn't open it's own window when it runs:
```
esptool.py -p /dev/ttyUSB0 write_flash blah && miniterm.py /dev/ttyUSB0 115200
```

(To exit miniterm, type Ctrl-].)

closed 2018-04-16 by projectgus

vinifr, 2018-04-16:
Well, I can flash and keep console programs opened at the same time on stm32 and tm4c1294x microcontrollers. But I guess this has something to do with hardware capacity?

projectgus, 2018-04-16:
What programs are you using to flash those two micros over serial?

vinifr, 2018-04-17:
st-flash for STM32 [https://github.com/texane/stlink](st-flash) and lm4flash [https://github.com/utzig/lm4tools/tree/master/lm4flash](lm4flsh) for TM4C129X.

vinifr, 2018-04-17:
By the way, both the micros, STM32 and TM4C19X, create a USB ACM device instead ttyUSB0.
_cdc_acm 2-1.6:1.2: ttyACM0: USB ACM device_

projectgus, 2018-04-17:
Thanks for the update. It turns out neither of these configurations are flashing via the serial port itself. The programmer software is talking to a different programmer/debugger interface on the chip (STLINK for STM32 and ICDI for the TI chip) via a lower-level USB interface. The USB devices on the boards are "Composite", so they have the serial interface and the programmer interface in the same device.

On the other hand, esptool.py uses the serial port and nothing else to communicate.

It is possible to have a separate programmer for ESP32 as well, we have this on the ESP-WROVER-KIT and there are some third-party boards which also integrate JTAG adapters. However unfortunately we don't yet have support for flashing ESP32 via JTAG, only debugging. The flashing support is a work in progress.