espressif/esptool #302: Custom flash tool flashing data and not verifying

nkolban, 2018-04-21:
I am working on a custom flash tool in JavaScript using Node.js.  My tests are progressing well and I thought I had it all working.  I can write into arbitrary flash and I had tested it by reading back and validating that the checksums on what I wrote was the same as what I read back.   However, the esptool verify_flash is reporting a problem.

I have a file called bootloader.bin.  This wants to be flashed to address 0x1000 and has a length of 18032 bytes.  I run my Node.js app and it claims it succeeds.  At this point I have bootloader.bin on my filesystem and the belief that what is in flash is now the same.

I now use

```
esptool read_flash 0x1000 18032 temp.bin
```

I now have two files.  My original bootloader.bin and temp.bin (as read back from flash).  When I take a checksum using `md5sum` against both files they report identical and using `cmp -l` no bytes differ in comparison.  They are the same file.   This raises my confidence that what is in flash is what what I expected.

Now comes the mystery ... when I run verify_flash using:

```
esptool verify_flash 0x1000 bootloader.bin 
```
we get the following response:

```
esptool.py v2.3.1
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Verifying 0x4670 (18032) bytes @ 0x00001000 in flash against bootloader.bin...
-- verify FAILED (digest mismatch)

A fatal error occurred: Verify failed.
```

... and I am at a loss to explain.

nkolban, 2018-04-22:
Closing after reading and understanding ... https://github.com/espressif/esptool#verifying-flash

closed 2018-04-22 by nkolban

zaphod-42, 2019-01-22:
Did you manage to get a nodejs version of esptool working without the python dependency? I was hoping to implement the same thing and came across this thread while looking for existing options... thanks!

alexandruradovici, 2020-09-22:
Same question as @zaphod-42 here about the node implementation.

kolban-google, 2020-09-22:
I got it running ... but it was work for hire for a 3rd party who owns the rights and source.  This means that there is little I can say about it.  In the end, I did find that study of the esptools and the specification of the data structure as documented by Espressif was sufficient in the end.