espressif/esptool #304: ESP32 erase_flash timeout after 120 seconds

jgehlsen, 2018-05-04:
I am having an issues similar to that discussed in https://github.com/espressif/esptool/issues/300 but this is on a custom board using an ESP32.

* Operating system: Windows 10
* Python version: 3.5.0
* ESP hardware in use: ESP32-D0WDQ6

# Full esptool.py command line as run: python esptool.py --port COM4 erase_flash

# Full output:
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision (unknown 0xa))
Features: WiFi, BT, Dual Core, VRef calibration in efuse
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content

The error occurs 120 seconds after I executed the command and it is consistent.

Here is the trace output (without the trace records before the "Stub running..." message:

Stub running...
Erasing flash (this may take a while)...
TRACE +0.000 command op=0xd0 data len=0 wait_response=1 timeout=120.000 data=b''
TRACE +0.000 Write 10 bytes: b'\xc0\x00\xd0\x00\x00\x00\x00\x00\x00\xc0'
TRACE +0.004 Read 1 bytes: b'\xc0'
TRACE +0.000 Read 8 bytes: b'\x01\xd0\x00\x00\x00\x00\x00\x00'
TRACE +120.002 Timed out waiting for packet content

A fatal error occurred: Timed out waiting for packet content

Here is the output from the flash_id command:
 
esptool.py v2.3.1
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision (unknown 0xa))
Features: WiFi, BT, Dual Core, VRef calibration in efuse
Uploading stub...
Running stub...
Stub running...
Manufacturer: ff
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...

All other commands work.  I can get the chip_id, read_mac, etc.  Any command that has anything to do with the flash memory fails.

I have taken the cover off of the ESP32 and I see nothing out of the ordinary.  All of the connections look fine.

This is a custom board that uses several of the pins for input signals and several of the output pins for output signals.

Inputs
GPIO14: I2C data
GPIO32: I2C clock
GPIO8: input signal
GPIO16: input signal
GPIO36: input signal

Outputs
GPIO4-11: output signal 
GPIO13-15: output signal
GPIO17: PWM output

From what I have read, if the flash_id command shows a manufacturer/device of ff/ffff, the chip cannot communicate with the flash memory.  Does this have anything to do with the GPIOs we are using or do I just have a bad board?

Any ideas, suggestions, thoughts?

projectgus, 2018-05-07:
Hi @jgehlsen ,

>  Does this have anything to do with the GPIOs we are using or do I just have a bad board?

Yes. GPIOs 6-11 are all used for SPI flash connections. If your board is using a WROOM module then these connections are made internally to the on-module SPI flash. You can check the ESP32 Hardware Design Guidelines or ESP-WROOM-32 reference design for the details. ([Both can be found here](https://www.espressif.com/en/support/download/documents)).

If nothing is connected to these pins but they are broken out on the board, check for shorts where the module or chip is soldered to the board.

jgehlsen, 2018-05-08:
Thanks @projectgus.

There is circuitry connected to these pins so we are going to have to go back and fix some things.

Thanks for getting back to me so quickly.

closed 2018-06-13 by projectgus