espressif/esptool #305: ESP32 Timed out waiting for packet content make: ***

max3l, 2018-05-08:
**Operating system:** Windows 10 Pro Version 1709
**Python version:** v3.6.3:2c5fed8
**ESP hardware in use:** [ESP-Wroom-32](https://de.aliexpress.com/item/ESP32-ESP32S-WIFI-Bluetooth-Module-240MHz-Dual-Core-CPU-MCU-Wireless-Network-Board-ESP-WROOM-32/32817688406.html?spm=a2g0s.9042311.0.0.JrQSeF)

## Full esptool.py command line as run:
`$make flash`

## Full output from esptool.py:

```
WARNING: Toolchain version is not supported: 1.22.0-73-ge28a011
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: 1.22.0-73-ge28a011
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed atyour own risk.
Flashing binaries to serial port COM8 (app at offset 0x10000)...
esptool.py v2.3.1
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 20096 bytes to 11826...

A fatal error occurred: Timed out waiting for packet content
make: *** [/esp/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Fehler 2
```

## What is the expected behaviour?

It should Flash the ESP32-IDF on the Board.

## Do you have any other information from investigating this?

I ran the espfuse.py with following results:

**command:** `python.exe espefuse.py --port COM8 summary`
**output:**
```
espefuse.py v2.3.1
Connecting.......
Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1100 R/W (0x0)

Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
[... 165 lines cut ...]
          }
          if (currentLine.endsWith("GET /gruen")) {
            Serial.println("Alle LEDs werden gruen");
          }
        }
      }
      client.stop();
      Serial.println("Client Disconnected.");
    }
  }

}

void loop() {

}
```

After the error occured first I succesfull flashed it again using make. But the Arduino IDE still refused to write my program onto the ESP32. So I tried to flash it again and was careful the Secure Boot mode was disabled but I still got the error message. And this is the point where I am now. The ESP32 is soldered onto a proper [adapter PCB](https://de.aliexpress.com/item/DIY-Kits-ESP-32S-Adapter-Board-Pinboard-Convertor-For-ESP32S-Wireless-WIFI-Module/32805395595.html?spm=a2g0s.9042311.0.0.HSbFAb) and is powered by a 5V 3A power supply which is stepped down to 3.3V using a LD33V voltage regulator and a [FT232RL USB to TTL Serial](https://www.amazon.de/gp/product/B01N9RZK6I/ref=oh_aui_search_detailpage?ie=UTF8&psc=1) modul to communicate with my computer. It never experienced very high tempretures and operated always between 19-23°C.

My Connections look like this. I'm not very sure about the buttons for boot mode, because these are given by the adapter.

[image]

Is the ESP-Wroom-32 completly broken or can I fix the problem?

projectgus, 2018-05-09:
Hi @max3l,

It looks like the flash chip connected to your ESP32 (inside the WROOM module) is no longer responding. This is usually because of a short or an external connection to GPIOs 6-11 (the SPI flash pins). Or possibly something is connected to GPIO12 and is pulling it high on reset. This sets 1.8V mode which will cause the WROOM internal flash to brown out.

Your schematic doesn't show any connections to these pins, but maybe check the breakout PCB soldering for any shorts (they may not be visible if they're under the module, but a multimeter should be able to detect them). If the WROOM breakout is in a breadboard then try removing it from the breadboard, as breadboards can add a lot of capacitance to the signalling pins.

BTW, the breakout adapter PCB you linked may cause problems with your ESP32 as it doesn't integrate any external power capacitors. The ESP-WROOM-32 module datasheet recommends 10uF + 0.1uF on a breakout board such as this (section Peripheral Schematics), but the breakout designers chose to leave this off.

Angus

max3l, 2018-05-09:
Thank you for your quick reply. 

I checked GPIOs 6-11 for short circuits but there are none. Also there is nothing connected to GPIO 12, which could pull it high. The Board is loose and not connected to a breadboard, only the power pins and the RX/TX pins.

Is there a way to find out if the flash chip is broken? 

Max

projectgus, 2018-05-11:
Hi Max,

You can take the RF can off the WROOM module and attach a SPI flash programming clip, while holding the WROOM module EN line low to keep it disabled. This is probably more trouble than it's worth, compared to swapping in a new WROOM and seeing if this fixes the problem!

Unfortunately the usual way to check the flash chip via the ESP32 itself is to use esptool.py... and esptool.py indicates the ESP32 can't talk to it.

projectgus, 2018-06-13:
Hi @max3l ,

Did you make any progress with this problem?

Angus

max3l, 2018-06-15:
Hi @projectgus,

No, I bought a new WROOM module and am waiting for it to arrive.

Max

projectgus, 2018-07-08:
Hi @FriendlyScientist,

Thanks for contributing this, this does sound like a very plausible cause.

These voltage efuses can be burned from the command line with the espefuse tool:
https://github.com/espressif/esptool/wiki/espefuse#setting-flash-voltage-vdd_sdio

projectgus, 2018-12-20:
Closing due to inactivity.

closed 2018-12-20 by projectgus

hal9000-x, 2019-03-15:
i had the exact same problem. tried a good power supply, no result. but after cleaning and re-soldering the ESP-WROOM-32D module, it's now working.

nitoniso, 2020-01-29:
Hi. in my case something is connected to GPIO12 and that is spi sdo/miso for an lisd3dh accelerometer. how can I solve the issue. I have already soldered everything on the PCB.

garygru, 2020-01-29:
hi @nitoniso, if you can't move to another GPIO, you can try a pull-down resistor. If this doesn't help, try to burn the fuses as described above.

ansh4786, 2020-08-18:
plz someone help i am doing it for the first time

error is occuring that

esptool.py v2.6
Serial port COM5
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:62:ab:fc:0e:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2048.0 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

plz help me

chrismerck, 2020-08-18:
We had the same issue. Problem was we didn't realize that WROVER-E version has different strapping pin requirements (GPIO12) due to having 3.3V- rather than 1.8V-based.

Angus provided a useful link above, but I'll quote the full command needed to override the GPIO12 strapping for WROVER-E:

```python `which espefuse.py` --port /usr/local/tty.bond set_flash_voltage 3.3V```

ansh4786, 2020-08-19:
But how to do i m doing it for the first time

wamiquemohd, 2021-07-05:
Try pressing pulling low EN pin for 1 sec and simultaneously pull the Boot pin low.
Now release the EN pin (Button) and continue holding the boot button and your code will be flashed

#its same process we do for some arm boards

limaarduino, 2022-02-06:
I had this problem and in my case it was the internal flash chip, pin 8, it was cold soldered. I resoldered and flash was detected again.

[image]