espressif/esptool #308: Explain boot log messages in wiki

marcelstoer, 2018-05-16:
https://github.com/espressif/esptool/wiki/ESP8266-Boot-ROM-Log is fairly naked without any explanations. How about adding the explanations from Espressif at https://bbs.espressif.com/viewtopic.php?t=896? That post mentions an official documentation but I failed to find it.

projectgus, 2018-05-30:
I know this wasn't your request, but I've added equivalent explanations for ESP32 here:
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#boot-mode-message

ESP8266 will be coming soon. :)

marcelstoer, 2018-11-12:
Any chance we can get this as a Christmas present 🎄?

projectgus, 2019-10-18:
Looks like @AABatteries added this information to the wiki in January. Thanks, @AABatteries! :)

closed 2019-10-18 by projectgus