espressif/esptool #310: ESP8266 getting stuck after using the stub with no_reset option

MatthiasJentsch, 2018-05-18:
* Operating system: Windows 10 64bit
* Python version: 3.6.5
* ESP hardware in use: ESP8266EX on ESP-12
* esptool version: 2.3.1

# Full esptool.py command line as run:
--port COM6 --baud 115200 --chip esp8266 --after no_reset erase_flash

# What is the expected behaviour?
After executing this command the last line is "Staying in bootloader". But now you can't connect with esptool.py anymore. You need to reset the chip. I found the same behaviour with the chip_id, read_mac and flash_id commands.
However if you use the --no-stub option with the chip_id, read_mac and flash_id command further connections are possible without resetting the chip.
I think there is a problem with the ESP8266 stub program. Maybe it will not exit and prevents further connection.
I would not have to reset the chip after each command.

Maybe that's similar to https://github.com/espressif/esptool/issues/249

projectgus, 2018-06-13:
Hi @MatthiasJentsch ,

Thanks for being patient while I got back to you.

I think the underlying problem is that the stub loader doesn't respond to either autobauding or the sync packet which is used to establish the connection. Can fix the second thing but the first thing is tricky. 

You should be able to use the new (in v2.4.0) `--before no_reset_no_sync` option, though. This option (with a matching baud rate) should just keep sending commands to the stub loader, as if esptool never left.

Angus

MatthiasJentsch, 2018-06-13:
Hi @projectgus ,

I've tested with version 2.4.0 and the **--before no_reset_no_sync** option. But it's not working at all. Here is the log:

```
esptool.py --port COM6 --baud 921600 --chip esp8266   --after no_reset erase_flash
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:9c:d7:d3
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.2s
Staying in bootloader.

esptool.py --port COM6 --baud 921600 --chip esp8266  --before no_reset_no_sync --after no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 512KB 0x0 "xxx.eagle.flash.0x00000.bin" 0x10000 "xxx.eagle.irom0text.bin.0x10000.bin" 0x7C000 "esp_init_data_default.bin" 0x7E000 "blank.bin"
esptool.py v2.4.0
Connecting...

A fatal error occurred: Timed out waiting for packet header
```

MatthiasJentsch, 2018-06-21:
Any news for this issue? I'm still looking for a solution or at least a workaround for it.

Jason2866, 2020-09-03:
Any news on this? The issue is still there