espressif/esptool #311: erase_flash problem

bardhandebashree, 2018-05-22:
(Please delete any lines which don't apply.)

* Operating system:windows 10
* Python version: (`python -V` to check this)3
* ESP hardware in use: esp8266

# Full esptool.py command line as run:PS C:\Users\hp\Downloads\esptool-master\esptool-master> python esptool.py --port COM6 erase_flash

# Full output from esptool.py (please copy and paste all lines of output)
A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0xF0)

# What is the expected behaviour?

`

12345 | esptool.py v1.2.1Connecting…Running Cesanta flasher stub…Erasing flash (this may take a while)…Erase took 0.9 seconds
-- | --

`

projectgus, 2018-05-22:
Hi @bardhandebashree ,

> Full output from esptool.py (please copy and paste all lines of output)
>
> A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0xF0)

Please copy and paste all lines of output from esptool.py.

> esptool.py v1.2.1Connecting…Running Cesanta flasher stub…Erasing flash (this may take a while)…Erase took 0.9 seconds

It's not entirely clear to me what's happening here. Do you have a piece of hardware where esptool.py v1.2.1 currently works to erase but the esptool master branch is failing?

Thanks,

Angus

bardhandebashree, 2018-05-23:
i am not getting why my esptool download tool always fails?
`C:\Users\hp\Downloads\flash_download_tools_v3.6.4_0 (1)\FLASH_DOWNLOAD_TOOLS_V3.6.4\ESPFlashDownloadTool_v3.6.4.exe:96: wxPyDeprecationWarning: Using deprecated class PySimpleApp.
('self.chip: ', 'ESP8266')
RF option applied...
C:\Users\hp\Downloads\esp_iot_sdk_v1.5.0_15_11_27\esp_iot_sdk_v1.5.0\bin\at\noboot\eagle.flash.bin
1 kill
1 kill
C:\Users\hp\Downloads\esp_iot_sdk_v1.5.0_15_11_27\esp_iot_sdk_v1.5.0\bin\at\noboot\eagle.irom0text.bin
2 kill
C:\Users\hp\Downloads\esp_iot_sdk_v1.5.0_15_11_27\esp_iot_sdk_v1.5.0\bin\blank.bin
3 kill
C:\Users\hp\Downloads\esp_iot_sdk_v1.5.0_15_11_27\esp_iot_sdk_v1.5.0\bin\blank.bin
4 kill
======
CONNECT BAUD: 115200
============
....._____....._____....._____....._____....._____....._____....._____....._____....._____....._____[2018-05-23 08:37:13,351][ESP8266Loader_spi[1]][espDownloader.pyo][line:351][ERROR]: Chip sync error: Failed to connect to ESP8266: Invalid head of packet ('\xf0')
[2018-05-23 08:37:13,352][ESP8266Loader_spi[1]][espDownloader.pyo][line:439][ERROR]: ESP8266 Chip sync error esp_sync_blocking.`

and my esptool.py also doesnot work:
 Failed to connect to Espressif device

projectgus, 2018-05-23:
Thanks for verifying that the Flash Download Tool doesn't work correctly either. This is useful information.

However:
> Please copy and paste all lines of output from esptool.py.

projectgus, 2018-06-13:
Hi @bardhandebashree ,

Have you made any progress with this? If not, are you able to provide the information that I asked for in the comments above?

Angus

projectgus, 2018-07-19:
Closing due to a lack of information.

closed 2018-07-19 by projectgus