espressif/esptool #312: write_flash from HTTP url

abdollahpour, 2018-05-31:
It would be good if user can download and write_flash directly from HTTP URL instead of file. When I want to send ROM to my clients I can simply send them a command and because it's Python, it would be valid for any operating system. If I want to use any other tools like curl to download it first it would not be cross platform.

projectgus, 2018-05-31:
Hi @abdollahpour ,

Just to be sure I understand you correctly, the feature request is to be able to do something like:

```
esptool.py -p PORT - b BAUD write_flash 0x0 https://some.web.site/someurl/image.bin
```

Is that correct?

abdollahpour, 2018-06-01:
Exactly!

dobairoland, 2022-11-30:
I'm sorry but I'm closing this for the following reasons:
1. My experience with https://github.com/espressif/esp-idf/blob/master/tools/idf_tools.py tells me that Python-based downloading could not be as reliable as tools specifically designed for that. Python-based downloading uses the operating system's certificates which is in most cases outdated. For that reason, most modern web browsers use their own set of certificates. 
2. I don't think that every tool should be able to download and read e-mails. Using a downloading tool specifically made for that purpose maintained by a team who has knowledge and experience specifically tailored for that purpose sounds more reasonable to me.

closed 2022-11-30 by dobairoland