espressif/esptool #318: New board for magic home?

ZoltanSzilagyiCse, 2018-06-14:
Alright, I just bought a Magic Home LED controller from AliExpress and when i opened it I got this:

[image]

[image]

I flashed both version (the 1.0 and the 2.0 too), then I reboot it (disconnected then connected the power) but the WiFi name never appeared. I think it's one of the cheapest version of this OR this maybe a total new board? (I already flashed 2 ESP modules before, so I did everything correctly)

projectgus, 2018-06-14:
Hi @ZoltanSzilagyiCse ,

Sorry, I'm not sure I understand the issue with esptool here.

ESP8285 is an [Espressif WiFi MCU chip](https://www.espressif.com/en/products/hardware/socs) (first announced in 2016) and is supported by esptool.py.

> I flashed both version (the 1.0 and the 2.0 too)

I'm not sure what this means. Espressif can't provide support for a third party's hardware design using Espressif chips. You'll need to talk to whoever produces the Magic Home product or firmware to get support - or maybe reverse engineer it yourself if no support is available.

If you think that esptool.py is not flashing the ESP8285 chip correctly, please let me know and I can reopen this issue. But this sounds like a firmware problem with this particular device's firmware (maybe the firmware you flashed is not compatible with ESP8285). I'm sorry but I can't help you with this.

closed 2018-06-14 by projectgus

coolineho, 2018-12-14:
i have the same board too, did you manage to flash this?

ZoltanSzilagyiCse, 2018-12-14:
Yes, need to change some pinout number in the source code. It looks like the board has different layout and pinouts. Just a note: I only used the 1.0 version not the newer 2.0.

ZoltanSzilagyiCse, 2018-12-23:
I uploaded the file here: [Google Drive](https://drive.google.com/open?id=1agCNaM6bx8TqClvsVJ2V9b8KqS6vGM00)

Just flash it or upgrade an existing one

coolineho, 2018-12-24:
i got it to flash ;)

thanks