espressif/esptool #319: A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

papahabla, 2018-06-16:
I'm not able to connect to any of the three  ESP8266 Feather HUZZAH boards I have when using version 2.4.0 of the esptool. After downgrading version to 2.3.1, the issue is resolved. If I then upgrade back to 2.4.0, this issue is re-introduced. 

I have tried multiple boards and cables. Also, I have tested many versions of the esptool, version 2.4.0 is the only version to exhibit the issue.

* Operating system: OS 10.12.16 
* Python version: Python 3.6.5 :: Anaconda, Inc.
* ESP hardware in use: ESP8266 Feather HUZZAH

# Full esptool.py command line as run:

`esptool.py -p $UPY_ESP8266_PORT erase_flash`

# Full output from esptool.py

```
(upy) ✔ ~/Make/fauxwindow
12:47 $ pip install esptool
Collecting esptool
Requirement already satisfied: pyaes in /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages (from esptool) (1.6.1)
Requirement already satisfied: ecdsa in /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages (from esptool) (0.13)
Requirement already satisfied: pyserial>=2.5 in /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages (from esptool) (3.4)
Installing collected packages: esptool
Successfully installed esptool-2.4.0
(upy) ✔ ~/Make/fauxwindow
12:47 $ esptool.py -p $UPY_ESP8266_PORT erase_flash
esptool.py v2.4.0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
(upy) ✘-2 ~/Make/fauxwindow
12:48 $ pip uninstall esptool
Uninstalling esptool-2.4.0:
  Would remove:
    /Users/awarner/anaconda2/envs/upy/bin/espefuse.py
    /Users/awarner/anaconda2/envs/upy/bin/espsecure.py
    /Users/awarner/anaconda2/envs/upy/bin/esptool.py
    /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages/espefuse.py
    /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages/espsecure.py
    /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages/esptool-2.4.0.dist-info/*
    /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages/esptool.py
Proceed (y/n)? y
  Successfully uninstalled esptool-2.4.0
(upy) ✔ ~/Make/fauxwindow
12:48 $ pip install esptool==2.3.1
Collecting esptool==2.3.1
Requirement already satisfied: pyserial>=2.5 in /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages (from esptool==2.3.1) (3.4)
Requirement already satisfied: ecdsa in /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages (from esptool==2.3.1) (0.13)
Requirement already satisfied: pyaes in /Users/awarner/anaconda2/envs/upy/lib/python3.6/site-packages (from esptool==2.3.1) (1.6.1)
Installing collected packages: esptool
Successfully installed esptool-2.3.1
(upy) ✔ ~/Make/fauxwindow
12:49 $ esptool.py -p $UPY_ESP8266_PORT erase_flash
esptool.py v2.3.1
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.3s
Hard resetting via RTS pin...
```

projectgus, 2018-06-18:
Hi @papahabla ,

Thanks for reporting this.

I think this may be same underlying cause as #320, the port argument being ignored (esptool v2.4.0 actually tries last serial port which is found, and aborts if this connection is unsuccessful - so in this case it may have been trying to sync with some other serial port). The only other difference in connection behaviour was number of connection attempts (2.3.1 made 10, 2.4.0 makes 7), but the log shows 2.3.1 succeeding on the second attempt.

Could you please try the current master branch (v2.4.1-dev) and let me know if it fixes the problem? You can install via pip with `pip install https://github.com/espressif/esptool.git`.

(If updating to v2.4.1-dev doesn't fix the issue, please re-run esptool.py v2.4.1-dev with `--trace` and post the log output.)

Angus

papahabla, 2018-06-18:
@projectgus Yes, I used release version v2.4.1 and issue is resolved. Thanks!

closed 2018-06-18 by papahabla

projectgus, 2018-06-18:
Great to hear!