espressif/esptool #321: Keeping DTR asserted after reset-to-bootloader

me21, 2018-06-18:
During connect attempt, the DTR line is deasserted after completing reset-to-bootloader sequence: https://github.com/espressif/esptool/blob/d915e7789b7ffcdde3530360cb6241080beef405/esptool.py#L415

Is there any reason for doing so? I'm using STM32 as USB-Serial bridge on my custom board and rogerclarkmelbourne's Arduino core for STM32 interprets inactive DTR state as indication that the port is closed from the host side, discarding any data I'm sending to the host.

projectgus, 2018-06-19:
Hi,

I think it should be possible to keep DTR asserted for the whole time while flashing, this only pulls IO0 low which shouldn't change anything in the flashing process. This long-standing behaviour though, so there might be some side effects I'm not thinking of right now.

Angus

radimkarnis, 2021-04-29:
I am closing this issue because this hasn't been requested ever again in 3 years. Feel free to reopen if you think it should be reconsidered.

closed 2021-04-29 by radimkarnis

me21, 2021-04-29:
I admit this is very specific to my setup, where I use STM32 chip programmed with Arduino core as USB-to-serial converter.So I frankly don't know what's the best course of action. Well, at least this is now documented as an issue, and someone who might have the same problem will find it by keywords.