espressif/esptool #322: esptool.py not connecting to nodemcu on windows

qwerty12345asdf, 2018-06-19:
* Operating system:Windows
* Python version: Python 3.6.5
* ESP hardware in use: nodemcu amica v2.0 

# Full esptool.py command line as run:
the command i ran is 
"esptool.py -port COM4 flash_id"

# Full output from esptool.py (please copy and paste all lines of output)
There is no output it just indicates connecting....
as shown in the screenshot.
[image]
# What is the expected behaviour?
the command should show me device information on successful execution.

#Do you have any other information from investigating this?
Initially I thought that i did not correctly select the port but on checking with device manager i found that i was using the correct COM port.
i have included the device manager screen shot too.
[image]
# Is there any other information you can think of which will help us reproduce this problem?
>I simply installed python 3 on my windows os .
>after installing python i installed pip3 .
>then i installed esptool.
>to check if flashing or reading from the device was working i opened the cmd in admin mode
  and typed command "esptool.py --port COM4 flash_id"

Other info:
>i was able to successfully flash micropython firmware on the device and use the esptool.py command using a linux os (ubuntu).
>if you need any other info pls inform me.
Thanks

projectgus, 2018-06-19:
HI @qwerty12345asdf ,

Thanks for the comprehensive bug report and sorry for the inconvenience. There was a bug in v2.4.0 with port selection, fixed in v2.4.1 which was released on Monday. If you update to this release then it should work.

If you're still seeing problems with v2.4.1, please let me know and I'll reopen the issue.

(See #320 for more details)

Angus

qwerty12345asdf, 2018-06-21:
Thanks, the issue has been resolved 👍

qwerty12345asdf, 2018-06-21:
Although i did notice one small issue, when using the auto find feature for port selection it tries to connect to one of the ports and does not go to the next one.
I have included a screenshot of the same 
>In the first command "esptool.py --port COM4 flash_id" esptool successfully connects to the device but
>When i try "esptool.py  flash_id"(the autoconnect feature) it stays on connecting...
[image]

projectgus, 2018-06-21:
Hi @qwerty12345asdf,

That's interesting, thanks. What's connected to COM7? If you run `esptool.py --trace flash_id`, what output do you get?

Angus

qwerty12345asdf, 2018-06-21:
I ran the trace command the output is as shown in the screen-shot also there is nothing connected to any other COM port other than COM4 (my esp8266 nodemcu).
[image]

I noticed that the process is not complete as the cursor keeps blinking after executing the command.

projectgus, 2018-06-21:
> I noticed that the process is not complete as the cursor keeps blinking after executing the command.

Thanks. What I would expect to see here is some more `...___...` type output, and then esptool moves to the next port after a few seconds.

* What driver/device is COM7 (internal motherboard COM, usb/serial adapter, something else?)
* If you run `esptool.py --port COM7 flash_id`, is the behaviour any different?

Thanks again for the additional info,

Angus

qwerty12345asdf, 2018-06-21:
>The COM7 port seems to be a bluetooth link as shown in the device manager
[image]
>on running "esptool.py --port COM7 flash_id" it is stuck on connecting...  similar to the earlier scenario.

qwerty12345asdf, 2018-06-21:
Hey, small update
Since the device was not connecting I to COM7 i decide to try and update the drivers for that port nothing much happened, i.e the driver was already up-to-date 
>But now a new output has cropped up for the same commands as shown in screenshot:
inputting the command "esptool.py flash_id" i.e autoconnect  the tool tries to connect to COM7 fails and stops.
[image]
>I don't know why is it indicating Access denied because cmd is running on admin mode.
*Note:
>Connection to COM4 directly has no problem either :
[image]

*Hope it helps, pls. inform if you require any other info.

projectgus, 2018-06-21:
Ah OK, this is useful.

I'm guessing the BT connection was stuck in some way (so the write to the serial port was hanging while the driver tried to establish the connection to the BT device) and now it's timed out or something and realised the connection is permanently down.

I think two things need to happen here:
* Allow for serial port exceptions (like access denied) while scanning, so the scan continues.
* See if timeouts can be applied when writing to Windows serial ports (it's possible the driver is hanging internally, in which case we can't time out from the esptool end, but I'll take a look).

Thanks!

projectgus, 2018-07-19:
Hi @qwerty12345asdf ,

Thanks for your patience while I looked into this.

Setting a write timeout on the serial port was added in #326 and is part of esptool v2.5.0.

Correctly handling serial port OSErrors while scanning has been added to master branch now, and will be in the next release.

Angus

closed 2018-07-19 by projectgus

marcelstoer, 2018-08-25:
Angus, I tried the auto-port feature on 2.5 for the first time today. What I got was this:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError: [Errno 16] Resource busy: '/dev/cu.nuraphone392-CSRGAIA-2'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 2849, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 2842, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2543, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/usr/local/bin/esptool.py", line 244, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "/usr/local/bin/esptool.py", line 213, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/local/lib/python3.6/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.6/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 16] could not open port /dev/cu.nuraphone392-CSRGAIA-2: [Errno 16] Resource busy: '/dev/cu.nuraphone392-CSRGAIA-2'
```

The `OSError` there at the beginning looks like what you fixed in cac97e1d64f8de22e673e3d943e976b28b0d3b12, is that right? Hence, in the upcoming 2.5.1 or 2.6 this will be fixed?

projectgus, 2018-09-27:
@marcelstoer Sorry for the slow reply. This should be fixed in v2.5.1 (just released).

commit cac97e1d (2018-07-19): Handle OSError correctly when scanning for port

Closes #322