espressif/esptool #323: Under VMWare fusion with USB2.0 controller, CH340 has glitch on RESET which prevents bootloader from starting

jeremyherbert, 2018-06-26:
This is a really bizzare one. I don't think there is anything you can do, feel free to close this. I'm reporting it to hopefully save someone else some time tracking this down and to make sure it shows up in google searches. 

Board: Lolin D32 Pro (ESP32-WROVER), uses a CH340 USB to serial converter.

I have a Xubuntu 18.04 VM running under VMWare fusion on OSX High Sierra. If the USB controller type is set to USB2.0, when performing the reset sequence, there is a glitch on the RESET line when releasing IO0 which causes the device to reset and not enter bootloader mode:
[image]

However, if I shut down the VM, change the controller type to USB3.0 and boot it up again, the glitch disappears and the board programs fine. I have no idea if the issue is in the linux kernel or in VMWare, or if there is a workaround. I tried removing the windows workaround from `_setRTS`, but it doesn't change anything.

jeremyherbert, 2018-06-26:
Actually, looking at it now, I have my labels the wrong way around; the glitch is on the IO0 line, not the RESET line. But everything else stands.

projectgus, 2018-07-03:
Hi @jeremyherbert ,

Thanks for the detailed report. This looks like a variation on #136, where USB host stack timing and the "auto reset circuit" creates this short glitch on the IO0 line (when both DTR & RTS are momentarily asserted) which can trick an ESP32 into not booting into the bootloader correctly.

We fixed this on Espressif development boards by adding a capacitor to the EN line, so it rises slightly slower. The D32 Pro schematic indicates it also has such a capacitor (C4), but you could try increasing capacitance on this pin and see if it works around the issue. It may also be that the VMWare host introduces enough additional USB latency that it's not possible to work around it with capacitance only.

I'm going to close this as I don't think there's anything we can do about in esptool itself, but please let me know if you think there is something.

closed 2018-07-03 by projectgus