espressif/esptool #324: esptool.py AttributeError: 'Serial' object has no attribute 'dtr'

gojimmypi, 2018-06-29:
* Operating system: 
Debian 8.x 64 bit running in a VMWare 14.1.2 build-8497320 VM Workstation on Windows 10 host
Linux version 3.16.0-6-amd64 (debian-kernel@lists.debian.org) (gcc version 4.9.2 (Debian 4.9.2-10+deb8u1) ) #1 SMP Debian 3.16.56-1+deb8u1 (2018-05-08)
(Removable device manually connected to VM via menu removable devices, disconnected from host)

* Python version: (`python -V` to check this)
Python 2.7.9 (default, Jun 29 2016, 13:08:31)
[GCC 4.9.2] on linux2

* ESP hardware in use:
Amica NodeMCU ESP8266 w/cp2102

# Full esptool.py command line as run:
`~/workspace/esptool/esptool.py --port /dev/ttyUSB0 --baud 115200 flash_id`

# Full output from esptool.py 
```
esptool.py v2.4.1
Serial port /dev/ttyUSB0
Connecting...
Traceback (most recent call last):
  File "/home/gojimmypi/workspace/esptool/esptool.py", line 2821, in <module>
    _main()
  File "/home/gojimmypi/workspace/esptool/esptool.py", line 2814, in _main
    main()
  File "/home/gojimmypi/workspace/esptool/esptool.py", line 2515, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/home/gojimmypi/workspace/esptool/esptool.py", line 242, in detect_chip
    detect_port.connect(connect_mode)
  File "/home/gojimmypi/workspace/esptool/esptool.py", line 441, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "/home/gojimmypi/workspace/esptool/esptool.py", line 400, in _connect_attempt
    self._setRTS(True)   # EN=LOW, chip in reset
  File "/home/gojimmypi/workspace/esptool/esptool.py", line 372, in _setRTS
    self._port.setDTR(self._port.dtr)
AttributeError: 'Serial' object has no attribute 'dtr'

```

# What is the expected behaviour?
return flash_id of device without error

# Do you have any other information from investigating this?
I'm trying `esptool.py` in this Debian VM as I have other esptool.py errors on the Raspberry Pi as noted here:

https://github.com/espressif/esptool/issues/269#issuecomment-401212234

The RPi is giving a `ImportError: No module named tools.list_ports` error that I've also been unable to resolve; wondering if both issues are related due to serial library. The esptool.py definitely used to work with the RPi using my MicroPython building script here:

https://github.com/gojimmypi/build_esp8266/

I also tried a completely different board: the Geekcreit ESP12E DEVKIT V2 with the same results.

gojimmypi, 2018-06-30:
The solution was found in the https://github.com/espressif/esptool#manual-installation instructions:

> esptool.py depends on pySerial version 3.0 or newer for serial communication with the target device.

despite a fresh apt-get update/upgrade, I found my version reported was still old (2.6.1):
```
~$ apt-cache policy python-serial
python-serial:
  Installed: 2.6-1.1
  Candidate: 2.6-1.1
  Version table:
 *** 2.6-1.1 0
        500 http://ftp.us.debian.org/debian/ jessie/main amd64 Packages
        100 /var/lib/dpkg/status
```
So to fix, ensure pip is installed:

`sudo apt-get install python-pip`

Try to upgrade python-serial:

`sudo pip install --upgrade pyserial`

Note results show pyserial-3-4 found, but IGNORED since existing version is "owned by OS":

```
Downloading/unpacking pyserial from https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl#sha256=e0770fadba80c31013896c7e6ef703f72e7834965954a78e71a3049488d4d7d8
  Downloading pyserial-3.4-py2.py3-none-any.whl (193kB): 193kB downloaded
Installing collected packages: pyserial
  Found existing installation: pyserial 2.6
    Not uninstalling pyserial at /usr/lib/python2.7/dist-packages, owned by OS
Successfully installed pyserial
Cleaning up...
jscarletta@ha-vm-debian:~$ apt-cache policy python-serial
python-serial:
  Installed: 2.6-1.1
  Candidate: 2.6-1.1
  Version table:
 *** 2.6-1.1 0
        500 http://ftp.us.debian.org/debian/ jessie/main amd64 Packages
        100 /var/lib/dpkg/status
```

So the issue needs to be forced:

`pip install --ignore-installed pyserial`

Then... tada!  `esptool.py --port /dev/ttyUSB0 --baud 115200 flash_id` results in:

```
esptool.py v2.4.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

closed 2018-06-30 by gojimmypi

projectgus, 2018-07-03:
Hi @gojimmypi ,

I'm glad you found a solution. The other way to do this is to create a virtualenv (dedicated Python install with only the modules you need), which can be separate from the OS Python.

You can read about virtualenvs here: https://virtualenv.pypa.io/en/stable/installation/ or a slightly simpler tool called pyenv, here: https://github.com/pyenv/pyenv#README