espressif/esptool #327: Cannot install on OSX following instructions letter-by-letter

markaaronky, 2018-07-06:
(Please delete any lines which don't apply.)

* Operating system:Mac OSX 10.11.6
* Python version: (`python -V` to check this)2.7.10
* ESP hardware in use:NODEMCU

# Full esptool.py command line as run:pip install esptool

# Full output from esptool.py (please copy and paste all lines of output)Could not find a version that satisfies the requirement esptool (from versions: )
  No matching distribution found for esptool

# What is the expected behaviour? esptools should be installed

# Do you have any other information from investigating this?The original error message complained that the requirement "pyserial>=3" wasn't met.  I'm on an older Mac, I assume this is referring to the type/speed of the USB port I'm using.  I changed the requirement from 3 to 2; still get a failure.

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2018-07-09:
Hi @markaaronky,

Thanks for contacting us, sorry esptool isn't working for you.

> No matching distribution found for esptool

That's unusual. esptool certainly should be available in a default pip install (the [project is here](https://pypi.org/project/esptool/)). Was your internet connection maybe down or misbehaving at the time you were running pip?

If you try `pip install esptool` again and it's still not working, can you please try running `pip -v install esptool` and then copy-paste the entire output from pip into this issue.

> The original error message complained that the requirement "pyserial>=3" wasn't met. I'm on an older Mac, I assume this is referring to the type/speed of the USB port I'm using. I changed the requirement from 3 to 2; still get a failure.

Not exactly. There's a [Python library called pyserial](https://pypi.org/project/pyserial/) that esptool relies on, and it needs version 3.0 or newer of this library. The pyserial version numbers don't relate to USB 2.0 vs USB 3.0 at all, that's an (unfortunate) coincidence.

esptool won't correctly if pyserial isn't version 3.0 or newer.

projectgus, 2018-07-19:
@markaaronky Did you make any progress with this issue?

vincnico, 2019-01-16:
Hi
I am totally new here and with esptool.
I follow some tuto to try to install esptool on my macbook with OS 10.10.5 Yosemite, in order to flash a nodemcu with ESPeasy.
But it appears I have the same issue : 
$ pip install esptool

You are using pip version 6.1.1, however version 18.1 is available.

You should consider upgrading via the 'pip install --upgrade pip' command.

Requirement already satisfied (use --upgrade to upgrade): esptool in /Library/Python/2.7/site-packages/esptool-2.7_dev-py2.7.egg

Collecting pyserial>=3.0 (from esptool)

  Could not find a version that satisfies the requirement pyserial>=3.0 (from esptool) (from versions: )

  No matching distribution found for pyserial>=3.0 (from esptool)

Does someone can help me ?
Thank you

projectgus, 2019-01-16:
> You are using pip version 6.1.1, however version 18.1 is available.

I think you probably have the same issue as SabeehY (above) where the version of pip is so old it can't verify the pypi archive server TLS certificate is valid any more.

Suggest trying the command posted by SabeehY above, or research another way to update pip on Yosemite.

vincnico, 2019-01-16:
hello

I tried : 
curl https://bootstrap.pypa.io/get-pip.py | python
but I have the message :
Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/pip-6.1.1.dist-info/DESCRIPTION.rst'
Consider using the `--user` option or check the permissions.

I have no idea what to do....
Does anybody can help ???

vincnico, 2019-01-16:
I have also tried:
sudo pip install --upgrade pip

but I receive: 
The directory '/Users/vincentnicolas/Library/Logs/pip' or its parent directory is not owned by the current user and the debug log has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/vincentnicolas/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
You are using pip version 6.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
The directory '/Users/vincentnicolas/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already up-to-date: pip in /Library/Python/2.7/site-packages

...? ? ? what to do ?????

vincnico, 2019-01-16:
I also tried:
$ pip install --upgrade pip
You are using pip version 6.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Requirement already up-to-date: pip in /Library/Python/2.7/site-packages

which doesn't seem consistent to me... any idea ?

projectgus, 2019-01-17:
You seem to maybe have a local pip install in your user directory, and a system-wide one. You can confirm by running `which pip`.

Try `pip --user --upgrade` (no sudo) to upgrade the user-specific one.

vincnico, 2019-01-17:
hi
something new ! 
I tried:
$ python get-pip.py --user

which seems to give something positive, but I received:
The script wheel is installed in '/Users/vincentnicolas/Library/Python/2.7/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed pip-18.1 wheel-0.32.3

so I tried to add to PATH with:
export PATH=$PATH:/Users/vincentnicolas/Library/Python/2.7/bin

and then relaunched the command: pip install esptool
but still have:
You are using pip version 6.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Requirement already satisfied (use --upgrade to upgrade): esptool in /Library/Python/2.7/site-packages/esptool-2.7_dev-py2.7.egg
Collecting pyserial>=3.0 (from esptool)
  Could not find a version that satisfies the requirement pyserial>=3.0 (from esptool) (from versions: )
  No matching distribution found for pyserial>=3.0 (from esptool)

what is the problem?

projectgus, 2019-01-20:
If you type `which pip`, you'll probably find that there is a system-wide pip which is being run before the local one you just installed.

To preference the local one, put it first in the PATH list - ie:

```
export PATH=/Users/vincentnicolas/Library/Python/2.7/bin:$PATH
```

vincnico, 2019-01-20:
Hi
Thanks for this, it seems I'm doing progress....
So did I.
But then, I have another one:
$ pip install esptool
Traceback (most recent call last):
  File "/Users/vincentnicolas/Library/Python/2.7/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named _internal

any idea ?